nguy/artview #3: ppi/rhi check too restrictive

tjlang, 2015-02-20:
ARTview appears to only consider the possibility that valid radar.scan_type attributes can be 'rhi' or 'ppi'. In fact, valid, plottable scan types include 'sector' and possibly other names. I changed the 'ppi' test in the #CreatePlot section from
if self.radar.scan_type == 'ppi':
to 
if self.radar.scan_type != 'rhi':
And I was able to plot up my 'sector' scan_type files just fine. I recommend more flexible checking for PPI vs. RHI style scans, or to just remove the tests and let ARTview fail with a warning if the scan type is really odd. 

tl;dr - Right now, to avoid the possibility that the program might fail on an unusual volume, the program is failing on perfectly valid volumes.

nguy, 2015-02-20:
@tjlang Thanks for the input! 
Paul H also had trouble with a surveillance scan, with what looks like the same reason.

I took your suggestion and it seems to work well with sector files I have locally.  This should also take care of surveillance scans.  I also changed another section to reflect this looser structure.

nguy, 2015-03-06:
@tjlang I think that fix worked for you.  Paul was having the same issue and the fix worked for him.

tjlang, 2015-03-06:
Looks to be fixed, agreed.

closed 2015-03-06 by tjlang

nguy, 2015-03-06:
Thanks for checking this @tjlang