nguy/artview #6: Zoom and Pan capability

nguy, 2015-03-23:
@pfhein
Paul added zoom and pan functionality into ARTView.  Values were not being passed back.

I worked through this issue I believe and both functionalities seem to work.  In addition, they now report back to the new popup window to modify all limits.

I generalized the code a bit more.  Check out the new version I uploaded this morning and see what you think.  You should be able to 'fetch' and then merge with your fork if you like the changes.

I am thinking that I would like to try to make the Zoom/Pan function a selectable option from a dropdown menu.  What do you think?

nguy, 2015-03-23:
Also, made it optional via a button menu (ComboBox).

nguy, 2015-03-23:
Hmmm, just realized that when the the scroll is connected to zoom, it doesn't disconnect.

pfhein, 2015-03-23:
I am just looking over your version right now.  I am glad to see you 
have figured out how to get all the limits in one box.  I was hitting 
brick walls trying to do that.  I do see that zoom is not disconnecting 
when you go back to no tools.  Right now there is way too much plotting 
going on and that really is slowing things down. I like the idea of 
turning zoom on and off and having it one of several future tools.

nguy, 2015-03-24:
Okay, got a little obsessed with this and I think I have a solution.  I'll post the results soon. 

I had to disconnect the events, which is simple enough in theory, but it took me a while to figure out that this had to be done within the ZoomPan class.  I reconfigured the ZoomPan class to now have a connect() and disconnnect() function.  I also renamed a few variables to something I thought was more indicative (that's open for debate).

By the way what is the format_coord supposed to do?

pfhein, 2015-03-24:
Great!  For me nothing is coming easy.  The format_coord is suppose to 
display the coordinates (and the value at that point) but it has never 
worked in artview.py.  It should be removed for now.

nguy, 2015-03-24:
Gotcha.  Well it's not hurting anything for now, but we can start a new issue that could address this later.

nguy, 2015-03-25:
Going to close this issue now.  The zoom/pan seems to be working.  The the format_coord is addressed in another Issue.

closed 2015-03-25 by nguy