nguy/artview #12: Using classes

gamaanderson, 2015-04-16:
So I will just write something here, as to force myself to get to work.

I have some experience in dynamical visualization and I believe to have some good ideas about it. I have this in bit-bucket (https://bitbucket.org/gamaanderson/radar_viewer), but I wouldn't recommend since it is really a mess, but it work really good for what I use.

Any way, what I learned is that visualization software should use classes to different functions, in your case that initially means manipulating files should be separated from visualizing, that would allow for example visualizing the same file with two windows side by side but with different moments but the same elevation. Handling this is a little trick, but there is a nice way for doing it.

Ok, this was terrible writing, I can't really explain what I mean, but I will do some work in the code to show it.

nguy, 2015-04-17:
@gamaanderson This has grown beyond it's humble beginnings.  The original all-in-one file is no longer the best method once functionality began to grow beyond just looking at an image.  I've been mulling over how to reconfigure and I think I agree with the separation you propose.  I look forward to what you come up with, especially for the "trick" for handling the different classes and windows.

gamaanderson, 2015-04-19:
Please have a look into https://github.com/gamaanderson/artview. It is just an example, so functionalities are intentionally turned off or just broken, even if the menus are still there. 

There are only two things one can do, change tilt with arrow up/down and change file inside the directory with arrows left/right. Please take care with focus window to guaranty you are sending the key event to the correct window. This in the case o tilt, on the other hand files are govern by the main menu, but the displays are re-directing those events to the parent, so no problem there. 

There are three files exampling different configurations:
1. execute_simple.py: just one menu and one display
2. execute_two.py: one menu and two displays. The displays share the same radar, but have different tilts (focus the display and move up/down to see elevation change without changing the other display).
3. execute_copy.py: one menu and two displays. Displays share same radar and same tilt, that is, changing tilt in one display automatic change the other (not really useful, I agree) .

please compare the execute file to see how small and similar they are. This is exactly the point, this allows:
* first, while developing a new function one must not (directly) considered how the function integrate with the rest; this can be easily configured in the execute file later, in the case of easy interaction, more complicated interaction may need some thinking. 
* second, one can easily adapt the code to combine the functionalists he needs
* third programmer are less likely to get in the way of each other, since the different parts of the code are not only physically separated, they don't know about the existence of each other (the menu can exist with no display, and the display can exist with no menu, test it!).

execute like this
python <execute file> -d <directory> -f <field> (default field is 'reflectivity')

Hope you like and I look forward for comments

P.S. in case some one is asking, I'm working/contributing with pyart, and when I heard they were starting to develop a visualization tool I had to give it a look

nguy, 2015-04-28:
@gamaanderson Okay, I've been tinkering with this for the last couple of days (I was at a workshop all last week, which of course did not have a solid internet connection).

I think I am now getting a grasp on what you are doing here, but I need to think this through a bit more and figure out how to link changes to the specific cases of each display. 

I actually recall seeing your name in some PyArt development.  Thanks so much for contributing here.  You'll be listed as a contributor in the next "release".  

I will probably have some more questions/comments soon as I begin to transform the code a bit.  I'd like to modularize it a bit more by stashing some functions in  **ui**, **resources**, and maybe **scripts** directories.  The idea moving forward is to give some functionality that the NCAR Solo package had and to connect more easily to existing PyArt modules.

gamaanderson, 2015-04-28:
Okay @nguy thank you for spending your time trying to understand my ideas. I also have several other comments, but I will wait as question appear naturally.

As I never really develop this for a non-developer/inexperience user, I usually concentrate new commands in the keyboard, I was just simpler not to think about creating buttons. I usually navigate with the arrow and use a radio menu in the display to change moments, the color-bar is also a separated display with its own functionalities. That is why, I don't have a closed solution on how to structure menus,  nevertheless one could find a way of allowing displays to add/remove its own buttons/menus to the main window.

nguy, 2015-04-30:
@gamaanderson Questions:
How does the following line in plot.py work?
```python
        self.Vtilt = Vtilt
        QtCore.QObject.connect(Vtilt,QtCore.SIGNAL("ValueChanged"),self.NewTilt)
```
I don't see where the variable and value arguments come from?

I emulated your menu.py and plot.py to form a tilt radiobutton selection:
```python
## tilt.py
# Load the needed packages
from PyQt4 import QtGui, QtCore
from functools import partial

class TiltWindow(QtGui.QMainWindow):
    '''Class to display the Window with Tilt Buttons'''

    tiltClicked = QtCore.pyqtSignal()
    
    def __init__(self, Vradar, Vtilt, MainDisplay, name="TiltButtons", parent=None):
        '''Initialize the class to create the interface'''
        super(TiltWindow, self).__init__(parent)
        self.parent = parent
        self.name = name
        self.setWindowTitle(name)
        self.Vradar = Vradar
        self.MainDisplay = MainDisplay
        
        self.Vtilt = Vtilt
        QtCore.QObject.connect(Vtilt, QtCore.SIGNAL("ValueChanged"), self.NewTilt)

        self.CreateTiltWidget()
        self.setRadioButton = self.SetTiltRadioButtons()
        self.radioBox.show()
           
    ########################
    # Button methods #
    ########################
        
    def TiltSelectCmd(self, ntilt):
        '''Captures a selection and redraws the field with new tilt'''
        print ntilt
        #self.tiltClicked.emit()
        self.Vtilt.change(ntilt)

    def CreateTiltWidget(self):
        '''Create a widget to store radio buttons to control tilt adjust'''
        self.radioBox = QtGui.QGroupBox("Tilt Selection")
        self.rBox_layout = QtGui.QVBoxLayout(self.radioBox)
                
    def SetTiltRadioButtons(self):
        '''Set a tilt selection using radio buttons'''
        # Need to first create each tilt button and connect a value when selected
        tilt_indices = self.Vradar.value.sweep_start_ray_index['data'][:]
        elevs = self.Vradar.value.elevation['data'][tilt_indices]
        for ntilt in self.Vradar.value.sweep_number['data'][:]:
            btntxt = "%2.1f deg (Tilt %d)"%(elevs[ntilt], ntilt+1)
            tiltbutton = QtGui.QRadioButton(btntxt, self.radioBox)
            QtCore.QObject.connect(tiltbutton, QtCore.SIGNAL("clicked()"), \
                         partial(self.TiltSelectCmd, ntilt))

            self.rBox_layout.addWidget(tiltbutton)
		
        self.radioBox.setLayout(self.rBox_layout)
        #self.setCentralWidget(self.radioBox)
		
        return self.radioBox
    
    def NewTilt(self, variable, value):
        self._send_update()
        
    def _send_update(self):
        '''Send the signal for update in the plots'''
        self.MainDisplay.plot1._update_plot()
        self.MainDisplay.plot2._update_plot()
```
By adding the following lines to one of the execute scripts, I get the instance to display. 
```python
tiltbutton = TiltWindow(Vradar, Variable(0), MainDisplay, name="Tilt Selection", parent=MainMenu)
```
When I then click on a radiobutton, the output shows that the command is working, but the plot figures are updated.  Any ideas?

nguy, 2015-04-30:
Oh and to follow up on one point you made earlier.  We want to keep the keyboard shortcuts for the "power user", but also want to make the GUI interface more friendly for more potential widespread useage.  So I'd like to connect things like the tiltbutton above and enable them as we go along.  Once I get a better understanding of the functionality you've developed, I'd like us to port the features over.  There is potentially a reorganization then to make the package installable.

gamaanderson, 2015-04-30:
Ok, quick answer to the first question before I see the rest. This is exactly the point, I want plots, menus etc. to share variable without one knowing about the other. So I can't do Display(... tilt ...), I must put the value inside a mutable class (Vtilt), that is why there is a Variable class (in core.py) with the attribute .value that holds that value. This is the python equivalent of using a pointer in C.

However I also want the widgets to respond when someone else (or it self) change the value, for that one shall never do

    Vtilt.value = 1
but instead

    Vtilt.change(1)
this will make Vtilt send the signal "ValueChanged", 

so the line

    QtCore.QObject.connect(Vtilt,QtCore.SIGNAL("ValueChanged"),self.NewTilt)
say:
whenever `Vtilt` send "ValueChanged" execute `self.NewTilt`

`self.NewTilt` don't new to get the value, since is already in self.Vtilt.value,

gamaanderson, 2015-04-30:
Nevertheless, NewTilt does receive variable and value, this is set in core.py

    self.emit(QtCore.SIGNAL("ValueChanged"),self,value)
that is (self,value) is passed as (variable,value). However I didn't used those anywhere, they are just usable in the strange circumstance where I don't have self.V*** .

gamaanderson, 2015-04-30:
So, I just give it a look and have some comments. As it is, it should work as long as you connect the tilt with the plots, that means:

    Vtitl = Variable(0) # just one instance and all classes share that same instance
    plot1=plot.Display(Vradar,Variable(field),Vtitl,name="Display1",parent=MainMenu)
    plot2=plot.Display(Vradar,Variable(field),Vtitl,name="Display2",parent=MainMenu)
    tiltbutton = TiltWindow(Vradar, Vtitl, MainDisplay, name="Tilt Selection", parent=MainMenu)

However you do not need method `_send_update` nor variable `MainDisplay`, those updates are automatic as I explained above. You also don't need method `NewTilt`, but you may want to use it to move the Check if someone else changed Tilt (e.g. Display with arrow keys), this is done by

    tiltbutton.setChecked(True)
for that you will need to save the buttons in a list

The reason I pass MainMenu to other windows is that when MainMenu closes the other ones also close. This is not happening with the TiltWindow, although parent is correct, this is because you are not showing TiltWindow (`self.show()`) but rather radioBox. It believe you should set TiltWindow as parent of radioBox 

    self.radioBox = QtGui.QGroupBox("Tilt Selection",parent=self)
    self.setCentralWidget(self.radioBox)
    self.show()

nguy, 2015-05-01:
@gamaanderson thanks for the explanations and further examples.  I missed that I had .show() on the wrong instance.

I really think this is going to be a powerful way to work with ARTView going forward.  I've copied you files with modifications into the package and uploaded.  Take a look at these for reference please.  Also note, I changed a couple names for clarification making them slightly different than our discussion above.

First a thought:
So in playing around, I think it would be nice to have a "basic" and "regular" launch.  The "basic" being actually a power user mode with no buttons, etc. - just for browsing and such.  While the "regular" brings up buttons, like the ability to launch tilt selection, etc.  What do you think?  This may take some planning on the front end to interface properly.

As a proof of concept, I added some buttons back to the display is the plot.py script.  So now the TiltButtonWindow can be called from the individual plots and the functionality works.  However, if you run the **execute_two.py** script, the TiltButtonWindow that pops up does not send the signal back to the displays to change.  The change is recorded and sent back to the Variable instance in core.py, but no action is taken.  I believe this is because I am passing the Vradar with None value into the TiltButtonWIndow class.  I thought passing a specific Vradar from the exisiting plot1 or plot2 would work, but it did not.

gamaanderson, 2015-05-01:
@nguy Thank you for add my code, so I may make PR as we work further. Also thank you for changing the name, I was even going to suggest that. I do fill the need to develop a better nomenclature.

The Idea of having two launch may help, but I'm still interested in creating a way of satisfy both need in one design. Just as an initial point, I feel a drop-down menus is better for changing the tilts, I don't see the need of spending that much visual space in that task, changing fields on the other hand may justify a radio since the names may change from one file to the other.

Either way I don't think they should be floating around as windows, but rather place them inside the plot, even if that means they will repeat when there are two plots. Two points for that: first I consider field and tilt as two fundamental variables for a plot, that represent what sub-set of a file is being displayed, and therefore the plot should control them. Second I want to let the execute file as clean as possible, this will not always be possible, but in this case it may be.

I believe it would be useful now to make a list of the functionalities that you want to implement, so we may discuss where each one fits.

Now I will investigate what is happening with the **execute_two.py** and come back.

gamaanderson, 2015-05-01:
The problem with **execute_two.py** is that TiltButtonWindow don't have the same tilt as plot 1 and 2. When you pass `Variable(0)` to TiltButtonWindow you are creating a new variable that has nothing to do with the one you passed to plot. Do this instead
    Vtilt = Variable(0)
    plot1 = Display(Vradar, Variable(field), Vtilt, name="Display1", parent=MainMenu)
    plot2 = Display(Vradar, Variable(field), Vtilt, name="Display2", parent=MainMenu)
    tiltselect = TiltButtonWindow(Vradar, Vtilt, name="Tilt Selection", parent=MainMenu)

P.S. we do initialize Vradar with None, but in line

    MainMenu = Menu(Vradar, DirIn, name="Menu") #initiate Vradar
Vradar is already being updated, so by the time you call  TiltButtonWindow it is no longer None, but rather a pyart radar.

gamaanderson, 2015-05-01:
I remove my comment about the tilt as window, the Tilt select button is a great idea. The only thing that could be better is if that button could work both, they way it is now and as drop-down. May be a small arrow on the side could open the drop-down.

closed 2015-05-12 by gamaanderson