nguy/artview #14: Global Variables

gamaanderson, 2015-05-05:
It was suggested to include of having the drop-down tilt menu changing local tilt and the TiltWindow changing the global tilt. I open this issue to discuss this.

As for me, this is not possible, not should it be possible. I will explain why is that bellow.

gamaanderson, 2015-05-05:
I'm not saying I don't like the idea, but it have severe consequences for the code.

The current structure is based in the idea that **everything is local** and should work locally, there is not global variable and global structure is just the sum of local ones. Local structures can share or not variables, that is all.

This reflect in the fact that the `Variable` class do both: hold the value and emit `"ChangeValue"`signal. To make the kind of selectable alteration suggested one must separate this two functions in `self.tilt`and `self.Vtilt`. This is the first problem, increase code complexity.

But my fundamental problem is that having global variables implies we are thinking in a specific global structure, this result that alternative configurations will must probably suffer with unwanted behavior. Also, programing turns more difficult, since with every variable I will have to care where to use the local one and where to use the global/shared one, this reflect in time. And the most time we spend in this is the less time spent in functionalities.

gamaanderson, 2015-05-05:
As an alternative for this, I would be in favor of a class that connect/disconnect variables of other classes. I, however, didn't have a good idea of how to perform this, but I believe classes show need to inform what are the sharable variables. Uniform naming of those variables would also be important.

nguy, 2015-05-05:
@gamaanderson I'll need to wrap my head around this for a bit.  I'm new to this particular style with ```python Variable``` class.  I think I see how this works now, but I don't have a full understanding of the effects quite yet.  Your comments have been most helpful.

I agree that this functionality is a challenge in our current setup.  I like the connect/disconnect idea.  Maybe for now we can form a better plan and comment here as ideas strike us and tinker around.  I think the current way is not detrimental.

Let's leave this open, but I'll concentrate on other parts first before we tackle this issue.  Sound good?

gamaanderson, 2015-05-05:
Sound perfect

gamaanderson, 2015-05-11:
Just a heads up, I'm working in the connect/disconnect idea in a [branch](https://github.com/gamaanderson/artview/tree/issue_%2314), and its is becoming a big change. Everything works the same way but I had to create a abstract Class from which all components (Menu, Display, FieldWindow etc.) are derived from.  Components must also inform which variables are the shared/sharable. Its turning good, when I feel it is mature I will open a PR. Till then continue pushing commits as you like, I will make sure of keeping the branch up-to-date.

nguy, 2015-05-12:
Sounds interesting @gamaanderson can't wait to see it!

nguy, 2015-06-23:
@gamaanderson I'm going to close this issue if you have no objections. I think most of the issues are deprecated and some have been addressed (or will be) in other issues.

gamaanderson, 2015-06-23:
I was also thinking of closing this. I believe all needs are covered by `artview.components.component_control` and `artview.core.VariableChoose`

closed 2015-06-23 by gamaanderson