nguy/artview #15: Toolbox in Display

gamaanderson, 2015-05-05:
I open this to discuss how the Toolbox display should work.

gamaanderson, 2015-05-05:
Before I can form my mind about this I would like to make some questions:
1. Why do we need to select zoom/plot, can't it be active all time? There is a historical comment about this not being the best way. What was the original problem that motivated going for this solution?
2. Shouldn't restoring default be a button in the same dialog used for modifying it, do we need a general restore?
3. I would like some examples of custom functionalities that would be added in this box. Depending on that I could suggest to put those in the main menu. Also if they are really complex, they could have a display of it own: reusing code is important, but when the cost of doing that is too big, we should not be afraid of coping, even things that look similar in the beginning can evolve to become really distinct.

nguy, 2015-05-05:
I'll try to answer your questions the best I can.

1. The issue with Zoom/Pan was that the wheel scroll was being interpreted as a tilt or file change.  I think this had something to do with the way the original ComboBox menu was structured.  I was having similar problems (up-down arrows were going through the ToolBox ComboBox menu instead of being related to tilt) when I put it back in yesterday.  After changing to the drop down menu, this behavior went away.  
A second reason is that if using a touch mouse, they can be quite sensitive and send a signal out when not intended.

2. I have no objections against a restore for each button.  I only put the general restore in to get back to default conditions if mucking around with various tools.  This is mostly due to my laziness.

3. There are a couple of examples that will hopefully be coming soon.  I'll post again here when I add one.  My feeling is the main menu is becoming less useful as modifications will be done on a per display basis.  (At least that's the way I would want to use it.).  The tool that I'll include is a masking function, so you could apply a mask on one display (say removing all reflectivity under 15 dBZ) and compare to the other non-modified display.

gamaanderson, 2015-05-05:
Thank you for the explanation @nguy, I will need to reflect on this a little.

pfhein, 2015-05-05:
I would go the other direction.  I would make the onPick (value at a point)  into a Class and add it to the toolsBox.  (Actually I have done this on my local copy.)  My reasoning is two-fold.  First I feel that we should keep the core program as simple as we can.  The core program being just display the data.  That would help to keep things modular.  Second, I am trying to add a "select a region" class and I needed to turn off the onPick to free up the mouse for my "select a region" code.  And thinking off in the future a ways, I would put the toolsBox Classes in a separate folder to make it easier for others to add their own special Classes to fulfill their needs.  That is my thinking.

nguy, 2015-05-05:
I had not thought about the mouse.  That's a good point.
I am still having trouble with the arrow keys not responding after certain operations.
Maybe a short term solution would be to create a tools.py file that holds the tools and then we can move this over into a separate folder once everything is working properly?

pfhein, 2015-05-05:
@nguy - I am good with a tools.py file.  That would be a good separation.

gamaanderson, 2015-05-06:
I wouldn't go so fast in separating a file, this can deceiving, having separated files but interdependent code. I would suggest new tools to be added as naive as possible in separated branches, so we can analyze how it works and how it could interact with the rest of the program. @pfhein if you already have that "select a region" in some where, I would be interested in having a look. 

There is no reason why arrow keys couldn't work with zoom/pan, is just a questions of correcting focus. Several widgets, as buttons, handle arrow key (this is poorly described in the Qt documentation) and as tilt are handle in the window layer, anything can get in the way. But as I said, this is just a question of finding what is the problem and fix it.

About the filter, I have talked in issue https://github.com/ARM-DOE/pyart/issues/244 about pseudo-variables. That was how I implement filters in my personal project and the discussion in the issue indicates how this can be done in pyart. But if we were to implement this way, we would need a new class (as tilt window but more complex) to govern it, adding and removing pseudo-variables as to avoid misuse.  Eventually this class could use the main window to display it self, avoiding a new floating window.

nguy, 2015-05-06:
I implemented a different masking procedure in AWOT - another package I've been working on - but pretty much extended the existing PyArt to be more comprehensive.  I'll take another look at that issue thread.  I scanned it, but had not read it closely.

What I'll do the tools.py is create the file, but not link it.  That way we can see if it looks like something that might work.  I'll push another commit here shortly with a few fixes.

nguy, 2015-05-06:
I decided to wait until the next push to include tools.py.

pfhein, 2015-05-06:
@gamaanderson - Here is my broken code with all its warts.  There are 2 versions there with the second version commented out.  Both versions are not working.

    #from matplotlib.widgets import LassoSelector
    from matplotlib.lines import Line2D
    from matplotlib.path import Path
    
    ##########################
    # Select Region of Interest Class Methods #
    ##########################
    class ROI:
        ''' 
        Select a Region of Interest: The code modified from
        https://www.mail-archive.com/matplotlib-users@lists.sourceforge.net/msg00661.html
        '''
    
        def __init__(self, ax, fig, tilt, display):
            self.previous_point = []
            self.start_point = []
            self.end_point = []
            self.line = None
            self.verts = []
            self.ind = []
            self.xys = []
            self.xpts = display.x[:,tilt]
            self.ypts = display.y[:,tilt]
            self.xys = np.array([self.xpts,self.ypts]).transpose()
            self.fig =  fig
            self.fig.canvas.draw()
    
        def motion_notify_callback(self, event):
            if event.inaxes:
                ax = event.inaxes
                x, y = event.xdata, event.ydata
                if event.button == None and self.line != None: # Move line around 
                    self.line.set_data([self.previous_point[0], x],
                                       [self.previous_point[1], y])
                    self.fig.canvas.draw()
                elif event.button == 1: # Free Hand Drawing
                        line = Line2D([self.previous_point[0], x],
                                      [self.previous_point[1], y])
                        ax.add_line(line)
                        self.previous_point = [x, y]
                        self.verts.append([x, y])
                        self.fig.canvas.draw()
    
    
        def button_press_callback(self, event):
            if event.inaxes:
                x, y = event.xdata, event.ydata
                ax = event.inaxes
                if event.button == 1:  # If you press the right button
                        if self.line == None: # if there is no line, create a line
                            self.line = Line2D([x,  x],
                                               [y, y],
                                               marker = 'o')
                            self.start_point = [x,y]
                            self.previous_point =  self.start_point
                            self.verts.append([x, y])
                            ax.add_line(self.line)
                            self.fig.canvas.draw()
                        # add a segment
                        else: # if there is a line, create a segment
                            self.line = Line2D([self.previous_point[0], x],
                                               [self.previous_point[1], y],
                                               marker = 'o')
                            self.previous_point = [x,y]
                            self.verts.append([x, y])
                            event.inaxes.add_line(self.line)
                            self.fig.canvas.draw()
    
                elif event.button == 3 and self.line != None: # close the loop
                            self.line.set_data([self.previous_point[0],
    self.start_point[0]],
                                               [self.previous_point[1],
    self.start_point[1]])
                            ax.add_line(self.line)
                            self.fig.canvas.draw()
                            self.line = None
                            path = Path(self.verts)
                            self.ind = np.nonzero([path.contains_point(xy) for xy in self.xys])[0]
                            print self.ind
                            print "Closed Loop"
                            print self.verts
    
        def connect(self):
            self.motionID = self.fig.canvas.mpl_connect('motion_notify_event', self.motion_notify_callback)
            self.buttonID = self.fig.canvas.mpl_connect('button_press_event', self.button_press_callback)
    
        def disconnect(self):
            self.fig.canvas.mpl_disconnect(self.motionID)
            self.fig.canvas.mpl_disconnect(self.buttonID)
    
    
    #class RegionSelect:
    #    '''
    #    Class for Selecting a region of points
    #    '''
    #    def __init__(self, ax, radar, tilt, display):
    #        self.xpts = display.x[:,tilt]
    #        self.ypts = display.y[:,tilt]
    #        self.ax = ax
    #        self.tilt = tilt
    #        self.radar = radar
    #        #Need to have a 2D array of r and 2D array of theta for each point
    #        #arr1 = np.zeros(shape=(len(self.radar.range['data']),self.radar.sweep_end_ray_index['data'][self.tilt]-self.radar.sweep_start_ray_index['data'][self.tilt]+1,1))+1
    #        #arr1 = np.ones(shape=(self.radar.sweep_end_ray_index['data'][self.tilt]-self.radar.sweep_start_ray_index['data'][self.tilt]+1,1))
    #        #rng_pts = arr1 * self.radar.range['data']
    #        #azarr = (np.zeros(shape=(1,self.radar.sweep_end_ray_index['data'][self.tilt]-self.radar.sweep_start_ray_index['data'][self.tilt]+1))+1) * self.radar.azimuth['data'][self.radar.sweep_start_ray_index['data'][self.tilt]:self.radar.sweep_end_ray_index['data'][self.tilt]+1]
    #        #az_pts = arr1 * azarr
    #        #xpts = rng_pts * np.sin(az_pts)
    #        #ypts = rng_pts * np.cos(az_pts)
    #
    ##  need matplotlib.collections.PathCollection object
    #        self.lasso = LassoSelector(ax, onselect=self.onselect)
    #    
    #    def onselect(verts):
    #        print verts

gamaanderson, 2015-05-06:
Thank you @pfhein.
Is it possible that the only thing you need directly from Display is self.fig, if that is the case, it could really work independent of Display. The only question is if fig need be a V variable, I have the impression it does not. If you need a graphical place, it could be in the main menu, but we still have to make it receptive for external tools. 
As you are connecting with fig, and onPick is connecting with canvas, they may conflict. But if we get them to connect to the same instance they could work together (no need to deactivate one to use the other), lets see how this turns out.

nguy, 2015-05-13:
@pfhein and @gamaanderson  I put the above code into the tools.py file and got it working. See pull request #24 

One thing about this method is that you select the area with a left click and close the region with a right click.  Not sure yet how to get this info to user.  

The next step would be to figure out what to do with this region.  I'll probably open another issue for this.

closed 2015-05-13 by nguy