nguy/artview #16: the limits.py file

gamaanderson, 2015-05-05:
I have two problems here.

1. One of the must important things in using OOP (object oriented) is to know when NOT to use objects. I believe `DisplayLimits` is a example of that, unless this is to grow in the future, a function would work just as well, with the advanced that `Display` is less dependent of how this is organized, as long as it keep returning valid values. I also wouldn't like to see this grow larger that it is today, this kind of tasks of selecting appropriated values, although important, tend to grow into big ugly parts.
 
2. The class `Ui_LimsDialog` uses a Qdialog, that is, it blocks the program until it is closed. I don't know if this was the intention, but if so, there is not reason to make it that complicated, with V variables etc. A simple function could work just as fine, just as selecting new title/unit does, and let `Display` update Vlims. If that was not the intention, then I will wait you finish with the implementation before making further comments.

nguy, 2015-05-05:
1. I fully agree with this.  Originally I was thinking that ```python DisplayLimits``` would be encompass more, but as you point out it can get ugly and keeping it as the simple update function I think is the best way to go.  
I just changed this and it will be in the next push.

2. The original was to block the program until finished selecting new limits.  But I'm not sure this is the best way.  The idea was to keep the limits the same as title/unis, but it seemed that it required a great deal more because I wanted all limits to be able to be changed at the same time.

gamaanderson, 2015-05-05:
2. It is a little more complicated than title/units, but I'm confident it can be done as a simple function call. I also agree with blocking the program, because otherwise plot will be updated at every time a value changes.

nguy, 2015-05-06:
I've modified the Ui_DisplayLimits to be a function.  You were right @gamaanderson it's simpler and more effective I think.

gamaanderson, 2015-05-06:
Lets keep Vlims obsolete by now, however I have a feeling it may come back.

This issue is almost closed, but I would still want to concentrate calls of `limits.initialize_limits(...)` in `plot.py` into a single function and make a local import to decrease file interdependence. If you want I can tackle that in a small PR.

nguy, 2015-05-07:
Sounds good.

gamaanderson, 2015-05-11:
closed by PR #21

closed 2015-05-11 by gamaanderson