nguy/artview #17: Class Initialising Variables

gamaanderson, 2015-05-05:
I open this to discuss if and when class should initialize V variables. I usually don't like this, as it is giving class to much power, and therefore multiple classes can conflict with each other over this. 

This is happening in two places now: `Menu` initialize `Vradar` and `Display` initialize `Vlims`. The first one, I did, because the main function of `Menu` is to govern over `Vradar`, so it is not that bad that it open it. Nevertheless I believe this should be done some what differently, not sure exactly how.

gamaanderson, 2015-05-05:
This is a small issue, so I would wait the new style to settle before handling it.

nguy, 2015-05-05:
Sounds good.  Let's see how things fall out.  But my initial (ignorant) feeling is that if this is controlled in the execute functions, it may not be detrimental?

gamaanderson, 2015-05-05:
This is not a ignorant feeling at all, it is a pretty real concern, that is why it need some thought before doing.

gamaanderson, 2015-05-11:
Before closings this issue, I would like to make a question. @nguy has suggested a graphical interface for starting a component, in especial a `Display`. I believe this could work fine as a static method inside every component. There is just one complication, the shared variables (V Variables). There are other solution for that, but one would be have all shared variables as optional arguments.
However I don't like, AT ALL, of `Display` initializing `Vradar`. Are you as opposite to that as I am? As I said the are other options to that, for instance a more complex starting interface or make Display work (that is: not crash!) with `Vradar.value == None` and count on issue #14 to connect to a valid radar.

nguy, 2015-05-12:
I have actually changed my mind about a GUI for starting a component (for now).  I don't see the benefit in this.  I believe we should just have a default startup mode (same or similar to what we have now) and let the user add instances (another Display window for example) via the main menu.  

I don't think that Display should initialize Vradar either.  Right now the Vradar instance is started in ```python menu.py``` and I do like this structure.  If there was a main abstract class that contained the whole app, this would allow the Vradar instance to be passed easily (I believe) to a new Display.  In addition, we would like to be able to ultimately edit the data (i.e. via selectable regions and automated functions) and this would likely make that coding a bit more straightforward.

One question I would have from the above is: Do we need to make Display not crash when ```python Vradar.value == None```? We could put a check in that passes when this is the case?

gamaanderson, 2015-05-12:
I'm not quite sure about the implementation, but it would be kind of interesting let the Display blank while waiting for a radar. I think it wouldn't be difficult to get that.

nguy, 2015-05-12:
That might be an interesting approach.
FYI, just pushed some changes.

gamaanderson, 2015-05-12:
So we have made two decision:
1. Components may initialize a shared variable as long as it is an optional variable and it was not given by the user. If the user gives a variable it must have a valid value, in some cases `.value=None` may be a valid value, but not always. Components docstring shall inform that. 
2. Components are allowed to have mandatory variable, for instance `Vradar` is mandatory for `Display`

That being said, I'm closing the issue.

closed 2015-05-12 by gamaanderson