nguy/artview #18: Revision

gamaanderson, 2015-05-05:
This is just a reminder to handle in the future.

Several parts of the new code were copied from the old one, So I believe we should revise it to make sure it has no bugs hidden, or if it could be done better. I'm specially concerned with the _update_plot() and how the zoom/pan is working.

nguy, 2015-05-05:
Great idea!  ```python _update_plot``` is a core piece and it is quite possilbe it is not optimized or could be done in a better way. 

Just a comment on Zoom/Pan, I have changed the ToolSelect to no longer be used and have instead added a ```python self.zp``` that will allow a more straighforward way of keeping track here.  It should be noted that when Zoom/Pan is enabled, the arrow keys are blocked.

nguy, 2015-05-13:
And just to update this thread, The tool setup has changed to now be put into a list.  This was done to hopefully keep track of instances as the functionality grows and makes the checks easier.

@gamaanderson Do you feel better about _update_plot now with the change to V variables (adding strong and weak)?

gamaanderson, 2015-05-13:
It is an improvement, but I will not fell better till I sit down and understand every line of that code, which I have not done jet. From what I have seen there is a gap between `self.display` and `self.canvas`, I

nguy, 2015-05-13:
So on line 575 we're using the PyArt library to create figure in matplotlib that is associated to the ```python self.ax```.  This Axis instance was established in the ```python _set_fig_ax``` function, with fig and ax being associated.  The ```python _set_figure_canvas``` function uses the matplotlib backend that connects the matplotlib figure instance and Qt.  So once the figure has been made, L614 is used to actually update the Qt canvas with the matplotlib figure.
Does that make sense?

gamaanderson, 2015-05-13:
Actually it does, I will reflect over that.

closed 2015-06-23 by nguy