nguy/artview #19: Conflicting changeValue

gamaanderson, 2015-05-06:
I have realized that it can happen that several V variables are changed at the same time, e.q. Vradar and Vfield because the new radar doesn't have the previous field. I'm not fully sure what would happen then, but I believe _update_plot() would be called twice, this would be wasted time to say the least.

I will present my thoughts bellow.

gamaanderson, 2015-05-06:
At first I thought to hide the alteration of field calling

    self.Vfield.value = new_field

But that is really wrong, since other classes may be holding just the Vfield but not the Vradar, they how then not be informed of the change. I decided then to make two kinds of value change: a soft and a strong one and is responsibility of the Display to react to the right one. There are two ways of doing that:

1. Variable have two signal "ValueChanged" and "StrongValueChanged", normally both are emitted but in the case of a soft update just the first is emitted. Display on the other hand would just react to "StrongValueChange"

2. There is still just one signal, but V variable send and extra argument `strong=True/False`. Then display would do the following:

        def NewField(self, variable, value, strong):
            # do some thing 
            # ...
            if strong:
               self._update_plot()

Comments?

nguy, 2015-05-06:
I agree that calling ```python _update_plot()``` more than once is wasted time and/or resources.  I admit that I haven't checked the signals coming back into the Display instance.  I'm not sure I understand the difference between strong and not strong.

I just went back and looked at the code again.   In cases like say the ```python tilt.py```, is it necessary to have Vradar called?  If you comment that section out, there is seemingly no affect on the Display.  Can this be safely removed then?

I assumed that the .change takes the change and updates the Variable instance in the Display.  Is this the case?  And if so, why not drop the Vradar altogether?

Finally, are we talking about two different things here?

gamaanderson, 2015-05-06:
Yes, we are probably talking about different things. So I will not comment your questions, since I'm not sure if I understood.
I'm talking about, suppose for safety `menu.py` around line 313 is updated to the following

    radar = pyart.io.read(self.filename)
    if self.Vfield.value not in radar.fields.keys():
        self.Vfield.change('reflectivity')
    self.Vradar.change(radar)

this is just an example, menu don't have Vfield and this check should be done by Display, I believe. But any way, in this case, plot is update twice. I would like to do the following

    radar = pyart.io.read(self.filename)
    if self.Vfield.value not in radar.fields.keys():
        self.Vfield.change('reflectivity',strong=False)
    self.Vradar.change(radar,strong=True)

in this case Display can choose not to update the plot when Vfield is changed.

gamaanderson, 2015-05-11:
closed by PR #22

closed 2015-05-11 by gamaanderson