nguy/artview #25: Enhance Region of Interest

nguy, 2015-05-13:
A Region of Interest class was introduced in Pull request #24 that allows the user to select a region. The next step is use this defined region to perform operations. I am thinking the following steps could be done:

1. Save the region in a way that allow operation inside or outside.
2. Open a Dialog that allows the user to choose the operation.  This should probably be a call within the ROI class.
3. The modified field should be able to be saved into the Vfield variable instance. I was thinking that core.py could be changed to include a .add function.
4. The user then could save the new field to the file if possible.

Let me know what you think of these points. Do they seem reasonable?

Also, this is tangentially related to Issue #9 and I wondered if the rubberband idea has been outdated by the current zoom function or if it still has application.

gamaanderson, 2015-05-13:
Right now there is no need for Issue #9, but it is not outdate, selecting a area to zoom would still be interesting. Most image manipulation programs have both kind of zoom.
I can see you implemented this functionality as a tool, I wounder if this could be done with a component. This is a hard task, since we have not develop components that affect the figure, but eventually we will need this kind of function. 
 
About number 3, if we are making this kind of alteration in the radar/field, it should be done in a component of its own. That doesn't mean it will be one more floating window, it can use of the main menu area to show itself, issue #14 will already allow to do that.

nguy, 2015-05-13:
What is the difference between a tool and a component?  Is a component the same idea as a plugin?

nguy, 2015-05-13:
Oh one more note here, the ROI class will need a little work out the kinks.  I literally copy and pasted @pfhein code and made quick mods to get working.

gamaanderson, 2015-05-13:
Yep, that I the nomenclature I'm using, pretty much a component is a plugin. But better are example:

Menu, Display, TiltButtonWindow and FieldButtonWindow are all components and are added in the **execute**. Even when Display lanch TiltButtonWindow it is doing the same way it could be done in the **execute**.

Everything in *tools.py* and *common.py* are tools. Display has a specif function for every tool it is using.

nguy, 2015-05-13:
Okay, thanks.  We (@pfhein and @tjlang ) had previously had a conversation about using plugins for all of the tools.  
So the crux is that the tools as they are right now are not able to launch on their own and would need to instantiate some variable instances, etc. to become components, correct?  This could seemingly be done.  But I'm wondering what you gain when the tool can be called, but there is nothing to operate on?

One other note, is that once we decide how to tackle this, I'd like to think about a reorganization of the code to create a tools folder.  Then each of the tools could be pulled out into it's own script (and therefor component) to keep the tools separated.  This is basically just cleaner for me to think about.  This would require that the setup.py be written and it becomes a package then.  
A possible example of something to follow would be [SharpPy](https://github.com/nguy/SHARPpy).

gamaanderson, 2015-05-13:
I believe both tools and components have a use, some times a tool is better, other times a component is better. I have thought about it, but decided ZoomPan is good as a tool. The reason I like to see ROI as a component is that it will be used outside Display, so someone is going to need to get this information. If ROI is a tool Display would had to provide the information and the other part would have to ask for it, if its a component the information is already there in a V Variable for instance.

As for the organization I understand you point of view and it is an important one. I however have a different point of view, for me *plot.py*, *tools.py* and *limits.py* are all one file that we separated so its not that big. **One can not work properly without the other**. If I were to separate in folders I would create a folder for every component (core, menu, display, field and tilt).

nguy, 2015-05-13:
I agree with that line of thought.  Thinking of these types as fundamentally different is helpful in designing the code.  

I actually do not think that any of the other components (core, menu, display, field, tilt) require individual folders.  In my mind the tools are unique here.  This is the component where I see others being able to easily add functionality.  But it is a bit too vague maybe.  I would propose that there are two types of tools (following from the tools and components discussion).  The first, like ZoomPan is a display tools and is inherently dependent upon the display.  The second is a data tools, which would be where masking, and application of internal PyArt algorithms could be connected.  This second set is where I would propose the sole folder to be, with each tool being it's own script and therefore easier to track and understand.  Though I think connecting PyArt algorithms could be done in one single file as they are related in their location.

gamaanderson, 2015-06-25:
I am/was increasingly worried about this ROI, it is growing too big as a tool (actually I don't think we need tools at all, but this an other issue).
I would like to see ROI as component. I see PR #36 moving in the right direction, but it don't go far enough. It is not a sub-class of core.Component, plot still have a specific reference to it. And some other points around this.

I know it looks like I'm implicating with small issues, but I do think its important. Just see how the plugins folder is working, delete a file and it will be as if the plug-in was never there.

I would like to hear some feed back to this, I'm considering getting my hands to work, but this way create some behaviors in ROI that one may not like.

nguy, 2015-06-25:
So I am thinking PR #36 is an incremental step. Tying up a few items, getting them working before moving on adding some features. 

I flirted with some of what you mention as I was implementing ROI. The implementation of the selection ROI is to me a fundamental tool. However, its utility is in applying that selection for many potential applications. Just a few of which I added (save, open, table). I wanted to make sure this was feasible before going larger. That being said I agree that ROI is probably better as a plugin.

On a separate but related issue, components vs. plugins vs. tools is getting confusing as I think we are each using them slightly differently to describe things. I think @gamaanderson your use of Components is equivalent to my use of Plugins. I think we either need to change the 'components' directory name and/or decide how to unify our wording. My use of Tools was merely a decision to use that word in the moment of programming. Having some function that operates on the display and querying the underlying data. I'm happy to change naming if the group would like.
I don't think there is a perfect solution here, but let's decide what we would like to call things. @pfhein @tjlang could you weigh in?

So the ROI is a first step to building a Solo editor replacement in my eyes. I'll call this the ROI editor for now. And this fits nicely into the plugin concept probably. As @gamaanderson points out, the advantage is that if ROI editor is not in the plugin folder its like it doesn't exist. I think you could also say it's there, but unless activated it is also like not existing. It doesn't interfere. But allows ARTView and extension that is very useful to some.

This also hits on another point for me at least. I'd like to make sure we can keep the core stuff slim enough to have a "lightweight" version for just visualizing the data. That was the original intention of ARTView and I think we should preserve that if possible.

But thinking ahead to make ARTView extensible has become a focus as well.

Well I will now step down from the soap box and let someone else take it up. Whoever, does the next PR after #36 is accepted, feel free to move ROI to plugin status and we can hopefully begin to develop this. I need to figure out how to use that interface to see if there might be ways to improve useability. Okay, now I relinquish the soap box. :)

gamaanderson, 2015-06-25:
About the wording:
tools is clear, they are in **tool.py** and get imported in **plot.py**
Components and Plugins are similar, the are both sub-classes of core.Component and work accordingly, the difference is in the level of commitment: A component is a **fundamental** part of ARTview and they are designed to work together, plug-ins are **optional** parts of ARTview.

See that Menu is a component and therefore I write in **artview/__init__.py**: `from .components import Menu`. So ARTview will not import without the Menu; Display can still work if Menu is broken but not if **menu.py** is not there

gamaanderson, 2015-07-11:
1. Now we have two ROI's: one as a component and one as tool. I don't think prudent to keep both as we will have keep both up to date, at the same time the will of maintaining it as a tool has been expressed. So I ask what is the important point here? do you understand how ROI being add to Menu ([in this line](https://github.com/nguy/artview/blob/master/artview/scripts/standard.py#L34))? would you be happy to have something similar to Display, replacing the ROI in **tool.py**?

2. About having a Histogram in ROI: I would prefer to have it into a separated plug-in that connect with ROI (in a shared Variable). However for that we must decide one thing first: now in ROI there are two conflicting data storage: [data](https://github.com/nguy/artview/blob/master/artview/components/roi.py#L151) this is a numpy array and [table](https://github.com/nguy/artview/blob/master/artview/components/roi.py#L195) is a `common.CreateTable`. We must define just one as the main data carrier, both cases involve code change. So where will the histogram get its data: from data and from table?

nguy, 2015-07-12:
I'll comment to  2. first. The data numpy array is the primary data storage. I think the histrogram and any further calculations should be performed on this array. 
The creation of the table needs the data to fill itself, making a QTableItem (if I remember correctly) out of every data point. But this table acts as nothing more than a display really. And that was the intent. When the data array changes then a new table can be drawn.

I agree with your point 1. In my head I had this duplication going away when ROI became a component. We could likely make some of the interior procedures common functions. I was envisioning that ROI in in tools.py would be a limited subset of the entire functionality of ROI the component (and calling the same underlying instance). Ultimately, I would like ROI to become part of the editing portion of ARTView, see #34

gamaanderson, 2015-07-12:
The 2. is answered. The 1. is not clear at all and this may need some more discussion, but for the time being, I'm saying the I can do the code in **roi.py** be assessable in the tools **drop-down menu** and that would allow me to completely remove the code in **tools.py**. As usually I have very little opinion on how the user sees the functionally,  for me it all comes down to the code. (and it is probably good that we are pretty different in this aspect) 

Note to myself: we must improve how components interact, in a lego metaphor we may have all the pieces but to build a running car we must join them in the correct order. Should we rely so heavily in the scripting for this task (as we do now?)

nguy, 2015-07-12:
I would agree that the ROI be accessible in drop-down menu and via roi.py is fine with me. It doesn't need to exist in **tools.py** if it works from **roi.py**.  In fact, roi.py would give more flexibility for future development. 

We could probably port any display function in **tools.py** that uses ROI into a function of **roi.py**. This way it remains the same as it is now AND we could then have ROI the component perform the other tasks (like histogram). We just don't need to call the "full" ROI for the drop-down menu. Does that make sense?

To comment on your note to self, we should make things modular to the best we can, so that scripting isn't necessarily required.

gamaanderson, 2015-07-13:
So I think I didn't explain it right. Of course we can change code from one file to the other, the challenge is that **plot_radar.py** would not have a reference to **roi.py** would not even have to word roi, just like **meny.py**.

About the self note, modularity actually increase the need for scripts, take this [line](https://github.com/nguy/artview/blob/master/artview/scripts/standard.py#L56): **roi.py** needs it while **tools.py** do/did not. Scripts are getting bigger and more complex faster than I've foreseen, control mechanism are also unintuitive and may scare inexperience users.

nguy, 2015-08-06:
Closing this issue as these are discussed elsewhere

closed 2015-08-06 by nguy