nguy/artview #29: Documentation

gamaanderson, 2015-05-19:
Permanent issue on documentation.

gamaanderson, 2015-05-19:
Just as a starting comment, I think pyart documentation is really good, if we could do something similar would be nice

gamaanderson, 2015-05-20:
[Here](https://raw.githubusercontent.com/gamaanderson/artview/pick-changes/DEV_README.txt) is a small list of orientations for developers

nguy, 2015-05-20:
I like PyArt documentation as well, but more importantly keeping a similar structure is probably the right way to go given that ARTView is essentially an extension of PyArt.

gamaanderson, 2015-06-11:
I went ahead and implemented this in PR #33

nguy, 2015-06-13:
@gamaanderson - looking into this now. Been super busy with our field project deployment the last few weeks.

gamaanderson, 2015-06-23:
In README::USAGE
there is no .py in the following:

    python artview.py -d /some/directory/you/want/to/point/to

there is no longer a artview.py file

nguy, 2015-06-23:
Oops missed one, I'll fix that with the next push (hopefully soon).

nguy, 2015-06-25:
Fixed in PR#36.

nguy, 2015-06-25:
I realized I hit the wrong button, we can keep this open as we move forward with Documentation.

gamaanderson, 2015-07-17:
@tjlang just a heads up!

I just realized that pep8 complained about `#:` and changed to `# :`, however I do need this to document attributes according with [sphynx autodoc](http://sphinx-doc.org/ext/autodoc.html). There is no need to correct that now, I will make the changes in my fork and this will come back in some PR.

nguy, 2015-08-04:
Just made some updates to the doc-strings throughout the package.

gamaanderson, 2015-08-04:
It would be nice to see if you can rebuild the documentation, but it may be not that easy, I had to fight a lot to get it running.

first you will need to install Sphinx
then run `rebuild_docs.sh`  (in folder **docs/**)
if you do built it successfully, use `git status` to check for new files in **docs/build/html**, you need to add those files to git.

I tested, and got 23 Warning, normally I get 3, those are the following, I understand them and they are ok.

    checking consistency... 
    /home/gama/artview/docs/source/auto_examples/index.rst:: WARNING: document isn't included in any toctree
    /home/gama/artview/docs/source/reference/generated/artview.core.core.Component.rst:: WARNING: document isn't included in any toctree
    /home/gama/artview/docs/source/reference/generated/artview.core.core.Variable.rst:: WARNING: document isn't included in any toctree
    done
the other 20 warning should be addressed, if you have problems with changes not taking place remove all files from folder **docs/source/reference/generated/** (add `rm source/reference/generated/*` to `rebuild_docs.sh` if you want)

In any case, I'm here for help. Also remember I'm running everything in Debian, and I don't know if this will work in MacOSX

gamaanderson, 2015-08-04:
also, before all that, you may need to build/install artview to get the right version number in the documentation (one gets the version from the automatically created **artview/version.py** not **setup.py**).

nguy, 2015-08-04:
I updated the version number in ```setup.py```, if I'm not mistaken this is the only place the update needs to be applied and then the *new* ```version.py ``` is created by build/install.

Okay as for the documentation, I also get the 23 warnings after running ```rebuild_docs.sh```. Now can you help me understand what needs to be changed? Do I need to modify the **.rst** files where the warnings were thrown?

gamaanderson, 2015-08-04:
no, you don't. All those files are automatically generated.

gamaanderson, 2015-08-04:
most of those warning is because `artview.components.Display` changed to `artview.components.RadarDisplay`. You can start going after this in the doc-strings and see how many warnings are left

gamaanderson, 2015-08-04:
the only **.rst** files you need to see, are the tutorials:
**docs/source/script_tutorial.rst**
**docs/source/plugin_tutorial.rst**
those also need actualization, including the code examples

nguy, 2015-08-04:
Okay, I have the warnings down to 15. I pushed the update.
Many still the same as you note above (I fixed multiple instances), but I can't find any instances with a ```grep``` command.

Also, what is ```toctree```?

gamaanderson, 2015-08-04:
`toctree` is how sphinx organize the links, or something like that.

    /home/gama/artview/docs/source/reference/components.rst:14: WARNING: failed to import Display
    /home/gama/artview/docs/source/reference/components.rst:14: WARNING: failed to import TiltButtonWindow
    /home/gama/artview/docs/source/reference/components.rst:14: WARNING: toctree references unknown document u'reference/generated/Display'
    /home/gama/artview/docs/source/reference/components.rst:14: WARNING: toctree references unknown document u'reference/generated/TiltButtonWindow'

in **components/__init__.py** one need to actualize the initial doc string, to remove `Display` and `TiltButtonWindow` and add `RadarDisplay`, `GridDisplay` and `LevelButtonWindow`

nguy, 2015-08-04:
Ah, I see what you mean. Fixed and pushed. Down to 3 warnings.

gamaanderson, 2015-08-04:
nice, those 3 are the "normal" ones.

gamaanderson, 2015-08-10:
I have no real experience in it but I have made a [video](https://www.youtube.com/watch?v=iaNoGZTUhg4), for now it is unlisted, so only we can see it. It has no sound, since I do not trust my voice, but it has some sub-titles. @nguy, @tjlang, @swnesbitt it has your names in the end, so make sure you all agree with it, also please revise the English.

If someone wants to improve it or add sound, the mp4 and the subtitle srt are [here](https://github.com/gamaanderson/scripts). I may also provide the original recordings (12min+) if needed.

nguy, 2015-08-10:
Documentation updated.

I'll take a look at the video today hopefully.

nguy, 2015-12-09:
Want to note the short discussion in Issue #85 so I can close.

gamaanderson, 2017-01-29:
Version 1.3 is getting closer, so we will need to review the docs. Also new tutorial are important. I know my voice is not the best so it would be really nice if someone would do the tutorial. It is certainly a lot of work and would be perfect for some one that want to help but don't want or don't know how to get in the code. I would certainly add such a person as author in the presentation at the AMS conference.

nguy, 2017-02-03:
Good idea. I'll try to do some as well. I'll try to go through the code and update docstrings, etc. before doing a doc build as a final bit before release.