nguy/artview #30: Buried windows

gamaanderson, 2015-05-19:
How to avoid windows being buried unter other ones? what tool pyqt give to define the window position and size?

nguy, 2015-05-20:
Maybe [this](http://pyqt.sourceforge.net/Docs/PyQt4/qwidget.html#move-2) is a start?

gamaanderson, 2015-05-22:
Yes, in combination with [screenGeometry](http://pyqt.sourceforge.net/Docs/PyQt4/qdesktopwidget.html#screenGeometry) it should do the work.
[sizeHint](http://pyqt.sourceforge.net/Docs/PyQt4/qwidget.html#sizeHint) and [adjustSize](http://pyqt.sourceforge.net/Docs/PyQt4/qwidget.html#adjustSize) shall also be useful

A hint is to execute **execute_two.py** in the python terminal without `app.exec_()`, them you can call this functions directly.

P.S. sorry not answering before, really busy here.

gamaanderson, 2015-06-15:
I would suggest to add the following in **execute_two.py** just before `app.exec_()`. When you have the time verify if this work well for you

    desktop_rect = QtGui.QDesktopWidget().screenGeometry()

    height = desktop_rect.height()
    width = desktop_rect.width()

    menu_width = max(MainMenu.menubar.sizeHint().width(), MainMenu.sizeHint().width())
    menu_height = MainMenu.sizeHint().height()

    MainMenu.setGeometry(0,0,menu_width,menu_height)

    plot_size = min(height-60-menu_height, width/2)-50
    plot1.setGeometry(0,height-plot_size,plot_size,plot_size)
    plot2.setGeometry(width/2,height-plot_size,plot_size,plot_size)

nguy, 2015-07-09:
Is this issue closed with the recent updates?

gamaanderson, 2015-07-09:
I'm happy with the way it is, if you are also this may be closed for now

nguy, 2015-07-09:
I'm calling this closed, we could always revisit in the future.

closed 2015-07-09 by nguy