nguy/artview #31: Installable package

gamaanderson, 2015-05-19:
How to make artview installable/importable and how to structure it?

gamaanderson, 2015-05-19:
I was thinking of something like

    artview.
        Menu (class)
        Display (class)
        TiltButtonWindow (class)
        FieldButtonWindow (class)
        ComponentsControl (class)
        plugins. (module) # as it already is in PR #26 
          any plug-in defined in plugins folder
          _plugins (list of plugins classes)
        execute (function) standard execution, like execute_two.py today

nguy, 2015-05-20:
I think the current structure is right way to go, this is what I was getting at in earlier conversations as well.  One change is that I would consider making an installable execution file that can be stored in the local Python path so it becomes a system call. 
So that 
```python artview -d directory -r field```
works out of the box.

There is some cross reference in issue #25 also.
I'd like to hold off on the setup until we stabilize the changes to a degree (i.e. work out any kinks in the latest PR and add a couple of working Plugins).

gamaanderson, 2015-06-11:
I went ahead and implemented this in PR #33

nguy, 2015-06-23:
This was implemented with PR #33.

closed 2015-06-23 by nguy