nguy/artview #32: Grid and map

gamaanderson, 2015-05-20:
@nguy I just opened some new issues to lead us in a new phase of the development. Now, however I'm here as an user, I were in need a the following two functionalities: a display for grid data and execute pyart.map.grid_from_radars. The intent is to calibrate the interpolations option to get the best equilibrium between quality and execution time. 

If we were able to develop this in the next month or so would be great. I know you have other projects and so do I, so handle things as you got time.

nguy, 2015-05-20:
Interesting.  Yeah I'll work on these as time permits.  My next week and half is very busy with aircraft deployment.  When in the field, I may have down time (depending on weather) to work on these issues.

My first instinct would be to revisit if PyArt spits out any kind of message (or attribute) upon IO.  Let me put some thought into this.

gamaanderson, 2015-05-22:
I'm adding here a list o task to close this issue, the first two are the most important, the other are just accessory

- [ ] Add plot_grid.py similar to plot.py, with variable Vgrid
- [ ] Add component (plugin) with two variable Vradar, Vgrid. It execute grid_from_radars from Vradar and put the result in Vgrid. It have a form to fill up the interpolation options and a map button that start the mapping.
- [ ] Make Menu work with radars and grids
- [ ] Make TiltButtonWindow work with grid as well
- [ ] Rename plot.py to plot_radar.py or something like that

suggest changes to this list as you wish

nguy, 2015-06-25:
Sorry I'm just getting to this.

I forgot to do this in the latest PR, but I'd like to rename plot.py to plot_radial.py in an upcoming PR.

As for the grid, would it work to add a way to access the ```python grid_from_radar``` to the tools or plugins? Or do you have gridded files athat need read in as well?

gamaanderson, 2015-06-25:
Yes, reading grid files is also important. I don't think we should make Display work with grid, but rather have a GridDisplay.

nguy, 2015-06-25:
Okay, so you want to do the gridding in an ARTView function?
Versus using a [PyART](http://arm-doe.github.io/pyart-docs-travis/user_reference/generated/pyart.graph.GridMapDisplay.html#pyart.graph.GridMapDisplay) method?

gamaanderson, 2015-06-25:
no, still using PyART. I just don't want the Display class working with grids, it needs a different class in a different file, that probably start as a copy of Display. I already have this in a branch, I will make a pull request so that you can give a look.

nguy, 2015-06-25:
Sounds good.

gamaanderson, 2015-07-23:
Closing this issue may be more difficult than I thought. pyart `GridMapDisplay` is not as complete as `RadarDisplay` and has some issues. Here is a list of issues to address:

- [] `GridMapDisplay` don't make a title
- [] `GridMapDisplay` don't change images
- []  limits are relative to the left-down corner, not to grid origin
- []  there is overall projection issue with grid, see pyart ARM-DOE/pyart#294
- [] `ValueClick` for grid
- [] `getPathInteriorValues` for grid
- [] develop an interface for changing `lat_lines` and on_lines`

nguy, 2015-07-24:
Let's leave the issue open and see how the PyArt issue progresses. It will be a little bit before I can turn my attention to this. But per the list above, is there a reason that ValueClick etc. functions are more challenging on a grid? Would a nearest neighbor approach to choosing a point work?

gamaanderson, 2015-07-24:
`ValueClick` and `getPathInteriorValues` ISN'T any more difficult for grid than radar, is just a question of converting the existing one

tjlang, 2015-07-31:
I get this error running the map_to_grid script in ARTview:

{'roi_func': 'dist_beam', 'grid_origin': (-12.24916233, 131.04444878), 'radars': (<pyart.core.radar.Radar object at 0x110b37e50>,), 'bsp': 1.0, 'gridding_algo': 'map_gates_to_grid', 'grid_limits': ((2000.0, 3000.0), (-250000.0, 250000.0), (-250000.0, 250000.0)), 'grid_origin_alt': 50.0, 'weighting_function': 'Barnes', 'map_roi': True, 'refl_filter_flag': True, 'max_refl': 100.0, 'fields': ['AD', 'KD', 'ZD', 'AH', 'DC', 'ZC', 'VR', 'DZ', 'FL', 'SR', 'PF', 'PH', 'RH', 'HD'], 'h_factor': 1.0, 'grid_shape': (1, 500, 500), 'refl_field': 'DZ', 'min_radius': 500.0, 'toa': 17000.0, 'nb': 1.5}
mapping ..
2015-07-31 11:03:43.246 python[70686:507] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
Warning Discarded!
Traceback (most recent call last):
  File "/Users/tjlang/Documents/Python/artview/artview/components/plot_grid.py", line 421, in NewRadar
    self._update_plot()
  File "/Users/tjlang/Documents/Python/artview/artview/components/plot_grid.py", line 596, in _update_plot
    vmin=self.limits['vmin'], vmax=self.limits['vmax'],
KeyError: 'vmin'
Mapping took 1.071725s
Traceback (most recent call last):
  File "/Users/tjlang/anaconda/lib/python2.7/site-packages/matplotlib/backends/backend_qt5agg.py", line 91, in paintEvent
    stringBuffer = self.renderer._renderer.tostring_bgra()
AttributeError: 'FigureCanvasQTAgg' object has no attribute 'renderer'

Not sure how mature the grid mapping is, so no big deal but I just wanted to document the error.

gamaanderson, 2015-07-31:
plot_grid in master is still experimental and its out-of-date, I'm still working in a stable version. See PR #60 for a fast correction or [gamaanderson/grid](https://github.com/gamaanderson/artview/tree/grid) for my working branch.

nguy, 2015-07-31:
@tjlang I just pulled in @gamaanderson changes in PR #58 . See if this solves the problem

tjlang, 2015-08-03:
It worked! OK, so the scripting (-s)  aspect seems reasonably intuitive. Looks like you still have some work to do on bringing in the ability to do titles, save images, etc., with the grid display. But kudos so far. Solo never did this kind of cool stuff!

[image]

gamaanderson, 2015-08-10:
Pyart has merged all that I needed to complete this issue, and I shall complete my part soon. However I don't know if we should merge it, as it need the newest version of pyart, and not pyart 1.4 as we have now.

nguy, 2015-08-10:
That works. We can update and require the most recent pyart. We could do this with an updated subversion I think in a short while.

closed 2015-08-11 by nguy