nguy/artview #35: RHI and Airborne plotting not working

nguy, 2015-06-23:
During the reorganization process the ability to display RHI and/or Airborne data has been lost.

Need to track down why this isn't working. The program tries to pull something up and opens a window, which leads me to believe that it may be something we are doing with the Variables, etc?

gamaanderson, 2015-06-23:
This is probably broken for some time now, I would try to found out the last commit it worked

gamaanderson, 2015-06-23:
the new artview have never worked with rhi, because `_check_file_type` is not working properly. But the interesting part is that in the old `artview.py` this was also broken, but it was not being executed when it should, and that is why it works!!!.

any way correcting `_check_file_type` should solve the problem, I just don't have files to test it.

nguy, 2015-06-23:
Really! Okay I'll look into that as well then.
On a related issue, I wonder if it would be worth creating a shared folder (here or somewhere else) with a few example files so that functionality could be tested in the future.

nguy, 2015-06-25:
See PR #36
It works so, I"m closing this issue

closed 2015-06-25 by nguy