nguy/artview #43: Pyart Config

gamaanderson, 2015-06-29:
We are currently dealing with some ugly hacks, for instance [this one](https://github.com/nguy/artview/blob/5c230a218cbe3439d755086bfb38b2f1725e2006/artview/components/plot_radar.py#L580-L607), that I believe the correct way of dealing would be using pyart configuration tools with the follow principle: "artview should always work with a unconfigured pyart, but a more smooth user experience does presuppose pyart is correctly configured"

However, for that to work we need pyart to give more focus to this possibility, this could be as simple as mentioning the PYART_CONFIG flag in README. What do you think of it @jjhelmus?

Related to this, we should also configure standard values for the scale (vmin and vmax) and colormap for every (pyart-)field. We are currently using values that are only useful in reflectivity-like fields, I have dificulties even for visualizing radial velocity. Couldn't pyart.graph also benefit from those standards?

jjhelmus, 2015-07-09:
@gamaanderson I do not think there is a elegant solution to the multiple field name issue, but I would be open to examining any that you have in mine.  The variety of moment names and conventions has been a pain point in Py-ART that the configuration logic and PYART_CONFIG environment variable attempts to address.  One possibility would be to have ARTView use the `pyart.config.get_field_name` to retrieve the field names which are set in the default or user defined configuration file.  This is what Py-ART uses to generate field names within correction and other functions if they are not explicitly specified.  This would not solve the issue at hand with the multiple names but would at least support users who customization their configuration in Py-ART.

+1 on adding a note about PYART_CONFIG to the README, I'll create a Py-ART issue for this.  

Also +1 on working out a method for standardizing the scale and colormap which pyarty.graph and ARTView can use.  Having these customizable in the same manner as the field names would make sense to me.  There will be some challenges in standardizing these as "good" values often depend significantly on the radar hardware and settings.  For example radial velocities should have limits of +/- the Nyquist velocity which changes depending on the radar and in some cases even the sweep.  Codifying this into a function may be challenging.  I'll make an issue in Py-ART.

nguy, 2015-07-09:
@gamaanderson nearly all fields do have a vmin and vmax associated with them in ```components/limits.py``` in the ```_default_limits```. These should be pretty standard, because once the reflectivity field is properly read in (which is why the hack exists) all other fields are mapped accordingly (as far as I've seen - maybe this isn't true?).

The limits I chose were completely arbitrary to what I was working on at the time and don't even work for some of my own data. Right now the display vmin/vmax can be changes from the Display Options drop-down menu. I could envision that we could save a "User Preference" file that included this along with other values. And then this could be called at startup as well.

gamaanderson, 2015-07-09:
@jjhelmus I think you actually got the idea right. As for now we are trying to cover all possibilities, I would say to give up on that and just use `pyart.config.get_field_name`, if that don't work the user must select manually (it is not that difficult just 2 clicks). My point here is that although there is no world standard most organization have a standard of its own, so a custom `config` should work in most cases.

@nguy I'm not getting those limits at all, the names in my netcdf files are just different, so we should implement the use of Py-ART config. Done that I will test those limits and mount something we can offer to Py-ART, the nice thing here is that ARTview is just perfect to this kind of testing.
I don't like the idea of saving preferences, some time they are need but less try to avoid as long as we can, more interesting would be a plug-in that make the pyart config file

nguy, 2015-07-10:
@gamaanderson I didn't realize there were any problems with the fields, my bad!

I think it's a good idea to try the get_field_name and Py-ART config. It's probably not bad to put in a check for this.

gamaanderson, 2015-08-12:
with PR #78 merge this is closed.

closed 2015-08-12 by gamaanderson