nguy/artview #44: Release

gamaanderson, 2015-06-29:
With the new upgrades in PR #42 I find artview, at least from the functionalities point of view, is starting to get mature and I can already see in a near future making a release changing from 0.x to 1.0. But we are not quite there, documentation for instance is still an issue. 

So I would like to start making a list of things we must solve before taking this step (github milestones?). For me is also important hearing @jjhelmus some criticism on this.

I believe we could use this opportunity for making a little bit of publicity, may be making a small video to show how it work (for me it is really difficult to understand a graphical software with just pictures and text).

jjhelmus, 2015-07-09:
I do not have much of an opinion of what should constitute 1.0 release, that decision is unique to each developer and I do not want to influence the decision either way.  

As for using GitHub milestones to track progress toward such a release I know a number of open source scientific python program use these but I've only tried this once or twice in my own repositories.  A good easy to follow example of this is [scikits-image](https://github.com/scikit-image/scikit-image/milestones).

nguy, 2015-07-09:
@gamaanderson I'm not sure how much I'd use Milestones, but if you want to try it I'm game. 
We could use this thread to make a check list like you did elsewhere for what we consider what is needed for a v1.0 release.

gamaanderson, 2015-07-09:
So Here is my list, you should also add a comment with yours (even if it is empty now)

- [ ] Finish issue #32 
- [ ] Implement support for `pyart.config`
- [ ] Fully implement of remove VgateFilter from plug-ins
- [ ] Re-implement or remove Vlims
- [ ] solve issue #47 
- [ ] make None valid value for Vradar
- [ ]  avoid crashes, ARTview shall "never" crash because of user miss-use but warn it (now there are some known crashes)
- [ ] Decide what to do with Roi: is it already usefully or should it stay in test version? component, tool or both?
- [ ] As aways: documentation

A would have some other  Enhancement ideas, but this already a lot.

nguy, 2015-07-11:
This is my list:

- [ ] Move ROI to plugin, however keep basic ability as tool
- [ ] Add histogram to ROI plugin
- [ ] Create masking plugin for data editing (whole file or region subset)
- [ ] Replicate Solo bulk edit for a directory of files
- [ ] Allow user to save modified NetCDF via Pyart interface ```pyart.io.write_cfradial``` or ```pyart.io.write_grid```
- [ ] Enhance the airborne, RHI capabilities (intentially vague)
- [ ] Display multiple radar files via ```pyart.map.grid_from_radars```
- [ ] Explore a OpenGL plotting library (Vispy, other) to speed plotting
- [ ] Thing about whether to provide a web interface (e.g. bokeh)
- [ ] Documentation, documentation, documentation
- [ ] Switch to PyArt colormaps instead of matplotlib
- [ ] Add a list of PyArt plugins (how to standardize this???)

Okay that's extensive...

gamaanderson, 2015-07-11:
I understand what you mean by items 1, 2, 5, 7, 10, and 11 and have some idea on how to implement. The other ones I may need some explanation.

Do you think all those points are important for a first release? or some of them could come in 1.1 for instance?

nguy, 2015-07-12:
All of these are not necessarily important for a release, but I wanted to get them down. I think we can make the judgement after some have been implemented what can be "kicked down the road" to v1.1, and actually the ones that you point out about having ideas of implementation are in fact what I would got with for v1.0.  

The others I wanted to record digitally for record (my memory is lacking sometimes). :)

For the record, I'm a proponent for versions with major, minor, sub (i.e. 1.0.0). I think Major releases then have more meaning. And then the quick fixes can be added to the sub number upgrades that don't provide substantive enhances or fix a minor bug.

nguy, 2015-07-12:
Another point I should make for a future release, is I would like some editing capabilities on the data, not just visualization. Replacing values, selecting a region to delete, etc. This directly ties to 3 and 4 in my list above.

gamaanderson, 2015-07-12:
Ok, I agree with everything. Just point out some plug-ins already edit the data adding corrected fields, overwriting old fields is also possible.

nguy, 2015-07-13:
Just switched to PyArt colormaps. Also included PEP8 update from Tim.

nguy, 2015-08-04:
@gamaanderson @tjlang @pfhein

I am thinking of going ahead with a release (v1.0) even though we have not met all of the above. I feel like we have a pretty solid file browser with some neat functionality now. 

I started the Zenodo method to assign a DOI for citation purposes. Once we release, we get a DOI.

pfhein, 2015-08-04:
I am okay with a 1.0 release if we are done with all major changes for 
the near future.  I am amazed at how far we have come in such a short 
period of time.  It has become quite usable and a couple of weeks ago I 
set out an email to the CSU Radarmet group recommending it.  Thanks for 
all the hard work.

gamaanderson, 2015-08-04:
I think we can push some things and get all the edges round before the end of the month. I wouldn’t like to get a release as it is now, still need documentation (including the tutorials) to be revised (this is already underway), issue #62 and other minor stuff

gamaanderson, 2015-08-10:
With PR #68 merged, the code is ok for me to make the release. As I don't trust my-self in revising the documentation, that is the only part that I'm not sure, it also need rebuilding.

nguy, 2015-08-10:
I'm looking at #69 (which encompasses #68 as you noted). GOing to make sure everything works on my system. But hopefully I'll merge tomorrow. Looks like some nice work!

closed 2015-09-01 by nguy