nguy/artview #47: Python 2.6 Troubles

pfhein, 2015-07-02:
Python 2.6 does not support the -m option the same way as Python 2.7.  `python -m artview` does not work but `python -m artview.__main__` works with an error.  That is only a documentation change for Python 2.6.

I think it works but the error I believe is a different problem.  The error it produces is the same as when I try to run it in the "installed folder".  This may not be due to Python 2.6 but the error below is the error that I am getting.

`denali:~$ python -m artview.__main__`
`/usr/local64/lib64/python2.6/site-packages/numpy/oldnumeric/__init__.py:11: ModuleDeprecationWarning: The oldnumeric module will be dropped in Numpy 1.9
  warnings.warn(_msg, ModuleDeprecationWarning)`
`./`
`Traceback (most recent call last):`
  `File "/usr/lib64/python2.6/runpy.py", line 122, in _run_module_as_main`
    `"__main__", fname, loader, pkg_name)`
  `File "/usr/lib64/python2.6/runpy.py", line 34, in _run_code`
   ` exec code in run_globals`
  `File "/home/software/usr/local64/lib64/python2.6/site-packages/artview/__main__.py", line 15, in <module>`
    `artview.run(DirIn, filename, field)`
  `File "/usr/local64/lib64/python2.6/site-packages/artview/scripts/standard.py", line 22, in run`
    `MainMenu = Menu(DirIn, filename, name="Menu") #initiate Vradar`
  `File "/usr/local64/lib64/python2.6/site-packages/artview/components/menu.py", line 47, in __init__`
   ` super(Menu, self).__init__(name=name, parent=parent)`
  `File "/usr/local64/lib64/python2.6/site-packages/artview/core/core.py", line 95, in __init__`
    `super(Component, self).__init__(parent=parent, flags=flags)`
`AttributeError: 'flags' is not a Qt property or a signal``

gamaanderson, 2015-07-02:
indeed this is not a python problem, but a PyQt one. It would be nice to know that version you are using. Anyway if you want to execute it is totally safe to remove flags from this [line](https://github.com/nguy/artview/blob/master/artview/core/core.py#L95) and if we can solve the problem I would have no problem in making this removal definitive

as for the -m option, my work around would be using -c, but that unfortunately don't have all the command line options

    python -c "import artview; artview.run()"

pfhein, 2015-07-02:
Here is the PyQt version number, 4.6.2-9.el6, so it is not the most 
recent as it is always with RHEL and its clones.  Do you think that old 
version is the problem?

I am thinking that I got the -m option figured out for Python 2.6, so I 
think that is okay with the slightly different syntax.

gamaanderson, 2015-07-02:
yes, this is some what old, but by looking at the old documentation I have changed my mind, I'm not sure what is the problem: python, pyqt or qt! 
I can't test that of my own and I don't want to make you a guinea pig, but try replacing

    flags=flags
by

    flags=QtCore.Qt.WindowFlags(flags)

pfhein, 2015-07-02:
Thanks for the idea.  Unfortunately it did not change a thing.
     super(Component, self).__init__(parent=parent, 
flags=QtCore.Qt.WindowFlags(flags))
AttributeError: 'flags' is not a Qt property or a signal

gamaanderson, 2015-07-03:
So I really don't know and tracking this down from python to sip into C++ is not in my possibilities. Just ignore the flags they don't have any real use right now

nguy, 2015-07-03:
What does the flags keyword do?

nguy, 2015-07-03:
I've been seeing some Warnings thrown out by Numpy recently. It appears they eventually may deprecate some items and are getting people ready for that. One in particular I see a lot is about an integer. I'll have to actually look at that one of these days, maybe be proactive before it breaks something. :)

nguy, 2015-07-09:
@pfhein do you have a solution then that works to call ARTView?

pfhein, 2015-07-09:
For 2.6 `python -m artview.__main__` I believe is working.  Unfortunately working means giving an error, the same error I get when running from the "installed folder".  See above for the error.  Removing "flags=flags" just moves the error to "parent=parent".  Removing the entire line gives a different error.  That is where I am at in my debugging.  I hope to get back to it today.

gamaanderson, 2015-07-09:
bad, the parent in indeed important, not primordial but important. Removing the hole line will indeed create an error in any case, but `super(Component, self).__init__()` works

nguy, 2015-07-09:
@pfhein did this come to light with the latest updates to ARTView?

pfhein, 2015-07-09:
Yes - the version that I downloaded on July 2 which has major changes 
and thus the python 2.6 problems.

nguy, 2015-07-09:
Okay. Good to know. I put a note in the README.

pfhein, 2015-07-09:
Thanks @gamaanderson that worked but it may of broke something too.  My test file will not work any more. Here is the latest problem.
```
denali:~/tmp/nc$ python -m artview.__main__
/usr/local64/lib64/python2.6/site-packages/numpy/oldnumeric/__init__.py:11: ModuleDeprecationWarning: The oldnumeric module will be dropped in Numpy 1.9
  warnings.warn(_msg, ModuleDeprecationWarning)
./
Opening file /home/hein/tmp/nc/cfrad.20111124_230003.000_to_20111124_230758.000_toga_v2.3_PPI.nc
Warning Discarded!
Traceback (most recent call last):
  File "/usr/lib64/python2.6/runpy.py", line 122, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib64/python2.6/runpy.py", line 34, in _run_code
    exec code in run_globals
  File "/home/software/usr/local64/lib64/python2.6/site-packages/artview/__main__.py", line 15, in <module>
    artview.run(DirIn, filename, field)
  File "/usr/local64/lib64/python2.6/site-packages/artview/scripts/standard.py", line 27, in run
    plot1 = Display(Vradar, Variable(field), Vtilt, name="Display1", parent=MainMenu)
  File "/usr/local64/lib64/python2.6/site-packages/artview/components/plot_radar.py", line 114, in __init__
    self.NewRadar(None, None, True)
  File "/usr/local64/lib64/python2.6/site-packages/artview/components/plot_radar.py", line 352, in NewRadar
    self.rTilts = self.Vradar.value.sweep_number['data'][:] 
AttributeError: 'NoneType' object has no attribute 'sweep_number'
```

gamaanderson, 2015-07-09:
Opening file /home/hein/tmp/nc/cfrad.20111124_230003.000_to_20111124_230758.000_toga_v2.3_PPI.nc
    Warning Discarded!

pyart fail to open the file, the important lines are here https://github.com/nguy/artview/blob/master/artview/components/menu.py#L392-L403. See if you don't have the problem @nguy had with `delay_field_loading`, just remove the option

gamaanderson, 2015-07-09:
this crash (when pyart fails it crashes) is one that have been annoying me for some time, it is a quick fix but I just didn't get to it

pfhein, 2015-07-09:
Yes , @gamaanderson,  removing the option worked.  Zoom/Pan works. 
"Click for Value" works. "Select a Region of Interest" does not work.  
We are making progress but it coming slow.  Here is the latest error.
```
Traceback (most recent call last):
   File 
"/usr/local64/lib64/python2.6/site-packages/artview/components/plot_radar.py", 
line 454, in toolROICmd
     self.ax, self.display, parent=self.parent)
   File 
"/usr/local64/lib64/python2.6/site-packages/artview/components/tools.py", line 
369, in __init__
     self.CreateROIWidget()
   File 
"/usr/local64/lib64/python2.6/site-packages/artview/components/tools.py", line 
482, in CreateROIWidget
     self.ROIbox = QtGui.QGroupBox("Region of Interest Selection", 
parent=self)
AttributeError: 'parent' is not a Qt property or a signal
```

gamaanderson, 2015-07-09:
the same problem, remove parent from artview/components/tools.py", line 482

pfhein, 2015-07-09:
Many thanks for your help, @gamaanderson!  It is working and looking very nice!

gamaanderson, 2015-07-09:
No problem @pfhein, enjoy it and let me know what you think. I will think on ways of making the needed python 2.6 correction without affecting other version, so that you can keep up to date with master.

nguy, 2015-08-05:
@pfhein and @gamaanderson Is this issue closed with the latest updates?

gamaanderson, 2015-08-05:
that is probably for @pfhein to say.

pfhein, 2015-08-05:
@nguy almost there.  I had to make one fix to get it to work in Python 2.6.
In artview/components/roi.py Lines 201-202 need to be
        self.ROIbox = QtGui.QGroupBox("Region of Interest Selection")
        #self.ROIbox = QtGui.QGroupBox("Region of Interest Selection",
        #                              parent=self)

pfhein, 2015-08-05:
One more thing, for Python 2.6 you need to do
python -m artview/__main__
to run it.

pfhein, 2015-08-05:
One more thing, for Python 2.6 you need to do
> python -m artview/__main__
to run it.  (with proper markdown)

gamaanderson, 2015-08-05:
ok, @pfhein. Note that the version without python (just `artview [options]`) should also work in python 2.6

pfhein, 2015-08-05:
I need to preview my work. One more thing, for Python 2.6 you need to do

> python -m artview/\_\_main\_\_

to run it. (with proper markdown)

pfhein, 2015-08-05:
Nice that command `artview` works (also with Python 2.6).

nguy, 2015-08-05:
That's great! I put a note in the README to use the simplified form.

closed 2015-08-10 by nguy