nguy/artview #53: User Check list

gamaanderson, 2015-07-13:
I would like to know how easily can an inexperience user go around artview. For that I've made the following list

- [ ] can you understand and correctly use inline option -d, -f, -D and -s?
- [ ] can you open a radar file?
- [ ] can you change files and sweeps with the arrow keys?
- [ ] can you change sweep/field?
- [ ] can you use zoom and pick-value?
- [ ] can you change axes limits, title, units and colormap?
- [ ] can you add/remove rings to the plot?
- [ ] can you save a plot image?
- [ ] can you use ComponentControl to get the 2 displays showing diferent fields but always the same sweep? if you change the sweep from one display does the other change together?
- [ ] can you clear the Main menu from all widget, letting only the menubar?
- [ ] can you start a new 3th Display?
- [ ] can you find and use the pluging that executes pyart.correction.dealias_unwrap_phase? can you see the corrected velocity in the Display? can you get help on the options?
- [ ] can you start ROI (from the "Components" menu)?
- [ ] can you select a region with ROI?
- [ ] can you view, save and open the ROI table?

I don't know if we even have guinea pigs to try to go this check list, however I wanted to write it down!

nguy, 2015-07-14:
This is good to think about now. Do you view this list as a test of how intuitive ARTView is to use or if works on different systems? Or both?

As for intuitiveness: I'd be very interested to hear about this.
As for system stability:We have a few different systems that have been running the software (mostly for development purposes), but it's far from extensive.
I think a third thing to think about: How easy is it for someone to help development? Add a plugin.

Some of this will be important for the AMS Radar Conference. There is a short course focusing on the different open source radar tools and ARTView is going to be included. If you have any ideas I'd be interested to hear it.

gamaanderson, 2015-07-14:
I was concerned about intuitiveness, I'm pretty confident of stability as we use nothing more than basic PyQt.

I believe anyone knowing something of PyQt wouldn't have a problem with writing a plug-in and the tutorials should help to get started. If I have any ideas I will tell you

P.s I will be without of internet til Thursday

tjlang, 2015-07-14:
Since I haven't done a ton with ARTview for a while, I might be a good candidate for going thru this checklist, but I don't have a timeline as I am traveling the next couple weeks.

nguy, 2015-07-14:
@gamaanderson enjoy your time away.
@tjlang that would be great!

tjlang, 2015-07-31:
Here are my answers to the questions. Kudos to you guys for putting together a tool that is pretty intuitive in terms of basic functionality. For just basic displaying and inquiry of data, this program runs laps around solo. It is faster, more stable, and much easier to change color maps, scales, zooms, etc. The intuitiveness breaks down for the advanced stuff (see below), but you can work on that.

1. can you understand and correctly use inline option -d, -f, -D and -s?

Mostly yes. I got the error that I mentioned in a recent comment with the -s command with map_to_grid, though it did initially bring up a menu to configure the gridding. The -D option is not documented and the program complains if you use it. Would be cool if it could install in such a way to just run the program by typing artview.

2. can you open a radar file?

Yes.

3. can you change files and sweeps with the arrow keys?

Yes. However, changing files with one key direction and fields with another is not intuitive. I could see a user getting confused by especially the ability to so quickly and unintentionally switch the file they were working on. The tilt switch keys are straight out of solo so they are more intuitive, except for the fact that in solo left/right keys switch tilt, not up/down.

4. can you change sweep/field?

Yes.

5. can you use zoom and pick-value?

Yes.

6. can you change axes limits, title, units and colormap?

Yes.

7. can you add/remove rings to the plot?

Yes.

8. can you save a plot image?

Regular save works. Quick Save does not appear to work. I can't find the image, even with locate on the command line. Should state clearly the path to the Quick Save image.

9. can you use ComponentControl to get the 2 displays showing different fields but always the same sweep? if you change the sweep from one display does the other change together?

Yes, but this was not intuitive and required some trial and error. The ComponentControl portion of the program is very confusing. First up, the term "ComponentsControl" (CC) makes no intuitive sense to an end user. Then, when I select the CC menu item, it brings up a cryptic widget, with an unexplained "IndependentWindow" option selected. Hitting Start gets me yet another Menu, but now with multiple drop-down menus. There are misspellings (e.g., "conect"), poor choice of words (I would recommend "link" not "connect"), and words for things that are not clear. What is vRadar or vGrid, etc.? I got things to sync up mostly by accident by playing with the drop-down menus until more radio button options appeared. Even then, there were more shorthand terms with Vs in front of them. Just write out clearly what is what, don't use the program's native variable names for widget labels. Keep that stuff internal. Be descriptive in the GUI.

10. can you clear the Main menu from all widget, letting only the menubar?

I don't know what this means. I can kill the individual Display windows, and be left with the Menu window. But then I don't know how to get back a Display window. If I kill the Menu window, the program exits.

11. can you start a new 3th Display?

No. I don't understand how to do this.

12. can you find and use the pluging that executes pyart.correction.dealias_unwrap_phase? can you see the corrected velocity in the Display? can you get help on the options?

I found it. It didn't seem to work. The Help discusses the native Py-ART function's options fine, but what I really need is help on how to work the menu widget itself. The Radar/Find Variable drop-down portion is especially confusing. I expect to see a list of radar variables, instead I get ComponentsControl stuff.

13. can you start ROI (from the "Components" menu)?

Yes, but from the Toolbox menu on an individual Display. The Components menu is not intuitive as described above.

14. can you select a region with ROI?

Yes. However, there needs to be simple directions (maybe on the ROI widget) for how to operate the mouse to successfully select a region (e.g., left-click to add a point, right-click to end, etc.)

15. can you view, save and open the ROI table?

Yes. However, the column/row labels don't save in the CSV, limiting the reusability of the CSV by someone who didn't create it.

tjlang, 2015-07-31:
Not sure what's up with the numbering, stupid markup ...

gamaanderson, 2015-07-31:
I must leave now, I will check this later

gamaanderson, 2015-07-31:
Thanks you a lot @tjlang, the problems are pretty much where I thought they would be. Unfortunately it is not that easy to change that, my latest idea would be to use "drag and drop". For instance the display has a menu variables, where you can drag radar to dealias_unwrap_phase etc.

Some specific comments:

    can you understand and correctly use inline option -d, -f, -D and -s?

 sorry it is not -D, it is -F (for --file). About running direct I agree with you, for register some help in how to do this is [here](http://stackoverflow.com/questions/874521/python-install-script-to-system)

    can you change files and sweeps with the arrow keys?

I have the idea of adding a plugin for a tree folder view to move between files. This is however not in my priorities, may be I should change this.

    can you clear the Main menu from all widget, letting only the menubar?

You can use the layout menu to close the CC, we should change that for a bottom in CC, this is not as straightforward but should be doable.

    can you find and use the pluging that executes pyart.correction.dealias_unwrap_phase? 

My plugin is working, but incorrect setting of "vel_field" may send an pyart error to the terminal. I will try to intersect any error to show it in the screen as well as improve the radar selection (drag-drop?!)

Finally, do you understand what the "IndependentWindow" option mean? test un-marking it. Can you suggest something to make it more explanatory?

Thank you again, I will try my best to address all points you raised

tjlang, 2015-08-03:
OK, -F works fine for me.

For the arrow key issue, maybe just consider some helper text in the window. For example, "Left and right arrow keys to change file, up and down keys to change tilt." You've got a lot of blank space in the window to put a few important instructions. Or maybe print a message (not a popup with an "OK" button) stating "File changed to X" or "Tilt changed to Y."

In terms of the Menu window and ComponentsControl, it's hard for me to make suggestions as they just are not intuitive at all. A new user will just not understand what to do with them. The display windows make a lot more sense. I know I can use the Menu window to link the display windows, but I had to approach it from a roundabout way. So you need a button on the Menu window, stating "Link Displays." The user clicks that, and it brings something akin to this: 

[image]

But instead of "conect/disconect" you have link/unlink. Then instead of "Vxxxxx" you have radio buttons for "Radar Objects," "Color Maps," "Fields," etc. And the dropdown menus only have the Display windows as options for linking, not Menu and ComponentsControl as the user is not going to know what it means to "link" those.

Similarly, you need a menu item for "Add Display Window." User clicks the button, and it immediately brings up a Display window.

By the way, when ARTview is more mature I strongly recommend creating Youtube videos for demonstrating how to use it.

nguy, 2015-08-03:
Thanks for the great feedback @tjlang !
I'll address some of the points and comment. Let me know questions or if I misunderstood something.

I agree that the advanced stuff needs a little work/explanation, we should consider how to address this. Maybe even provide a link to documentation that could be clicked. I'll try to dive into the wording used to see if it improves the useability. 

One idea: Using tooltips to popup explanations. Do you find hover-over tooltips to be a useful way of giving information?

The quicksave function seems to have disappeared at some point. It should be easy to add back.

I think adding helper text might be a good thing. Just need to think how to do this without cluttering the display. One thought is to have a true popup window upon startup that has a quick "Introduction". This could become a nuisance later if you use the software a lot. For now wording "Changed to X..." occurs in the terminal window. I can look at putting this into the window somewhere.
Actually using the status bar for this may be possible...

Maybe I can make a few changes and run them by everyone before implementation.

I agree with the ```Add Display Window```.

@gamaanderson suggested YouTube videos also.

tjlang, 2015-08-04:
Yeah, limit popups with OK buttons to dismiss. That is by far the worst aspect of Windows, and it would be bad to include it in ARTview. Hover-over tooltips could be useful. Don't need a lot of space, but some text indicating current sweep, file, and variable would be helpful. Not sure you want to rely on ARTview's default title text to do that job for plots since the user can change the title text.

nguy, 2015-08-04:
I thought the same thing about the title. That's why I included an experimental text field at the top of the display with the buttons now.

I am think about making a Help link in the ```Tilt``` and ```Field``` drop-down menus that would open a help window with simple instructions.

nguy, 2015-08-05:
@tjlang @gamaanderson Should an 'Add another Display' function be under the 'File' or 'Components' menu.

gamaanderson, 2015-08-05:
Do we really need this button? the menu components::display already do that. The problem is that it was not intuitive enough.

tjlang, 2015-08-05:
"Components" and "ComponentControl" should be renamed because they are too vague to be useful to an end user. Once that is renamed, then you can make a decision about what goes where.

nguy, 2015-08-05:
Any ideas on the renaming? "Advanced Tools"?

gamaanderson, 2015-08-05:
rename components to plug-ins

tjlang, 2015-08-05:
Plug-Ins or Advanced Tools is fine.

gamaanderson, 2015-08-05:
to rename ComponentControl you have to rename the class it self

gamaanderson, 2015-08-05:
go with advanced tools, plug-ins have a more specific meaning for programers ( us )

tjlang, 2015-08-05:
Hmmm, I guess I see a problem with using class names as menu labels, etc. Classes are for developers, menu labels are for users.

gamaanderson, 2015-08-05:
also rename the function Menu.addComponent( ) and related ones

gamaanderson, 2015-08-05:
the first user is the developer and the last developer is a user

gamaanderson, 2015-08-05:
the ComponentsControl is so abstract, that I can't image of a label that will give any real help in understanding what it does. The only way to understand is reading some long help or playing with it.

tjlang, 2015-08-05:
Well, what are the range of tasks that ComponentsControl enables? It appears to handle the plug-in interface. Are there other tasks?

gamaanderson, 2015-08-05:
no, it does what you saw and nothing more.

tjlang, 2015-08-05:
OK, the only task I have been successfully able to complete with the ComponentsControl window is to link Display1 and Display2 in some way (e.g., fields, tilts, etc.). Can you complete additional actions with that window? If not, then the obvious menu label is "Link Displays," and the additional dropdown options in the current ComponentsControl beyond Display# need to not be there. If one can do more with the window, then I need to educate myself on those tasks before suggesting a new title.

gamaanderson, 2015-08-05:
It link not just displays, but all components/plugins. For instance to open two different radar you can unlink the displays from the menu and opening a new file will just affect the other display

gamaanderson, 2015-08-05:
Components Control also appear in that list, but it has no linking possibility. This could be tested to avoid CC there.

nguy, 2015-08-05:
This will come up again, but I forgot to mention that when you unlink the ```radar``` and advance files you seem to only be able to advance one window. This isn't necessarily a bug depending on your point of view. I think it could even be sold as a "feature" - see where you started!

nguy, 2015-08-05:
Being neither a good developer or a good user I see both perspectives. I do think we need to cater more to a user base if we want this software to keep going. It has serious potential in my opinion. Once we release a version I am going to send out emails and we can post to PyAOS. Once we get more users, we'll really understand where we can make improvements.

nguy, 2015-08-05:
One question: Will 'Advanced Tools' be confusing when we have 'Toolbox' in the Display already?

Also, right now the menu has "Advance File"? I don't really like that, any suggestions?

gamaanderson, 2015-08-05:
file changes is controlled by Menu, whatever window is not linked will not go with it. This kind of versatility is an important feature (without quotes)

gamaanderson, 2015-08-05:
Who would confuse 'Advanced Tools'  with 'Toolbox'?  they have different name and different locations

from my simple English I always thought it should be "change file", for me "advance" means "forward"

nguy, 2015-08-05:
Glad to hear the tools are distinguishable. I like the 'Change File'.

gamaanderson, 2015-08-05:
Correction: most of labels used are not the component class name, but the component name, it is a string and the default is the class name it self. So you don't need to change the class name to change the label. But that don't change my opinion, if the label is "Link Display", the class should be `LinkDisplay`, clarity is just as important in the code side as in the user side.

Changing the class name is not a problem, components are independent, so the only changes are in the class itself, in `__init__.py`, in the scripts and in the documentation (4 places, but 4 really specific places with no need for back-ward compatibility or tendency to create conflicts). By the way, ROI could really become RegionOfInterest (but, against my original recommendation, ROI is not independent, so changing is a little more complicated)

nguy, 2015-08-05:
I have no problem with removing white space. It even fits the somewhat commonly followed class naming structure.

Okay I'll make the LinkDisplay change and push to #66 to see what it's like.

gamaanderson, 2015-08-05:
Wait, Wait Wait, LinkDisplay was only an example. It does more than linking the display, the closest thing would be LinkPlugins or LinkComponents, but I'm not sure of this name

nguy, 2015-08-06:
uh-oh. Well I can undo this. But I don't see where it does more than Link Displays...

I did keep ```component_contorl.py``` intact, because more was going on in there.

nguy, 2015-08-06:
Also ```RadarDisplay``` is problematic. I find that name to more descriptive, but would like an "Add Display" menu item.

gamaanderson, 2015-08-06:
it link and unlink all variables from all components/plugins, the ones that exist and the future ones.

nguy, 2015-08-06:
Okay see ongoing conversation in pull request #66.

nguy, 2016-03-01:
I think this thread has become outdated now. Shall we close?

gamaanderson, 2016-03-02:
Yes, this is pretty outdated and if I want to re-see this list I can still look in the closed issues.
Clossing.

closed 2016-03-02 by gamaanderson