nguy/artview #59: New Data Container

gamaanderson, 2015-07-27:
This has been in my mind for some time now, it is probably something to add to pyart, but I think it is better we to develop it for internal artview usage and them suggest to pyart

For now pyart has two data containers, as I call them, `Radar` for polar data and `Grid` for Cartesian data. I would like to add a new one (not sure the name) for unstructured data, that is just as list of points (like RoiData). It would be similar to grid in the point that it has a `fields`and a `axes` dict. The axes could be of several kinds (all optional) polar coords, cartesian coods or index referring to the original object (RayIndex etc.). I thought about adding a reference to the original data container, but I'm inclined not to: if the user wants to use the index axes he must know to what radar that is referring.

nguy, 2015-07-31:
I've wondered on and off about this as well. I think there may be some files out there that would benefit from this approach.
I'd be game for us tackling this here if you like.

gamaanderson, 2015-07-31:
actually, I had a lot of free time this week, and already started a [model](https://github.com/gamaanderson/artview/blob/points/artview/core/points.py), you can start from there as well as change everything if you like to. I also didn't really tested it.

gamaanderson, 2015-08-06:
can you add the file I linked in the previous comment to a file in **artview/core/points.py** (unless you want to change the name). I need a container to make roi work for grid, as there is not azimuth etc., the columns change and I don't want to make a ROIgrid just to change the columns.

nguy, 2015-08-10:
Can this be closed with PR #71 ?

gamaanderson, 2015-08-10:
yes, it can

closed 2015-08-10 by gamaanderson