nguy/artview #61: Shared Variables Names

gamaanderson, 2015-08-03:
@tjlang complained about the unexplanatory name of the variables in the ComponentControl, suggesting to use "longer" names instead of the internal name. Although I agree with him from the user side, I'm not inclined to do that, since this means that plug-ins writers would have to register their variables names inside artview. This is contradictory with our intention to make life as easy as possible for writing plug-ins and incrementing artview.
I much more inclined to change the name of the variables for more explicit ones. Here is a list of the shared variables names being used:

* radar
* grid
* container
* field
* tilt
* levelZ
* levelY
* levelX 
* level   
* lims     
* cmap     
* gatefilter

`lims` and `cmap` are the one that most worry me, better name could be `zoom` and `colormap`; `container` is just an alias for `radar` or `grid`, so in theory it doesn't appear for the user

nguy, 2015-08-04:
Where are these defined? I'm having trouble finding them.

gamaanderson, 2015-08-04:
The are listed [here](https://github.com/nguy/artview/blob/master/artview/core/core.py#L29-L71) and are used in several parts of the code (with the capital V in front)

nguy, 2015-08-04:
Sorry my question wasn't clear. When the list is made for the ```menu``` instance, where do the names come from? For some reason I was having trouble tracking that down last night.

gamaanderson, 2015-08-04:
it is done directly from the sharedVariables instance defined in every component individually, and that name MUST match the variable name being used. Changing this is a pretty big modification. 
If you want, the label the user sees in being defined [here](https://github.com/nguy/artview/blob/master/artview/components/component_control.py#L143), from where you can start tracking back. However I do not suggest changing that label, as artview do not control variable name from plug-ins, that list is only a suggestion

nguy, 2015-08-04:
Okay, I see now why that makes it much more difficult.
One thought is that we could add a label to each Vclass instance that would be a "human readable label".
Since we have a limited number of V instances, this would work, correct?

Then in the line you highlighted, instead of making the QLabel the variable name, it could be that label part of the class.

gamaanderson, 2015-08-04:
The problem it that this not a limited list, but a expanding one and plug-ins writer are allowed to add what ever the want. 
simple example, I make two plug-ins, one take radar and put them in a Vradars (with s) and another plug-in use the radars. But because Vradars is not in that list I can't use ComponentsControl? one way would be to allow plug-ins to define also the long-name, but is one more thing that a plug-in writer must do.
I don't think this is really a problem, link/unlink variables is not fundamental for using artview, and user by playing with it will figure it out, as long as names are clear (no abbreviations). That the find variable instance was criticized is what worries me the most, but I think this can be done a lot nicely with drag-drop.

nguy, 2015-08-04:
Okay, I see your reasoning. For now I think we leave it be. In the future we may need to revisit.

gamaanderson, 2016-01-17:
I think the time is closing up to handle the original issue here (change `Vlims` to `Vlimits` and `Vcmap` to `Vcolormap`!). and also remove the value argument from the `ValueChanged` signal (which it totally redundant). This are pretty straightforward changes but they are really all over the place. Or we do it now or those are going to be crystallized such that changing is not worth it.

nguy, 2016-01-18:
I'm fine with closing this as well. It might be wothwhile removing the value argument to reduce future confusion. We can work on this after your grid PR moves through. I just want to make sure that is working first.

gamaanderson, 2016-02-05:
any sugestion on changing `VPathInteriorFunc`?
here is the variable description

        +-----------+-------------------+------------------------------------+
        |VPathInte\ | Hold auxillary    |function like :py:func:`~artview.\  |
        |riorFunc   | function          |components.RadarDisplay.\           |
        |           |                   |getPathInteriorValues` or None      |
        +-----------+-------------------+------------------------------------+

nguy, 2016-02-05:
Since this only holds point data (as far as I can tell), would it make sense to have a ```Vpoints``` or ```Vpointdata```?
This shared variable could also be used for any unstructured data in theory as more plugins are developed.

gamaanderson, 2016-02-05:
no, it does not hold any data, it holds a function that receive a list of paths (matplotlib) and return Points

nguy, 2016-02-05:
Ah, I missed that step. This name is a tough one. Could just have ```Vfunction```, which could be utilized in future plugins?

gamaanderson, 2016-02-05:
yes, it is tough. It can't be just `Vfunction`, as it is not any function, it is a specific one and components make assumption over that.

nguy, 2016-02-05:
If this variable were to be a dictionary with 'name' and 'action' (or whatever) attributes, it could become a generic holder. Is that potentially worthwhile? If not, maybe ```Vpathfunction```?

gamaanderson, 2016-02-05:
It is an interesting Idea, I have to think about it.

gamaanderson, 2016-02-25:
I end up not changing `VpathInteriorFunc`, I don't want to make it any longer that it already is, and `VpathInteriorFunction` don't really help much.

closed 2016-02-25 by gamaanderson