nguy/artview #62: Fast  and Simple Visualisation

gamaanderson, 2015-08-03:
I was thinking: to make artview more useful for pyart users we should add some convenience functions that allow a fast and simple visualization, with no need of mounting the application; that is, you are working with a radar, making corrections and stuff, and now you want to visualize, just run something like `view(radar)`. For that I would like to mount function like this

* `artview.view.view(radar)`
* `artview.view.view(radar, "reflectivity")`
* `artview.view.view(grid)`

also to allow applications with more than one display:

* `artview.view.start() # alias to app = QtGui.QApplication([ ])`
* `artview.view.addRadar(radar)`
* `artview.view.addGrid(grid)`
* `artview.view.execute() # alias to app.exec_()`

This is easy to be done, I would just like to hear some opinions. I also plan to have this before the September  presentation, as it can really come to hand to work/present together with pyart (@scollis @jjhelmus).

jjhelmus, 2015-08-03:
+1 on the ability to start the application within a shell.

I'm unclear on the benefit of a view function over using Py-ART's display classes.  Would the `artview.view.view(radar)` function start the GUI or spawn a interactive window?  Depending on what exactly is planned in this function, it might be a better addition to Py-ART itself rather than ARTView.

gamaanderson, 2015-08-03:
yes, `artview.view.view(radar)` would open an iterative window, with all field and sweep selection features of artview. It is artview, but instead of opening a file the user provided the Radar him self.

the function, just for an idea, would be equivalent to this

    def view(radar, field="reflectivity", sweep=0):
        import artview
        from PyQt4 import QtGui, QtCore

        app = QtGui.QApplication([])

        var = artview.core.Variable
        plot = artview.component.RadarDisplay(var(radar), var(field), var(sweep))

        app.exec_()

of course if you don't have artview it is just better to use direct pyart, but if you do have it is more convenient.

nguy, 2015-08-04:
I"m not sure this gains anything over just opening a file. there really isn't that much overhead and I'm having trouble imagining many situations in which this would be useful.

nguy, 2015-08-04:
That being said, it would be nice to keep this in mind in case ARTview continues to grow. If it slows down, then we should definitley consider a "light" version. Of course, I don't think it hurts to implement and test this.

jjhelmus, 2015-08-04:
Being able to start ARTview from a shell by passing a radar or grid instance sounds like a nice feature that I would use, especially in the middle of an interactive processing session.

gamaanderson, 2015-08-11:
This is done in PR #68 and already available in version 1.0

closed 2015-08-11 by gamaanderson