nguy/artview #64: Mouse selection events overrun

nguy, 2015-08-03:
With the tools, I've noticed that selecting one tool after another can cause some behavior that could effect the use of ARTView. For example in Zoom/Pan is selected and then ROI is selected the user can no longer select curved lines. Instead the image pans. This can be alleviated by resetting the default. 

Maybe a reset of the tools could be added or disconnecting mouse events before connecting for each tool.

gamaanderson, 2015-08-03:
I have known of this for a while, but it never annoyed me, reset was all that I need. If you are to implement something, just be sure not to disconnect ROI, this is much better done by closing the roi window.

nguy, 2015-08-07:
This was fixed in PR #66 with the addition of a reset of tools and not just reset to default.

closed 2015-08-07 by nguy