nguy/artview #65: Filename in Display

nguy, 2015-08-04:
@gamaanderson I want your opinion here. I need to bring the filename over from the menu (or Variable) class to the Display to assign in the new QLabel at top. Which do you think would be easiest?

I was thinking it may be as simple as passing a ```filename```  keyword to the ```plot_radar.py``` Display class?

gamaanderson, 2015-08-04:
no, passing this kind of information must be done with a variable, but I also wouldn’t what to add a Vfilename. You could in theory smuggle it in radar, by

    radar = pyart.io.read(filename)
    radar.filename = filename

and then test

    if hasattr(radar, "filename"):
       print radar.filename

gamaanderson, 2015-08-04:
But when one things get complicated, this is a signal that there is something wrong. In this case, artview is build around pyart, and pyart does not recognize the existence of files (Jonathan will say `Radar` is an abstract structure). Therefore ARTview in theory should follow the lead of py-ART and don't care about filename.

gamaanderson, 2015-08-04:
Just to be clear, I'm just say why of this uncomfortable limitation, not because of the Vvariable being a limitation, but this limitation is showing a concept conflict. I would also want `Radar` to be more closely related to disc files

nguy, 2015-08-05:
Consider it smuggled....for now.

nguy, 2015-08-07:
Everyone okay if I close this issue? It can be reopened if future change is desired.

gamaanderson, 2015-08-07:
ok, you may close it

closed 2015-08-07 by nguy