nguy/artview #74: Python 2.6 needs importlib module

pfhein, 2015-08-11:
Unfortunately we introduced a minor problem with Python 2.6.  Importlib is not in Python 2.6.  Fortunately there is a very small module to download that adds importlib.  My gut feeling on it is that Python 2.6 is not worth supporting.  I am seriously considering moving away from Python 2.6.  It is too old.  Python 2.7 seems to be the current standard and python 3.4 seems to be the future.

gamaanderson, 2015-08-11:
the use of importlib is because of python 3.4 being annoying to import. This is easy to go around with a check, I'm sure it is worth doing. However in the future this will continue to happen, so updating to python 2.7 will be needed soon or later. Any way, when we definitely drop 2.6, we shall let a branch with a "historical" version for 2.6 and comment that on the README.

tjlang, 2015-08-11:
@pfhein I strongly recommend CSU upgrading to 2.7. Py-ART and ARTview are now supporting Python 3, and I have upgraded all my NASA radar modules too. With the Python 2 to 3 transition process, the recommendations I have seen is that the developer shouldn't even try to support anything less than 2.7 going forward.

pfhein, 2015-08-11:
To the documentation we can add a dependency of importlib for Python 2.6, and leave that as a minimal fix.

gamaanderson, 2015-12-11:
can this be closed?

nguy, 2015-12-11:
I think so if other do. 
How long should we go before "officially" not supporting < Python 2.7?

I have been holding off transitioning to Python 3 because I figure I'm a good test environment. One of these days, I need to get both 2.7 and 3 running on my machine.

pfhein, 2015-12-11:
We can close this.  I am ready to stop supporting Python 2.6.  The needed importlib module for 2.6 is one example of how out of mainstream and behind Python 2.6 is today.  We can add to the Readme that Python 2.6 is not supported but with little effort one can get it to work.

jjhelmus, 2015-12-11:
As another data point Py-ART is going to drop support for Python 2.6 after the 1.6 release, ARM-DOE/pyart#386

gamaanderson, 2015-12-11:
that means will (need to) drop when we update to pyart 1.7  (we are still in 1.5).

closed 2016-01-21 by nguy