nguy/artview #77: Obtain a list of shared variables only

nguy, 2015-08-12:
I'm working on Issue #57 right now, creating a GateFilter plugin.

It seems that being able to get a list of just variables would be ideal for this application. I've tried a couple of different methods without success thus far. I've been concentrating on something like ```SelectRegion``` where a display is chosen, but I'd like to drill down to the variables somehow from this.

The reason being that a list of variables is required to create the window output for potential masking on each field.

gamaanderson, 2015-08-12:
I didn't really understand? when you say variable you mean `artview.core.Variable`?

I would like to avoid if possible the need of a display as input, like in `SelectRegion`. I do it there because SelectRegion must work over a display or it has no meaning.

nguy, 2015-08-12:
Basically, what is the best way to get a list of Vradar instances?

nguy, 2015-08-12:
Specifically from ```VariableChoose``` class. This class sets up a UI in which a tree can be navigated. Possibly two questions/comments:
1. It works, but I'm not sure the "average" user is going to successfully navigate the tree.
2. Would it be useful to have some internal methods that we could use to just pull out ```Variable```s, etc. that are buried in this tree?

Just wondering as I begin to really think through plugins.

gamaanderson, 2015-08-12:
I have thought on that, and I haven't found a way using a shared variable list. So we must use a static way, that is, we rely on the variables not changing during the operation. This is an acceptable condition as the values may change, only the variables instances must remain the same. You may have problems if you need to respond to changes (i.e. execute `NewRadar`) as it will probably be difficult to use `connectAllVariables()`, which means your component will not work well with `LinkPlugins` but that is also not that bad.

Ok, how you do that in the end.

1. receive in init `[Vradar1, Vradar2, Vradar3 etc.]` (script time)
2. at execution time use the `ChooseVariable` tool to add new Vradar, for use example see the [plugins](https://github.com/nguy/artview/blob/master/artview/plugins/dealias_unwrap_phase.py#L143-L151)
3. In the future the `ChooseVariable` may be replaced with a drag-drop utility, but this will still take a while.

gamaanderson, 2015-08-12:
responding:
1. your right, that is why I want to change that to a drag-drop
2. how you know what variable you want to get?

nguy, 2015-08-12:
I started from ```calculate_attenuation``` and began modifying. I do have the point #2 in there, but maybe it's not quite right in my code - I'll look.

Basically I'd just like to make a list of Vradar from Display1 and Display2, for example, and let the user choose which they would like to use to start a new Display window. Then all the information I need should be available.

gamaanderson, 2015-08-12:
On getting variables I don't have much more then what is being used in the Display GUIstart. The points is that is not because a variable is not called Vradar that it may not have a radar in it, may be some plug-ins wants to call its radar variable as VradarAirborne! Also Displays are not the only things with radar, Menu also has Vradar and Mapper has Vgrid

gamaanderson, 2015-08-12:
A suggestion I would do is first get the `__init__` start working, and make a script that start this plugins in the script. After that we can concentrate in how to GUI start this plugins. I do recognize our solutions for GUI are still not sufficient and improvements must be made.

gamaanderson, 2015-08-12:
Also if you want, you can get access for the list of components in `core.core.componentsList` and the variables (name) of each component in `component.sharedVariables.keys()`.

nguy, 2015-08-12:
I have the script working through the ```__init__```. I am using exactly what you note. It seems I am returning a ```NewRadar``` instead of ```Vradar``` instance. I need to look at the code a bit more.

gamaanderson, 2015-08-12:
if you upload what you have to a branch, I can give it a look

nguy, 2015-08-13:
After some trial and error and looking at ```VariableChoose.choose_variable``` the following pulled out the ```Variable`` instance instead of the ```NewRadar``` instance:
```getattr(core.core.componentsList[selection], str(variable))```
where selection and variable are integer and string Combobox selections!

Let me get a little further, but I may post later.

nguy, 2015-08-14:
Closing this discussion as superceded by Issue #57

closed 2015-08-14 by nguy