nguy/artview #79: Greater file support

nguy, 2015-08-14:
We should look into adding a secondary check to try ```pyart.aux_io``` read routines to support a greater variety of radar files.

gamaanderson, 2015-08-14:
I agree, however I fear that in some circumstance it could fail some how, a error is not raised but a empty/invalid radar is returned. May be we should allow the user to choose. Its basically one of the classic dilemmas:

* `if: else:` (ask for permision)
* `try: except:` (ask for forgiviness)

I must add that some python gurus advocate for the second.

P.S. this is just some phobia of mine, the important point here is I AGREE.

nguy, 2015-08-17:
I would advocate for the ```try: except:``` method. One problem I do see is that that ```pyart.aux_io``` routines don't have a parent ```read``` routine that calls the "correct" function. This would mean a series of calls or as you mention allowing the user to choose an interface (with an explicit message that all ARTview functionality may not work).

closed 2016-01-21 by nguy