nguy/artview #80: crash after opening file

vanandel, 2015-08-18:
With conda 3.16.0 on OX 10.10.5, installed latest 
artview : de80c70b6f15d4915153f286a79bd92450e94b4a

and pyart: a07b186fd7b9603d5e718daef1ad088bc2510567

I opened a CFRadial file and got the following crash:
Opening file /scr/eldora2/rsfdata/front/cfradial/spol/qpe/20150404/cfrad.20150404_224032.000_to_20150404_224032.000_SPOL_v664_s00_el85.00_SUR.nc
Traceback (most recent call last):
  File "/usr/local/anaconda/bin/artview", line 5, in <module>
    artview.__main__.main(sys.argv)
  File "/usr/local/anaconda/lib/python2.7/site-packages/artview/__main__.py", line 18, in main
    artview.run(DirIn, filename, field)
  File "/usr/local/anaconda/lib/python2.7/site-packages/artview/scripts/standard.py", line 50, in run
    parent=MainMenu)
  File "/usr/local/anaconda/lib/python2.7/site-packages/artview/components/plot_radar.py", line 137, in __init__
    self.LaunchGUI()
  File "/usr/local/anaconda/lib/python2.7/site-packages/artview/components/plot_radar.py", line 166, in LaunchGUI
    self._set_figure_canvas()
  File "/usr/local/anaconda/lib/python2.7/site-packages/artview/components/plot_radar.py", line 672, in _set_figure_canvas
    self.layout.addWidget(self.canvas, 1, 0, 7, 6)
TypeError: arguments did not match any overloaded call:
  QGridLayout.addWidget(QWidget): argument 1 has unexpected type 'FigureCanvasQTAgg'
  QGridLayout.addWidget(QWidget, int, int, Qt.Alignment alignment=0): argument 1 has unexpected type 'FigureCanvasQTAgg'
  QGridLayout.addWidget(QWidget, int, int, int, int, Qt.Alignment alignment=0): argument 1 has unexpected type 'FigureCanvasQTAgg'

gamaanderson, 2015-08-18:
@vanandel Thank you for reporting this bug.

Apparently (py)Qt fail to recognize matplotlib canvas as a Widget, I'm not sure why is that, since `matplotlib.backends.backend_qt4agg.FigureCanvas` is a subclass of `QtGui.QWidget` and for me `isinstance(QtGui.QWidget, self.canvas)` returns `True` as expected.

The following information could help me in debugging this:
python version
Qt version
PyQt version
Matplotlib version

gamaanderson, 2015-08-19:
I have found just one reference for a similar error: (http://stackoverflow.com/questions/6723527/getting-pyside-to-work-with-matplotlib)

Any way here is a test script that should isolate the problem :

    #!/usr/bin/env python
    import sys
    import matplotlib

    # matplotlib.use('Qt4Agg')
    # matplotlib.rcParams['backend.qt4']='PyQt'

    from matplotlib.backends.backend_qt4agg import FigureCanvasQTAgg as FigureCanvas
    from matplotlib.figure import Figure

    from PyQt4 import QtCore, QtGui

    app = QtGui.QApplication(sys.argv)

    # generate the plot
    fig = Figure(figsize=(600,600), dpi=72, facecolor=(1,1,1), edgecolor=(0,0,0))
    ax = fig.add_subplot(111)
    ax.plot([0,1])
    # generate the canvas to display the plot
    canvas = FigureCanvas(fig)

    win = QtGui.QMainWindow()
    # add the plot canvas to a window
    win.setCentralWidget(canvas)

    win.show()

    sys.exit(app.exec_())

if this don't work, as I expect, try uncommenting the matplotlib line

gamaanderson, 2015-08-19:
I have successfully replicated this error. This will happen if one has both PySide and PyQt4 installed and PySide is the default for Matplotlib (for instance because it has installed first). The solution is just to tell Matplotlib to use PyQt (`matplotlib.rcParams['backend.qt4']='PyQt4'`) as in PR #81. 

again thank you @vanandel for reporting this bug and helping us making ARTview better.

nguy, 2015-08-19:
@vanandel The above solution was merged in #81 . Can you update the code and let us know if this works for you?

nguy, 2015-08-27:
If there are no objections, I'd like to close this issue since it in particular has been solved.
Maybe we can open a second issue with the details of the discussion we had in #81 to keep it alive so we can look into this later?

gamaanderson, 2015-08-27:
That can be done, but I don't see the point in closing an issue just to open an other one.

nguy, 2015-08-27:
Good point. Probably should figure out what we are going to do and just take care of it.

gamaanderson, 2015-09-11:
I have give some tough on the matplotlib problem, and decided that I'm in favor of not testing in execution time, simple because it is not really our responsibility to interfere with matplotlib and the environment flags. I don't really agree in how matplotlib is doing things, but I must live with that. 

We should rather inform about the need of have matplotlib importing PyQt4 in README as a kind of pre installation testing, we can even make a script that does this testing. If one thinks this is too much for README we can add a INSTALL file.

I also don't think now is the time to expand to other Qt front ends. This decision should be done later, in especial in the transition to PyQt5. To prepare for that we should however centralize all imports to one file, I think this will cost us nothing and may spare some headaches.

If there is no one against it, I may implement. They are small alteration but code-wide.

nguy, 2015-09-11:
When you say have matplotlib import do you mean ```(matplotlib.rcParams['backend.qt4']='PyQt4') ``` or the answer that was given on the StackOverlow post?

I agree that we should make a note in the README. I would guess this issue won't come up that often, but this way it's a known limitation under specific circumstances.

Do you happen to know the status of PyQt5? I haven't kept abridged.

gamaanderson, 2015-09-11:
by matplotlib import I mean that this

     from matplotlib.backends.qt_compat import QtCore, QtGui
should be equivalent to

     from PyQt4 import QtCore, QtGui

PyQt5 is already stable and ready for use. However transition is still not complete and Qt4 is still official supported til the end of this year.  But this does not mean that we should make the transition by then, for instance KDE (that works over qt) have still no single version made with qt5, which means that even if Qt5 can be installed and used in parallel , the KDE application will all still run in Qt4.

So there is for now no reason for going PyQt5, unless there is a feature we want to use. And PyQt4 must still be supported for several years.

nguy, 2015-09-11:
One question: when was the backend compatibility implemented? We may need to do a try statement here and then go to directly import from PyQt4 if failure occurs.

Sounds like we should take our time in migrating to qt5 then.

gamaanderson, 2015-09-11:
I don't remember when, but it was not too long ago, but that is why I say we concentrate the import in one file, so the other can import from there safely.

nguy, 2015-09-11:
Okay, I say go for it and see how it turns out. I'll not make any changes unless something comes up at the conference and a quick fix is needed.

gamaanderson, 2015-09-11:
okay.

nguy, 2016-01-21:
Closing this issue for now as it has not come up again recently. Can always reopen a similar ticket later.

closed 2016-01-21 by nguy