nguy/artview #83: Include style sheet change for matplotlib

nguy, 2015-08-27:
Allow user to set the style sheet for those that like dark background. Would need to institute something like the following in the plot_*.py files:
```python
with plt.style.context(('dark_background')):
    d.plot_ppi('reflectivity',0)
```

gamaanderson, 2015-08-27:
This could be added to the Display Options menu, while doing one must also allow a way of going back to white without resetting.
This would however means that display would always open in white, do you think it would be important to allow a change of default to open in black? there is no easy/clear way of doing this since we have no configuration setting.

nguy, 2015-08-27:
As of right now, I don't think it's important in terms of configuration. Actually come to think of it, this type of configuration would be better suited to PyART if anything. What are your thoughts @jjhelmus to include something like this in the config file. Of course it would need to be added to the ```RadarDisplay``` and the like.

gamaanderson, 2015-08-27:
I disagree, I think this is the first example we have of a artview configuration not a pyart configuration. This can be done in pyart as an function argument if they want to.

nguy, 2015-08-27:
Okay I'm open to exploring this. The cool thing is that the configuration could actually call a user's settings if they've set their own style sheet I think. I feel like a .cfg file could be loaded on startup with this any any future settings as yet to be determined.

jjhelmus, 2015-08-27:
I agree with @gamaanderson, this option does not belong in Py-ART.  The same functionality could be accomplished in Py-ART either using the context manager example you gave on a per call basis or globally by having a user set their `matplotlib.rc` file.  Since Py-ART target audience is writing scripts or IPython notebooks they can directly manipulate matplotlib settings using matplotlib's functionality.  ARTView need to provide a this functionality directly since users cannot directly manipulate these settings.

nguy, 2015-08-27:
Alright, we'll explore adding this option to ARTView. It's low on the priorities, but just wanted to get this down in writing before I forgot.

gamaanderson, 2015-08-27:
Thanks @jjhelmus for the comment, it gave me the idea, this can be set in the script with

    matplotlib.rcParams['axes.facecolor']='black'

As we do incentive user to write/modify scripts this a perfectly valid way of changing the default.  I have nothing against adding a `config` to `artview`, but if we can avoid it, I think it is better.

nguy, 2015-08-27:
Will this only turn the face black though and not the adjust labels, etc.?

Another option if we wanted to "brand" ARTview would be to create our own stylesheet.
Here's an [example script](http://matplotlib.org/users/customizing.html#matplotlibrc-sample). 
Just a thought, I'm not endorsing any ideas right now.

gamaanderson, 2015-08-27:
matplotlib.style.use(('dark_background'))
also works

nguy, 2015-08-27:
Yep that's the one that I used and it takes care of a multitude of settings (like axis font colors, etc.).