nguy/artview #87: Display Options -> Adjust Display Limits

andrevf, 2015-08-31:
When we select the Data Min and Data Max the artview changes the color bar, but it is not masking the data outside the interval (Data Min, Data Max). For example, if the data is under 10, when we set Data Min = 15, everything becomes blue, but it is still plotted. I realized it while using plot_ppi(), and in this case the solution is very simple. One of the options of plot_ppi() is mask_outside=False (this is the default), if we set mask_outside=True in /artview/artview/components/plot_radar.py, in line 741:
# Create Plot
            self.plot = self.display.plot_ppi(
                self.Vfield.value, self.Vtilt.value,
                vmin=cmap['vmin'], vmax=cmap['vmax'],
                colorbar_flag=False, cmap=cmap['cmap'], mask_outside=True,     <------------------------
                gatefilter=gatefilter, ax=self.ax, fig=self.fig, title=title)

then data outside (Data Min, Data Max) will be masked and artview will display only the data between Data Min and Data Max.
This option, mask_outside, also exists in plot_rhi(), and in plot_sweep_grid().

gamaanderson, 2015-08-31:
pyart recently changed it default from `mask_outside=True` to `mask_outside=False`, we are still some what behind. Thank you for pointing this out @andrevf. Can you implement @nguy? I have to take my plane now!

nguy, 2015-08-31:
Thank you @andrevf The change has been made, just pull down the most recent version of the code. I am going to keep this issue open to make sure there are not unintended consequences before closing

andrevf, 2015-09-01:
Ok. Thank you.

closed 2015-12-09 by nguy