nguy/artview #88: scripts

gamaanderson, 2015-09-06:
I've realized that if we want user to take advantage of artview versatility we must first show them that versatility and give a starting point. I thing the best way of doing that is increasing the number of scripts and show they better, so I did it and also took some printscreens 

(https://github.com/gamaanderson/artview/blob/ng-gatefilter/SCRIPTS.md)

let me know what you think. And FYI this is build over the ng-gatefilter branch

nguy, 2015-09-08:
These are great! I've been meaning to do that @gamaanderson  
I just officially added the gatefilter portion. Do you want to do a pull request with these and then we can link the documentation later?

Also, I like your comments about "share" versus "link". I would not be against changing the naming conventions in the code and labels to indicate this.

gamaanderson, 2015-09-08:
Yes, scripts are kind of a free part of the code, so I have no problem with direct merges. I also don't have a problem with the documentation not having that images, they can be seen in github.

nguy, 2015-09-08:
Alright scripts are merged!

closed 2015-12-11 by gamaanderson