nguy/artview #94: Plugin for IO

gamaanderson, 2015-09-15:
Make plug-in for more advanced io operations. It must in any case have a tree view of directory. It could also have an optional cache of radar files.

nguy, 2015-09-16:
Something that might be nice is to pop up a list of files in directory and click on a file to display the data.

gamaanderson, 2015-09-16:
that is exactly what I mean, Qt has already specific tools for doing such.

gamaanderson, 2015-09-21:
I have this implemented in a [branch](https://github.com/gamaanderson/artview/tree/treeview). Use the view script (`artview -s view`) to give it a look. Use double clicks to move between folders (.. for back) and open files and right-click for more options.

I named this AdvancedIO, but for now it is only useful for opening. I have no good idea of how to add the write part.

nguy, 2015-10-22:
Closing issue since this was merged

closed 2015-10-22 by nguy