nguy/artview #95: Hand edit data - mask

nguy, 2015-09-21:
Would like to implement hand editing data in Solo-like fashion. The functionality desired is to uses the polygon selected by SelectRegion instance and apply mask to that region if desired.

nguy, 2015-09-21:
A first pass is available in [this branch](https://github.com/nguy/artview/tree/ng-regionmask). The functionality seems to be working however the Vgatefilter variable is not shared properly at present.

gamaanderson, 2015-09-21:
First, I'm not at all familiar with solo, so I don't really know what that means. But I got the idea from the commit

Ok, we real should start communicating more before doing thing. I already had some [developments](https://github.com/gamaanderson/artview/commit/b9e0c6d5f656b4e733525584049090439b417bb4) in this sense, but I was not sure if we would want to use it. Note that is not recommended to use privat pyart methods (i.e. `GateFilter._merge`), that is why I have this [alterations](https://github.com/gamaanderson/pyart/commit/5f393aac28eab50330350fd94aa3d40cd709aa3c) in pyart, that we should first try to PR.

As you already done your version, I invite you to see mine, may be I give you some ideas. I'll add my traditional code comments in the commit, take care to not get lost with the multiple discussions

nguy, 2015-09-21:
Yes I agree. I should have posted this issue before embarking on the code. This was spur of the moment last week at night during the conference. This was the specifically asked for by at least 4 different people. I'll look at your comments and code. Did you submit the changes in PyART?

gamaanderson, 2015-09-21:
No, I never get to submit to PyART.

gamaanderson, 2015-09-21:
Really, 4 people asked for that? I found the ability of erasing everything out of a region some what unnecessary, I could just ignore it.

nguy, 2015-09-21:
It is a highly used function in parts of radar meteorology for cleaning up data that just doesn't respond to QC filters.

nguy, 2015-09-21:
Making some of the changes you suggest and I'll push them over for further review.

gamaanderson, 2015-09-22:
Open PR for pyart: https://github.com/ARM-DOE/pyart/pull/377

nguy, 2015-10-21:
Can this be closed now or should we keep it open to remind ourselves of the minor change to make in the code?

gamaanderson, 2015-10-22:
let it open.

nguy, 2016-03-02:
I think this has also been addressed and can be closed now?

gamaanderson, 2016-03-02:
ok, lets close

closed 2016-03-02 by nguy