nguy/artview #96: User Communication

gamaanderson, 2015-09-23:
So the good news is that we have a small, but growing user community. The bad news is that is community is silent. This is absolutely normal, but it would be really important to create a conversation. Ideas?

gamaanderson, 2015-09-23:
So as our components are already creating some useless outputs and I don't think we will clean that any time soon, we could take the opportunity and add in the end of every script a print asking for feed back, I don't think this will annoy any one . We could also open a mail list.

nguy, 2015-09-25:
Yes, engaging can be difficult. I'm open to a statement print out. If it's in the terminal screen or text in the Menu Window it would be very unobtrusive.

closed 2016-01-21 by nguy