nguy/artview #97: Manual Unfold

gamaanderson, 2015-09-24:
This I idea just came to me: add a plugin for manual velocity unfolding, that is, the user select a region (with the `SelectRegion` plug-in or with some cluster identification tool) that he knows is folded and we unfold it! Using a `SelectRegion` it is pretty easy, just need to link Vradar and Vpoints and perform the operation.

@jjhelmus, this is probably a pure ARTview functionality, but I just wanted to let you know, may be you have some idea that we can use/contribute to pyart.

jjhelmus, 2015-09-24:
Nick and I discussed this at the AMS conference last week and it was a commonly requested feature of ARTView.  What would be nice in such a plugin would be to have a button/option to first perform unfolding using one of Py-ART's dealiasing routines and then allow hand unfolding of the results.  The idea here being that the automatic method can get you a long ways but the resulting volume may still need some minor modifications.  Exposing the regions used in the region based dealiasing method to the user to that they could easily select and unfold a region where velocities are similar would also be helpful.

gamaanderson, 2015-09-24:
There are already some plug-ins implementing pyart unfolding, so this already partially done. I could really fast implement a simple manual version, but I think using it will probably be "uncomfortable". I would really like to user could just click in a region and we identify the borders. Also if we could separate the already implemented pyart methods in two parts: identification and unfolding, we could try to show/modify the identification results (GateFilter ?) before applying. Would this separation be doable?

jjhelmus, 2015-09-24:
Definitely doable, and in fact already partially implemented.  The `dealias_region_based` function unfolds a volume sweep by sweep.  Identification of regions of similar velocities is accomplished using the [_find_regions](http://arm-doe.github.io/pyart-docs-travis/dev_reference/generated/pyart.correct.region_dealias._find_regions.html#pyart.correct.region_dealias._find_regions) function.  This function requires numpy arrays which would need to be extracted from the Vradar but should be doable.  Injecting this back into the unfolding algorithm would take a bit more work.

nguy, 2015-09-25:
Doh, I forgot to mention this one in the email! Thanks for bringing it up @gamaanderson 

I wonder if adding a button for the manual unfolding to the pyart plugins would be the way to go. That way the regions that do not get completely unfolded can be dealt with by popping up a SelectRegion instance.

gamaanderson, 2015-09-25:
Just added a initial plugin in this [branch](https://github.com/gamaanderson/artview/tree/manual-unfold). For convenience use the *manual_unfold* script (`artview -s manual_unfold`), to avoid having to link, you may need however to give the nyquist velocity and change the colormap limits (no chage is made to the nyquist velocity of radar). This plugin work by its own, but of course you can open the pyart-unfold plugins in the advanced tool menu.

gamaanderson, 2015-11-06:
@deeplycloudy following your comments I improve the manual unfold. I still not done with it, but you can already see it in this branch:
https://github.com/gamaanderson/artview/tree/manual-unfold
you can start it in the modes menu under "manual_unfold". I have the felling it is useful.

one thing that is important to say is that, in order to allow successive selection/unfold, the unfold process pick the data from the original velocity and menge into the corrected velocity (if it exists), points outside the selection are not replaced.

The select region has already the "back" button, the unfold widget still need implementation. If someone miss the buttons that were in the `SelectRegion` widget, those are to be re-implemented in a `PointsDisplay` component and a respective "extract points" mode.

deeplycloudy, 2015-11-09:
@gamaanderson I've passed along your branch to two grad students in the program here, who will be giving it a try. One of them is looking at tornado RHI data, which is traditionally one of the more challenging unfolding problems and will require some manual intervention.

gamaanderson, 2015-11-09:
ok, looking forward for some feed back. I should said that in the way artview/pyart is done RHI should work just like PPI, but i have just tested with PPI files.

gamaanderson, 2015-12-11:
just to comment it hier, we merged that branch (together with some updates). There is also a [video](https://www.youtube.com/watch?v=1ehZXbp7000) that exemplify the use.

For continue this issue:
1. Are the current user iteration satisfatory?
2. How to better integrate that with pyart unfold? For now it is possible to make a pyart unfold and then improve that unfold manualy.

nguy, 2015-12-11:
@deeplycloudy @tjlang @pfhein @bdolan44
At your earliest convenience, could you take a look at this newest merge and provide some feedback?

deeplycloudy, 2015-12-11:
Sorry that the grad students didn't get back to you. I just looked at the new merge. 

I had to watch the video (and read your comment above) to figure out that I needed to use the 'mode' and not the 'advanced tool'. I presume that'll be cleaned up later once the manual unfold mode has been fully tested.

I like that there is a simplified interface for SelectRegion in the ManualUnfold mode.

I was able to draw a polygon but the unfold buttons didn't do anything, even though the corrected_velocity field was created and I got the Mac beachball, indicating that some serious work was happening. Is there any chance that this is because I was working with RHI data that had negative range (negative x-axis) values?

pfhein, 2015-12-12:
@deeplycloudy With a PPI data (named VR) I also did not get a change in my display.  I wonder if the change is happening but the display is not updating.

gamaanderson, 2015-12-12:
@deeplycloudy it turns selectRegion is not working correctly for RHI files, I will correct that as soon as I can.

@pfhein did you find and visualized a new corrected_velocity field, with bigger color range? If your file have different name then the pyart-default, you may need to change the names in the Manual Unfold.

nguy, 2015-12-14:
I think we may need to find a way to have corrected velocity use the same data limits as velocity (and let user change) or make that more straightforward. We'll have to think about these tweaks once we get more feedback on the operation.

nguy, 2015-12-14:
@gamaanderson has begun to submit fixes in PR #115 for a couple of these issues.

Per my above comment, it is easiest to connect the limits and cmap of the displays for checking results once the ```corrected_velocity``` field is created. Note you may need to refresh the display (by changing tilt or such to see the updated field).

pfhein, 2015-12-15:
@gamaanderson - It was my stupid mistake of not changing to corrected_velocity to see the unfolding.  It is working.  I appreciate all you are doing.  Thank you!

gamaanderson, 2015-12-21:
PR #115 is merged.

nguy, 2016-03-02:
I'd like to close this issue and refer later if needed. It will see full implementation with the 1.2 release and any bugs can be listed in new Issues.

gamaanderson, 2016-03-02:
i'm ok with closing

closed 2016-03-02 by nguy