nguy/artview #98: Display sizes

nguy, 2015-09-25:
On smaller screens the display sizes and window sizes can be skewed in aspect ratio and go off the screen. We may want to implement a dynamic resizing of windows depending on system.
Will have to look how to get these numbers from the OS.

nguy, 2015-09-25:
Width looks like it is hard coded.

gamaanderson, 2015-09-25:
the problem is that I'm in as 15 in screen and I am already missing more space, if the screen is much smaller don't mater how much er re-size, the two display model will still look bad,  some things as menus have minimum size.

That said, if you get some thing better that what we have [now](https://github.com/nguy/artview/blob/master/artview/scripts/standard.py#L73-L86), fell free to push.

nguy, 2015-09-25:
Yep, I just saw ARTView on a 13" screen for the first time. I am helping a student in the dept and converting them to python and ARTView :).
I will put it on my list and see what I can find.

gamaanderson, 2015-09-25:
For these use case I suggest to use the radar script, instead of the standard, or include all components as tabs in menu (`menu.addLayoutWidget()`). You lose the side to side comparison  and will have to keep changing tabs, but it does look better.

gamaanderson, 2017-01-29:
UPDATE: PR #177 should do an improvement on this, but no movement is being made to accommodate really small screens. This would mean changing to a more app like format, I don't think this is desirable right now.

nguy, 2017-02-03:
I agree. Small screens-focused app format is beyond the scope now.

closed 2017-08-28 by nguy