nguy/artview #99: Install Error

dcon691, 2015-10-03:
Has anybody had any issues installing ArtView? I was able to install Pyart without any issues but when I try to install Artview, I receive an error message.

The line I'm using to install the package into Anaconda is: 
conda install -c https://conda.anaconda.org/jjhelmus artview

With the error being: 
Fetching package metadata: ......
Solving package specifications: .
Error: Unsatisfiable package specifications.
Generating hint: 
[      COMPLETE      ]|###################################################| 100%

Hint: the following packages conflict with each other:
  - artview
  - python 3.4*

Use 'conda info artview' etc. to see the dependencies for each package.

Any help would be much appreciated! Thanks!

gamaanderson, 2015-10-03:
I didn't knew you could install artview with conda, apparently @jjhelmus has made us a favor and added it to his account. I will try to see what is the problem there, but in the mean while you can install it from git: 

* Download the source,
* unpack, 
* move to the unpacked folder
* in a terminal execute `python setup.py install` (more [here](https://github.com/nguy/artview/blob/master/INSTALL.rst))

Artview should run in python 3.4, but we are running it normally in python 2.7, so there may be unknown bug, please report if you found one.

Thank you for your interess in Artview, we are a small group and every new person using it a victory. We whould also be extremely happy with some feed back, so we can better plane future work.

gamaanderson, 2015-10-03:
The problem seems to be as simple as: the compiled version for python 3.4 are not available in conda. As I said that is not an official repository. @nguy wanted to make one, but I don't think he got to it.

dcon691, 2015-10-03:
For some reason this isn't playing nice during the install. The main reason why I tried the conda install is because the method you listed above isn't working for me.

Error Message:
testing matplotlib qt backend ...
Traceback (most recent call last):
  File "/home/dconrad/anaconda3/pyart/artview/tests/qt.py", line 17, in test_matplotlib_qt_backend
    from matplotlib.backends.qt import QtCore
ImportError: No module named matplotlib.backends.qt

... Test fail is an expected way! We would like to know why, please report in 'https://github.com/nguy/artview/issues'
Traceback (most recent call last):
  File "setup.py", line 191, in <module>
    setup_package()
  File "setup.py", line 159, in setup_package
    from numpy.distutils.core import setup
ImportError: No module named numpy.distutils.core

Numpy and matplotlib should have both been installed from conda (Able to do "Import Pyart" in Anaconda without any errors) which is what's really surprising. Any suggestions?

gamaanderson, 2015-10-04:
matplotlib is mandatory in ARTview, but just optional in Py-ART, so having Py-ART installed is no  guarantee. However you also have problems with numpy as well, so this is strange, if you could open a python terminal and do the following:

    import matplotlib
    import numpy
    print(matplotlib.__version)
    print(numpy.__version__)

this would at least guarantee you have both and give me their versions.

If their are properly installed, I don't know what could be happening, my only guess would be you have python2 and python3 installed, they are in one and you are running artview in the other.

dcon691, 2015-10-05:
Here's the version numbers for those two
>>> import matplotlib
>>> import numpy
>>> print(matplotlib.__version__)
1.4.3
>>> print(numpy.__version__)
1.9.3

Here's also the version numbers for Python and Anaconda

Python 3.4.3 |Anaconda 2.3.0 (64-bit)| (default, Jun  4 2015, 15:29:08) 
[GCC 4.4.7 20120313 (Red Hat 4.4.7-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.

I don't know if it makes a difference but I'm running it on CentOS 7.

Thanks again for all your help!

gamaanderson, 2015-10-05:
This just increases my wounder, you version are almost the same as mine and I run artview in python3 with no problems. I really don't know why those imports (mainly `from numpy.distutils.core import setup`) are falling.

One thing you could do is skip the installation process, it is not that important, and run artview direct from the folder (`python artview`). However two foolish `import version` will get in your way. This [commit](https://github.com/gamaanderson/artview/commit/2ad49a25c8985025565cb538de7d902a105e06cf) (still not in master) show how to bypass them.

I'm not confident that skipping installation will solve the problem, it will probably just bring a different import error. Be free to open the source code to make some debugging if you want, as I said I totally perplex.

jjhelmus, 2015-10-05:
Can't offer much in terms of help on the install.  I created an artview conda package from the 1.0.1 tag while at the AMS radar conference to try to help with the on site install.  The conda package did not work for some reason but I never removed the install files from anaconda.org.  Sorry about that and for any confusion it has created.   I'll move them to a locked channel shortly, since they are not going to help users install ArtView.  

Longer term I would like to provide conda packages of ARTView, just need to work out some logistics and do proper testing.  I will open a new issue for that discussion when I have some time to work on the task.

jjhelmus, 2015-10-05:
One possible issue is that Py-ART requires matplotlib but not a specific backend like Qt but ARTView does require Qt.  Using conda you should be able to install Qt with:

```
conda install pyqt
```

nguy, 2015-10-06:
Hmm, this is really an odd error. I also have nearly identical packages installed.

I just want to check, @dcon691 do you have PySide installed or any environmental variables set for python Qt?

nguy, 2015-10-06:
I just noticed that artview is a subfolder in pyart from the post from @dcon691 above. @gamaanderson and @jjhelmus would this be a potential cause for the error?

gamaanderson, 2015-10-06:
I don't think this would be an problem @nguy, nor would a qt environment be. The problem happens even before any part of artview is loaded.

dcon691, 2015-10-08:
@nguy No I haven't done anything special with those besides running jjhelmus's command of "conda install pyqt". 

Still getting the error of:

testing matplotlib qt backend ...
Traceback (most recent call last):
  File "/home/dconrad/anaconda3/pyart/artview/tests/qt.py", line 17, in test_matplotlib_qt_backend
    from matplotlib.backends.qt import QtCore
ImportError: No module named matplotlib.backends.qt

... Test fail is an expected way! We would like to know why, please report in 'https://github.com/nguy/artview/issues'
Traceback (most recent call last):
  File "setup.py", line 191, in <module>
    setup_package()
  File "setup.py", line 159, in setup_package
    from numpy.distutils.core import setup
ImportError: No module named numpy.distutils.core

gamaanderson, 2015-10-08:
Just to inform, the newest git version allows you to execute artview in place with `python artview` or `python3 artview`, with no need to install. I do expect this however to just result in a different error.

dcon691, 2015-10-08:
Actually, that worked. It did spit out an error (pasted below) but the program seems to be working fine. Was able to load some Level II in it and it ran fine. Thank you for everyone's help!

artview/../artview/__init__.py:47: UserWarning: No ARTview Version!
  warnings.warn("No ARTview Version!")
artview/parser.py:23: UserWarning: No ARTview Version!
  warnings.warn("No ARTview Version!")
artview/../artview/parser.py:23: UserWarning: No ARTview Version!
  warnings.warn("No ARTview Version!")

gamaanderson, 2015-10-08:
Well, I didn't expect that! But nice.

That error you see is just because you skip installation. You will continue to see that, as long as we don't figure out how to respond to this development.

@jjhelmus this is probably also interesting to you, since installation of artview follow the same process as pyart. For some reason (may be numpy was installed using conda) the `setup` function is not available.

jjhelmus, 2015-10-09:
I'm also a bit confused why this solved the issue, but good to hear.  

As we use compiled extension modules, Py-ART cannot be imported from a source tree without at least using `python setup.py build_ext -i` or another build/install command.   Py-ART does have some logic in the [__check_build](https://github.com/ARM-DOE/pyart/tree/master/pyart/__check_build) directory which will inform users that they need to install Py-ART and cannot use it from the source tree.  For example after a new checkout of the source:

```
$ python -c "import pyart"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "pyart/__init__.py", line 20, in <module>
    from . import __check_build
  File "pyart/__check_build/__init__.py", line 46, in <module>
    raise_build_error(e)
  File "pyart/__check_build/__init__.py", line 41, in raise_build_error
    %s""" % (e, local_dir, ''.join(dir_content).strip(), msg))
ImportError: No module named _check_build
___________________________________________________________________________
Contents of pyart/__check_build:
README                    __init__.pyc              __init__.py
_check_build.c            _check_build.pyx          setup.py
___________________________________________________________________________
It seems that Py-ART has not been built correctly.

If you have installed Py-ART from source, please do not forget
to build the package before using it: run `python setup.py install` in the
source directory.

It appears that you are importing a local pyart source tree. For this, you
need to have an inplace install. Maybe you are in the source directory and
you need to try from another location.
```

This feature was adapted from the [scikit-learn](https://github.com/scikit-learn/scikit-learn/tree/master/sklearn/__check_build).  It works but is probably a bit excessive for artview since it requires compiling a C extension.

nguy, 2015-10-14:
This still confounds me. I'm not sure the logic @jjhelmus should be necessary as he mentions. But maybe worth trying.

gamaanderson, 2015-10-14:
It is not worth trying, it is even contra productive. This logic test for a fail install (build), our problem here is that installing is falling, but artview still works perfectly well. I would add the following (or equivalent) to README or INSTALL and close this issue:

    If for some reason installation fails, artview can be run locally with no installation. This will however show the "No ARTview Version!" warning!

the other possibility would be to not use `setup` from `numpy`, but implement our own, which is quite complicated.

gamaanderson, 2015-10-14:
We could also add a list of known issues and work-arounds

jjhelmus, 2015-10-14:
Is using the setup is one provided by the standard library's [distutils module]([https://docs.python.org/2/distutils/index.html#distutils-index]) or [setuptools](http://pythonhosted.org/setuptools/) an option.  Py-ART uses numpy.distutils because it needs to compile C extensions and access numpy's include information.  For pure Python functions this might be over complicated.

gamaanderson, 2015-10-14:
This is also a possibility.

nguy, 2015-10-14:
We're actually using numpy.distutils.core.setup it looks like. We could easily use either of the above mentioned I think for what we do. What's the advantage of using numpy-based setup?

closed 2016-01-21 by nguy