nguy/artview #105: ARTView conda package

jjhelmus, 2015-10-06:
At the AMS radar conference I tried making a conda package for ARTView which would make installing easier for users.  My testing was promising but the package did not work.  I think this was due to a mismatch between the ARTView tag I choose (v1.0.1) and the version of Py-ART used (1.5.0).  

Is there an interest in a conda package for ARTView?  If available users could install a fixed version of ARTView using `conda install -c channel artview`.  

I currently make conda package for every Py-ART release and conda is the primary means by which users install Py-ART.  We highly encourage users to choose this over other install methods (setup.py, pip, etc) unless they need to use features that have not made it into a release and would need to install from source.  

In ARM-DOE/pyart#390 I have a task of creating conda packages for other radar tools.  If there is interest I'll add ARTView to this list.  To accomplish this it would be helpful if "good" version of ARTView were tagged with a version number.

gamaanderson, 2015-10-07:
That would be great @jjhelmus, we have done some progress since v1.0.1, but it is hard to say if we have a clean version. Our development progress is still not as organized as your with Py-ART. We must work on that, but our increasing use of PR has resulted that at least the master branch is quite stable.

Just a question: in conda is the `setup.py` script executed by the final user, I mean in his environment? or is it executed just when creating the conda package?

One comment is that we do incentive user to customize and add their own plugins to artview, installing by conda will make that a little more difficult, but I think we must accept the fact that 90% of the users will never want to even see the source code, for the rest git is still available.

jjhelmus, 2015-10-07:
In conda the setup.py script is executed only during package creation, not during install.  This creates a bz2 archive which is then unpacked into the user's site-packages directory.  The actual process is a bit more involved than this, but is the basic idea.  

The end goal of a conda package is to provide the user with a "binary" package which can be installed only by unpacking an archive file into the appropriate location.  No compiling, scripts, etc should be run.  This means that whatever is packaged is what the user gets, nothing extra gets pulled in during install.  Luckily updating is very easy with conda (`conda update package`) so upgrading to the latest version is much simpler.  

Do plug-ins need to be added to the source tree before building, or can they pushed into the artview directory post install?

gamaanderson, 2015-10-07:
Plug-ins, get added whilst `__init__` list the directory and try to import everything that ends with `.py` (and don't start with underscore). So adding it after installation should work, it just happen that the user will need to look after the place where it is installed.