nguy/artview #108: Py-ART Grid refactor.

jjhelmus, 2015-11-03:
I'm planning on a large refactor and reorganization of the **Grid** class in Py-ART and want to check on how much ARTView depends on the organization of the class.  If needed it should be possible to keep a compatibility layer which allow access to the old **Grid** attributes but if this is not needed I would like to avoid it.

nguy, 2015-11-03:
I think @gamaanderson can answer better, but it looks like axes information is the most used part of the **Grid** class in ARTView. What kind of changes are you considering?

We could probably write a check upon a **Vgrid** initialization that checks attributes to account for changes.

gamaanderson, 2015-11-04:
There is no need for a compatibility layer, we can copy the current version of `artview/components/plot_grid.py` to a new file and check pyart version while loading. All other uses of grid outside this file should not be dependent of any internal organization of grid (**thanks to artview's code independence model!**). Just inform us later which is the version to check for.

jjhelmus, 2015-11-04:
Good to hear.

I'll outline the new layout I have in mind for the **Grid** object in ARM-DOE/pyart#285 tomorrow to gather feedback before implementation.  I have it drawn out on paper but have not had the time to write it up nicely.

closed 2016-01-21 by nguy