nguy/artview #109: Vcmap color table value is an object or not

pfhein, 2015-11-10:
Vcmap changes depending on whether the color table.  For the color table `jet` if you `print dir(Vcmap.value['cmap'])` you will get 
```{'vmax': 65, 'cmap': <matplotlib.colors.LinearSegmentedColormap object at 0x7fe02770eed0>, 'vmin': -10}```
And you need to use `Vcmap.value['cmap'].name` to get the value `jet`.
However for color table `pyart_NWSRef` you will get
```{'vmax': 65, 'cmap': 'pyart_NWSRef', 'vmin': -10}```
So in a pyart color table  `Vcmap.value['cmap']` is not an object and it gives you  color table value `pyart_NWSRef`.  There should not be this difference.

nguy, 2015-11-10:
This is interesting. We'll see what we can hunt down. Does this happen whenever you select a new color table or just with this color table or in general?

pfhein, 2015-11-10:
To learn how to make a plugin, I thought I would create a simple plugin that reports the color table.  Right now it works only at plugin startup and then only with the Display 1 Window.  So it reports the current color table at plugin startup.

pfhein, 2015-11-10:
Another thing is that when I start up artview  field variable `CZ ` is given the color table `jet` but if I chose a color table `jet` is not an option and only pyart color table values are options.  So maybe the problem is that  color table `jet` is being chosen instead of a pyart color table value.

nguy, 2015-11-10:
Gotcha. Okay, I'm looking through the code now to see exactly where these are pulled. Can you paste your code to see what is going on? 

It's weird that ```CZ``` is not bringing up bringing up the ```pyart_NWSRef```. There is a check in artview.scripts._parse_field that should catch CZ as a valid reflectivity.

nguy, 2015-11-10:
I'd like to see how 'jet' is being loaded, because when a ```RadarDisplay``` is started, the Py-ART colortables are loaded and supercede the matplotlib currently. When they are loaded, the string name is saved to the Vcmap.value['cmap'] field.

pfhein, 2015-11-10:
Artview has always loaded the same color table `jet` for `CZ`.  I think it is `jet` as it does not match any of the pyart color tables.  Or is it pyart overriding artview and not recognizing `CZ`.  Here is my plugin code.  Note I copied from your plugin and it is not done.  I think it needs a button to run it properly.
```
"""
cmap_value.py
"""

# Load the needed packages
from PyQt4 import QtGui, QtCore
import pyart
from .. import core, components

class CMapValue(core.Component):

    @classmethod
    def guiStart(self, parent=None):
        kwargs, independent = core.common._SimplePluginStart(
                                            "CMapValue").startDisplay()
        kwargs['parent'] = parent
        return self(**kwargs), independent

    def __init__(self, Vcmap=None, name="CMapValue", parent=None):
        '''
        Print out Color Table Name
        '''
        super(CMapValue, self).__init__(name=name, parent=parent)

        self.central_widget = QtGui.QWidget()
        self.setCentralWidget(self.central_widget)
        self.layout = QtGui.QGridLayout(self.central_widget)

        if Vcmap is None:
            self.Vcmap = core.Variable(None)
        else:
            self.Vcmap = Vcmap

        self.sharedVariables = {"Vcmap": None}
        self.connectAllVariables()

        self.generalLayout = QtGui.QVBoxLayout()
        # Set the Variable layout
        self.generalLayout.addWidget(self.createVarUI())
        self.layout.addLayout(self.generalLayout, 0, 0, 1, 2)

        self.show()

    ######################
    ##  Layout Methods  ##
    ######################

    def createVarUI(self):
        '''
        Mount the Variable layout.
        User may select another Display
        '''
        groupBox = QtGui.QGroupBox("Choose Display Window ")
        gBox_layout = QtGui.QGridLayout()

        self.dispCombo = QtGui.QComboBox()
        gBox_layout.addWidget(QtGui.QLabel("Select Display Link"), 0, 0)
        gBox_layout.addWidget(self.dispCombo, 0, 1, 1, 1)

        self.DispChoiceList = []
        self.components = core.componentsList
        for component in self.components:
            if "Vcmap" in component.sharedVariables.keys():
                self.dispCombo.addItem(component.name)
                self.DispChoiceList.append(component)
        self.dispCombo.setCurrentIndex(0)

        self.chooseDisplay()
        groupBox.setLayout(gBox_layout)

        return groupBox

    #########################
    ##  Selection Methods  ##
    #########################

    def chooseDisplay(self):
        '''Get Radar with :py:class:`~artview.core.VariableChoose`.'''
        selection = self.dispCombo.currentIndex()
        Vcmap = getattr(self.DispChoiceList[selection], str("Vcmap"))
        #print Vcmap
        #print dir(Vcmap)
        #print dict(Vcmap.value)
        #print dir(Vcmap.value['cmap'])
        self.dispCombo.setCurrentIndex(selection)

        self.Vcmap = Vcmap
        try:
            print self.Vcmap.value['cmap'].name
        except:
            print self.Vcmap.value['cmap']

    ################################
    #         Other Methods        #
    ################################

#    def Get_Cmap(self, variable, value, strong):
#        print self.Vcmap.value['cmap']

_plugins = [CMapValue]
```

pfhein, 2015-11-10:
One more data point.  If I start up artview.  It starts with `CZ` and color table `jet`.  If I immediately do a "Toolbox Reset File Defaults", it sets the color table to `pyart_Carbone17`, so default is not `jet` but `pyart_Carbone17`.

nguy, 2015-11-10:
Do you have a pyart.config file?

nguy, 2015-11-10:
Would you be able to share a file with me?

pfhein, 2015-11-10:
I do not have a pyart.config file.  What would you like me to share?

nguy, 2015-11-10:
A radar data file, sorry.
I've just tried another file with alternative (other and Py-ART) naming conventions and sure enough jet was used! @gamaanderson any thoughts off the top of your head why the field check does not seem to be working any longer (I *think* it worked before).

pfhein, 2015-11-10:
Here is where you can get my test file.  I use it unzipped for pretty much all my testing.
http://radarmet.atmos.colostate.edu/ftp/hein/cfrad.20111124_230003.000_to_20111124_230758.000_toga_v2.3_PPI.nc.gz

gamaanderson, 2015-11-10:
As long as I know we have fully implement the "follow pyart config" Principe, so  `._parse_field` is not being used or only in some insignificant place, you should forget about that function, and CZ will only be interpreted as reflectivity if pyart does so with the `pyart.config`.

About the original bug @pfhein, I already knew it and (even if it was inconsistent with the documentation), I didn't care enough to change it. To found where does that come from, you have to go after where RadarDisplay gets its cmap, it will try to use pyart, but if it fails it will fall back to a default, and display also has manual selection of cmap. One of those is giving matplotlib colormap instances instead of names. If you found it you can correct it, it is just a small commit.

Since you are already in writing a plug-in, something I would found really useful is some way to see the color-maps before selecting it. Now I can just select based in the name. This is of course a plug-in, you need not/should not create a button in display. First independent plug-in code, than we decide about user functionality.

nguy, 2015-11-11:
Ah I didn't realize we fully removed the function for ```_parse_field```, I actually think this should be selectively added back in if it selects the "correct" i.e. not 'jet' color map. Just my 2 cents.

Great idea on the colormap display plugin. Check out (this notebook)[https://github.com/nguy/Notebooks/blob/master/pyart_colormaps.ipynb] for an idea how to create the plot. It could then be added in a popup window pretty easily.

nguy, 2015-11-11:
@gamaanderson @jjhelmus I know this goes back to an earlier conversation we had regarding default configurations. 
I think at that time, we decided that Py-ART shouldn't set the defaults for ARTView, but also that we decided to check if a user had a configuration locally? Is that right?

We could check the system for the ```PYART_CONFIG``` environmental path and use the ```pyart.config``` if it is set, otherwise use the ```_parse_field```. Thoughts?

jjhelmus, 2015-11-11:
I believe the discussion was in ARM-DOE/pyart#329 with the implementation in ARM-DOE/pyart#336.  The `default_configuration.py` file in Py-ART contains a **DEFAULT_FIELD_COLORMAP** dictionary which maps field names to colormaps.  Right now these only map the "simple configuration" field names but additional names could be added to the dictionary, more can be added, like those in the `Zlike` variable.

gamaanderson, 2015-11-11:
@nguy I absolutely don't remember that decision, I would remember it, since I would have been opposed to it. I would say we should pass to Py-ART as much configuration as possible, since Py-ART has a well done config. The other possibility is to make a config file for ARTview, but why do it, if Py-ART give us it.

By the way, Py-ART implemented default colormaps and default limits at our request.

nguy, 2015-11-11:
I agree that we want to avoid ARTView config and use as much Py-ART config as possible. 

Okay it's coming back to me now (apparently my memory is crap!). I'd love to have something like the ```Zlike``` variable.

jjhelmus, 2015-11-11:
More fields added, ARM-DOE/pyart#427

pfhein, 2015-11-11:
In my playing around today, Py-ART will use the `jet` colormap if it does not recognize the variable. (It does recognize `CZ` as "Corrected Reflectivity" in the plot title but it does not for cmap.)  It would be nice if it defaulted to one of its own colormaps.  In ARTview  when an unknown variable like `CZ` is reset to "File Defaults" the colormap ends up being `Carbone17`.  I think making use of  pyart.config for ARTview is a good idea though people have to know it exists and how to set it up.  We will never cover every possible name.  Also I really like the idea of a `treat_this_field_like` option (along the lines of `Zlike`).

nguy, 2015-11-11:
```pyart.config``` is used in ARTView currently. @pfhein do you have a ```PYART_CONFIG``` environmental variable set? This would seem to be the only reason ```Carbone17``` is used.

@jjhelmus talk about bleeding edge :+1:

pfhein, 2015-11-11:
I need to learn about `pyart.config` now that I know it exists. I can not run fast enough to keep up. :-)

jjhelmus, 2015-11-11:
The docs on [pyart.load_config](http://arm-doe.github.io/pyart-docs-travis/user_reference/generated/pyart.load_config.html#pyart.load_config) and the [default_config.py](https://github.com/ARM-DOE/pyart/blob/master/pyart/default_config.py) file should be what you want to look at.  

Do note that if you decide to use a custom config you may be in a small minority of users who do this (or even the only one).  So if something breaks or does not work, let us know, but it might be a bit before someone (most likely me) can get to fixing it.

pfhein, 2015-11-11:
Thank you, @jjhelmus

gamaanderson, 2015-11-11:
FYI I use it!

nguy, 2015-11-11:
I probably should use it ;). But those new additions might just be what we need.

If this solves the issues @gamaanderson I would propose we do some house cleaning and move the _parse_field and any other unused code into a legacy directory.

nguy, 2015-11-11:
We'll also need to make mention in our README that it is HIGHLY recommended to update to the latest Py-ART as our development is happening pretty quick now.

pfhein, 2015-11-11:
@nguy  I do not have `PYART_CONFIG` set but it is the final colormap else so when you reset to file defaults the colormap is changed to `pyart_Carbone17`.  It is in `components/limits.py`.

gamaanderson, 2015-11-11:
I don't think it is HIGHLY recommended, those are just more comfort question. In all my previous alterations, I tried to always let a fall back, in case of an old pyart.

I don't think we should recommend the use of an in-development version of pyart, as soon as a new pyart release is done we can update.

nguy, 2015-11-11:
@pfhein Okay so that's where it is coming from. The reset defaults versus upon startup?
I think I put that color table last because so many people were familiar with that from Solo.

@gamaanderson This just means that we need to keep in touch with @jjhelmus to time releases a little in the future.

pfhein, 2015-11-11:
The color table is `pyart_Carbone17` on reset but on startup it is `jet`, so there is a difference.

gamaanderson, 2015-11-12:
the better would be to redirect the reset tool to `RadarDisplay._set_default_cmap` and `RadarDisplay._set_default_limits`. `GridDisplay` probably also need correction.

jjhelmus, 2015-11-13:
> @gamaanderson This just means that we need to keep in touch with @jjhelmus to time releases a little in the future.

Would there be an interest to synchronize the version numbers of Py-ART and ARTView?  Say for example that ARTView 1.6 is designed to work with Py-ART 1.6.0?  This would take a bit of coordination between all of us but should be possible.  I can open a separate issue on the topic is this is of interest.

gamaanderson, 2015-11-13:
ARTView is still a secondary project for us, even if we are spending a considerable amount of time in it. That means that we can not guaranty to get a new version ready in time. I believe we should stay the way it is, that is, ARTView runs after Py-ART. It is just a question of internal organization that when we bring something new to Py-ART we test its existence till a new stable version of pyart is released.

If there comes a case where that don't work, we discuss that case individually.

nguy, 2015-11-13:
I like the idea synchronizing, but agree with @gamaanderson that we may not be able to keep pace with Py-ART. It would be good to make it more clear which version of Py-ART is expected, maybe even adding this to the ```About``` menu. 

We may even consider adding a Py-ART version test during installation if we roll versions that require specific Py-ART releases.

gamaanderson, 2015-11-14:
We do say we need Py-Art >= 1.5 and testing dependencies in installation should actually be the norm.

nguy, 2015-11-16:
I agree. That way would throw warnings and maybe even create install log that could help diagnose install and usage issues as well.

jjhelmus, 2015-11-16:
Sounds like a plan.

closed 2016-01-21 by nguy