nguy/artview #110: Py-ART aux_io

nguy, 2015-11-10:
In ```artview.plugins.io```, we should use a logic-based approach for the enabling the ```pyart.aux_io``` read routines. 
Either a try: except: statement or test for each instance name in ```pyart.aux_io.__all__``` before loading.

jjhelmus, 2015-11-10:
Just FYI, all the aux_io read function are available in the pyart.aux_io namespace regardless of if the optional dependencies are available (ARM-DOE/pyart#411), so testing for the names will no longer work.  

You'll either need to catch the **MissingOptionalDependency** exceptions during read or examine the private testing variables in the various modules.  This is probably not ideal for ARTView but I've been trying to improve the Py-ART user experience with this change.  Also  aux_io is not intended to have full support in Py-ART so... it does not have full support.

gamaanderson, 2015-11-10:
@jjhelmus is using `__all__` a save way to get all (aux) read function (even if some will not work). In the end `__all__` is a private/special method, we probably should not use it.

jjhelmus, 2015-11-11:
`__all__` is fine to use for package (or in this case) sub-package introspection, and I believe is the preferred method..  The convention (maybe standard?) in Python is that `__all__` defines all the public names in the package or sub-package.  It is also used to determine what names get imported when importing * from a package, eg  `from pyart.aux_io import *`. 

In general names that begin and end with two underscores, `__variable__` reflect a builtin method of variable.  Other examples include `__name__`, `__file__` and `__doc__`.  

Variables with a single leading underscore are private variables but can still be access with their given names.  Two leading underscores with a single or no trailing underscores also indicates a private variable which may also be [name mangled](https://docs.python.org/2/tutorial/classes.html#private-variables-and-class-local-references) in classes to further protect them from unintended access.

gamaanderson, 2015-11-15:
using `__all__` don't work, it also list modules and I just want read functions.

jjhelmus, 2015-11-16:
Interesting.  I did not know that `from .module import func` adds module to the symbol table.  Apparently this is due to how imports work in Python (PEP 302).  The modules in pyart.aux_io could be deleted from the pyart.aux_io symbol table or `__all__` could be but explicitly defined with the function names so that only the functions/class are listed.  Apparently both of these are considered a bad practices because the modules are objects in that namespace.  Keeping the modules in there also makes debugging and testing easier.  Without them `pyart.aux_io.gamic_hdf5.read_gamic` would raise a AttributeError.

jjhelmus, 2015-11-16:
You can get a listing of all the functions in a the aux_io namespace using the [inspect]( https://docs.python.org/2/library/inspect.html) module.

``` Python
import pyart
import inspect
aux_read_functions = inspect.getmembers(pyart.aux_io, inspect.isfunction)
```

nguy, 2015-12-02:
This is a duplicate of Issue #79

closed 2016-01-21 by nguy