nguy/artview #111: Windows 10 install issues

nguy, 2015-11-11:
Some notes on a recent install of ARTView onto a Windows 10 OS.
The steps from the (openradar course)[https://github.com/openradar/AMS-Short-Course-on-Open-Source-Radar-Software] were followed to install Py-ART. 
ARTView was then installed. 

First thing is that the executable was installed in ```...\Anaconda2\Scripts\``` and not ```...\Anaconda2\Lib\bin```. The other thing is that the artview script does not work even after adding the correct directory to PATH variable. Going in and using ```python artview``` works properly, but it does not become a system executable.

gamaanderson, 2015-11-11:
pyart also was example script (e.g. anytocfradial), does those become executable?

nguy, 2015-11-11:
I tried ```radar_info```, but this worked just fine.

nguy, 2015-11-11:
I know this all worked previously on a Windows 8.1 install I did. So I'm trying to figure out it it's an Anaconda thing. I posted a quick question in a (closed thread)[https://github.com/ContinuumIO/anaconda-issues/issues/509].

msarahan, 2015-11-30:
Here's a quick summary of the situation:

1. Scripts and entry-point executables should be installed into PREFIX\Scripts
2. 3rd party programs and dlls are beginning to be (but are not always) installed to PREFIX\Library\bin.  Many official packages install here.  Some community packages still install to PREFIX\Scripts, which used to be a dumpster for all binaries.

To completely cover your bases, you need PREFIX;PREFIX\Scripts;PREFIX\Library\bin on PATH.  If you are familiar with UNIX/Linux, we treat PREFIX\Library as a folder very much like /usr/local, which makes installation more straightforward (we just point the installation prefix at PREFIX/Library)

nguy, 2015-12-08:
@msarahan Thanks for info. I've tried this now on two separate Windows 10 installs and have no luck running the packages.

I'm much more familiar with unix/linux and it's very possible that I'm not understanding something necessary on the Windows side. In addition web searches don't bring up this problem, so I expect this may be isolated.

nguy, 2016-06-27:
For some reason this issue has seemed to stop recently (I watched a user install from scratch and it worked!).  I still suspect that this was something to do with Anaconda, though I was never able to track this down. If we fixed it, that was accidental after trying many things.
So closing for now and we can reopen if it comes back up.

closed 2016-06-27 by nguy