nguy/artview #117: Menu Design

gamaanderson, 2016-01-17:
We really need a new menu Design, the current one is "broken" since introduce of the modes (that were a success in my perception! :+1:  ). It works, but i don't help the user. We must solve this before the next release. 

As I said before, I don't think I'm the person to this task. I would really appreciate if someone else pick it up, may be start from the Solo menu. I'm not asking for any code change, a simple structure Idea that I can implement is already enough.

nguy, 2016-01-18:
I agree it's time to revisit and modify.
I really like the modes, that makes it much more usable. I don't mind making these changes, but let's have a conversation here before we do. I'd like to hear the opinions of others about the menu design - @pfhein,  @tjlang , @swnesbitt @deeplycloudy

I'm in favor of minimizing the menu as best we can. I think Solo menus were too extensive, which is why I began to put some functionality directly in the display drop-downs.

One thought right away would be to build a small component with buttons to adjust tilt and advance files instead of the current ```Change File``` menu option.

pfhein, 2016-01-19:
I agree with keeping it minimal or simple.  Modes are a success!  They help the user know what the decisions/steps that need to happen.  

Here is my thinking. First we need to have good defaults so that 80% of the users don't need to do much to get their results.  We do not need to get rid of the complexity.  We just need to hide it away so that the 20% of users that need to do more can and the 80% are not confused by too many options. I am not certain what that looks like but if we can reduce the number of steps needed for them to get their results the better the experience will be.  That is my dream but I can think of some things that would require major rewriting of the code.  That is not going to happen since we all have jobs. So we need to keep our changes small and incrementally improve things following the ideas above.  

To start, for the menu window design, I would make the "LinkPlugins" default be Display1 and Display2 on startup and I would change the name to be "Links".  I think a lot of this is just refining what we have.  The mode tab "Select Region" does not tell the use he needs to click in the Display1 window to chose the region.  Details like that would be helpful.  I am not certain what to do with the Mode or Advanced Tools menu items.  They need to be short and yet descriptive.  Can we make them more descriptive? 

For now those are my thoughts.

nguy, 2016-01-20:
This is all good stuff. I agree with refining much of what we have. I have no objections to the proposed name changes. 

I wonder if we could use a tool tip for the Select Region issue you bring up. At the present it defaults to the first found display, so in theory that doesn't have to be Display1 if it has been closed (I think).

gamaanderson, 2016-01-20:
The menu structure is defined in script time, so we are free to put the names we want.

nick is right about the Select Region, so we don't really know what axes it was connected to. I don't see a good and secure way to tell the user what axes it is connect, my closes option would be to change the picture in some way, add a symbol or something. Never the less we can still say he needs to click in a display and hope the user knows or can found out witch one (as we have undo this is a save operation.). We need to make a (video)tutorial about how to change the linking and add this to the Select Region help (this is important I'm opening an issue!).

About the LinkPlugins, with a few changes I can define the start selection at script time. This will just work on start up, but not if the user open the LinkPlugins afterward.

nguy, 2016-01-21:
As a first step, is anyone again removing Advanced Tools from menu?
It appears that every function is duplicated.

I was thinking of the following name changes from Advanced Tools to modes:
RadarDisplay --> add_RadarDisplay (currently radar)
GridDisplay --> add_GridDisplay (currently grid)

The only function that is not present is AccessTerminal.

tjlang, 2016-01-21:
Is there a reason menu names can't be real English words, with spaces? The lack of descriptive names for menu options is a major culprit in the nonintuitive nature of some menus.

nguy, 2016-01-21:
Currently the key is used as a name. But, correct me if I'm wrong @gamaanderson we could make the modes dictionary a dictionary of dictionaries to solve this problem.

gamaanderson, 2016-01-21:
There is no reason not to use spaces, just change the keys of this [dictionary](https://github.com/nguy/artview/blob/master/artview/modes.py#L159-L170).

The advanced tools can be removed, but I still need to access it somewhere. I have no problem with it being hidden in some corner, but I still want to access it somehow.  It is important to bypass the modes and have a greater control over artview components.

nguy, 2016-01-21:
Let's do this in incremental steps. See PR #119.
[image]

tjlang, 2016-01-21:
W00t!

nguy, 2016-01-21:
I only want to disable the ```Advanced Tools``` menu text and leave the code in

nguy, 2016-01-21:
For the next PR (if this is acceptable above) I'll disable the Advanced Tools text. I was also thinking of moving the stuff under About under the File header. About artview and Help there.
And maybe put Short and Long Radar info under the Toolbox dropdown?

[image]
[image]

gamaanderson, 2016-01-21:
There is no reason to create a lot of PR, we can work over this one.

"Add radar and grid" is redundant with "add radar" and "add grid", we could remove it. Compare radars, is actualy link till, if there isn't 2 open DisplayRadar it open new ones. But first, I create the modes based in the scripts, I not sure we need this "compare" modes.

We could create a plugins to make a raw access to radar data, those option went in there (better than in display, I think). In the mean while we could remove the short and move the long to file.

nguy, 2016-01-21:
Okay, we can put the menu changes in one PR, that's easy enough.

I agree that Add Radar and Grid is redundant and will remove. Let's also nuke the compare mode. This can be done with Add Radar or Add Grid manually and maybe more effectively since the user would know what they are looking for.

nguy, 2016-01-21:
I actually really like the Short Info as a quick reference and think this would do well in the Toolbox.
The long radar info could go under File as "Export Radar Description" or such.

gamaanderson, 2016-01-21:
Could we put both in the files? it is not like the is too much there. And if we add new plugins for file access, this will not grow, we could even eventually remove the file menu and pass it to a component.

nguy, 2016-01-21:
Hmmm, that's interesting. So do you suggest we make a file access Plugin, maybe like FileList and open it the same (for now) from File menu?

gamaanderson, 2016-01-21:
I'm not sure if the user knows what he want in the case of compare. The best way of doing this compare is with the LinkPlugins, but if the user don't know he can do it, he will not come to the idea and he will also initially not understand LinkPlugings. That would be the argument to maintain the compare mode.

On the other hand, this is already the standard script, so clicking this mode has initially no effect (and users get confused with effect-less clicks. )

gamaanderson, 2016-01-21:
Yes, that could be done in the future. So the files menu become a tab, more or less like the new Microsoft office (>2003) (that I hate by the way!)

nguy, 2016-01-21:
I'm going to git rid of putting under the Modes heading, but the code remains alive if for future use.

nguy, 2016-01-21:
Ha! I agree and maybe we shouldn't go that way if we hate that style. But a FileAccess Plugin would be a nice alternative.

nguy, 2016-01-21:
I'm going to turn off the ```_add_all_advanced_tools```. I think this is the most robust way.

gamaanderson, 2016-01-21:
ok,

just for information if you want the advanced tools as a sub-menu of file you would change line

    menu.addMenuAction(("Advanced Tools",), action)

to some thing like this

    menu.addMenuAction(("File","Advanced Tools"), action)

gamaanderson, 2016-01-21:
the is no need to define "File" before it, if the menu doesn't exist it will create it.

nguy, 2016-01-21:
Okay, great. By removing this we effectively get rid of the Plugins capability. I'd like to retain that is some way, but the current is too confusing. I think changing the name from Advanced Tools to Plugins (I know we've had this conversation before) and sub-menu it under File might be good?

gamaanderson, 2016-01-21:
I don't know if File is the best place, but in the lack of a better one.

nguy, 2016-01-21:
I thought the same. But let's try it and then go from there. I'll move a couple of things around and then see what people think.

gamaanderson, 2016-01-21:
You may want to separe the diferent modes in diferent menus. Do you realize how this is done in the case of "file_list"? This is a place of code interdependence, so take care with the mode names!

gamaanderson, 2016-01-21:
I just realize you add a dict of dict, if you want that way, this is not a problem. But you also don't need to. The variable name (e.g. `radar_mode`) is not used any where else, just the key (e.g. `'radar'`). There is no rule that variable name = key + '_mode'.

nguy, 2016-01-21:
This last point was purposeful. That way I could maintain a 'label' and 'action' key. It can be changed later if it becomes impractical.

gamaanderson, 2016-01-21:
ok

nguy, 2016-01-21:
Separating ```file_list``` was merely a conditional statement it appears. Let's see what it looks like when we reorganize and then we could split modes. But you may run into people calling operations different names or not realizing the category we select for each mode.

gamaanderson, 2016-01-21:
that is right. But basic (if we don't change our minds) almost every new development will be accessible in form of a mode. So this menu will grow big.

nguy, 2016-01-21:
Let's start a separate Issue on this topic to change set up if required in future.

nguy, 2016-01-21:
So here is the current setup. I find this cleaner, what do you all think?
Radar will go away in lieu of another Component like Filelist
[image]

gamaanderson, 2016-01-21:
ok nice, new question. Do we need (want) the "change file" menu? It is kind of raw. If we think it is important, we should develop it some how (now or in the future), if not we should think in a way to replace it (again now or in the future)

tjlang, 2016-01-21:
This looks good.

gamaanderson, 2016-01-21:
also, the modes are not in order. We should put then in a meaningful order. This is kind of trick, as dicts have no order.

nguy, 2016-01-21:
Thanks @tjlang !
I'm going to number to keep things straight.
1. The Change File menu is going to go away. It's clunky. I am going to work on creating a button based component with common icons. e.g.
[image]

The question I have here is do we include these directly on the displays (I think yes due to the ability to unlink radar in displays) or do we have a standalone component?

2. I don't know how to order the modes. Any ideas?

nguy, 2016-01-21:
2. I don't know how to order the modes. Any ideas?

gamaanderson, 2016-01-21:
1. While you are doing this, put also the name of the files, so one knows to which file he will change before he does it. Also it should be an standalone component, as Display is only responsible for displaying, not for changing files.

nguy, 2016-01-21:
Okay, I'll put something together and put it in the PR. I've been committing the changes one at a time to a single PR and we'll merge it when we deem it ready.

gamaanderson, 2016-01-21:
"2". order modes in groups (e.g display, extract, modification). If you question was about the practial implementation: you need a list of some kind: for instance you transform the dict of dict in a list of dict (the keys get lost!), or you make a list of keys (code redundancy = not good!). There are also ordered dicts, but you need to import from a library.

nguy, 2016-01-22:
Okay pushed the latest change, no more About or Radar in menu:
[image]

I have an [open Issue](https://github.com/ARM-DOE/pyart/issues/458) to figure out how to save the radar.info() output to a file.

nguy, 2016-01-22:
@gamaanderson - Just to revisit something you said earlier: "also, the modes are not in order. We should put then in a meaningful order. This is kind of trick, as dicts have no order. "
would making this a list be an easier method? You may have suggested that, sorry if I didn't get it.

jjhelmus, 2016-01-22:
> There are also ordered dicts, but you need to import from a library.

The OrderedDict[https://docs.python.org/2/library/collections.html#collections.OrderedDict] class is available in the Python standard library in the collections module as of Python 2.7.  For Python 2.6 you need to use a third-party library.

OrderedDics are nice, once I drop the Python 2.6 dependency from Py-ART I plan to uses the a bit.

nguy, 2016-01-22:
I'll look into that. It was an easy switch to a list for now.

nguy, 2016-01-25:
Can you take a look at PR #119 ? 
I think the Menu is how I'd like to see it. I have begun a new ```FileNavigator``` feature to replace and extend the old ```Change File``` menu. I do not have this connected correctly and am struggling a bit to figure this one out. Any ideas would be welcome.

nguy, 2016-02-04:
If there are no further comments, I'll close this thread as I feel it's run it's course for the time being.

gamaanderson, 2016-02-04:
I agree with clossing

closed 2016-02-05 by nguy