nguy/artview #118: Change SelectRegion linking

gamaanderson, 2016-01-20:
changing witch display is connected to SelectRegion is not really trivial, but is something users will want to do. Therefore we need to create a video tutorial on how to do this and also add the link to this tutorial in the help.

Is the SelectRegion help up to date?

nguy, 2016-01-21:
Well I'm sure it could be expanded, but I think it's still correct.

gamaanderson, 2016-02-08:
here are new and updated tutorials:
SelectRegion: https://www.youtube.com/watch?v=Hcz2YtpXBdM
ManualUnfold: https://www.youtube.com/watch?v=B_BmYV7GdCA
ManualFilter: https://www.youtube.com/watch?v=VXUZBiA3HfU
ExtractPoints: https://www.youtube.com/watch?v=iWBFSN6Thbw
Linking SelectRegion: https://www.youtube.com/watch?v=cd4_OBJ6HnA
Opening: https://www.youtube.com/watch?v=eoF7tZu39JA

suggestion and correction are welcome. I don't think those videos are good, but someone gotta do it!

nguy, 2016-02-19:
I think these are a great start and helpful. Thanks for making and linking them.

gamaanderson, 2016-02-19:
In this case, I will add them to the help!

closed 2016-02-25 by gamaanderson