nguy/artview #120: Modes separation

nguy, 2016-01-22:
As discussed in Issue #117 it may be worth splitting modes into categories later on. This is to remind us to revisit this topic in the future.

nguy, 2016-04-21:
Following a comment in PR #142 adding separator lines in the Modes Menu could help in splitting into categories.

Also, I think we should add a separator line and move Close to the bottom in Files menu.

nguy, 2016-05-05:
Okay, we didn't do this in #142 b/c I forgot about it. I propose the following separation under the Modes Menu:
```
Add RadarDisplay
Add GridDisplay
Edit Colormap (Radar)
Edit Colormap (Grid)
----------------------
Map Radar to Grid
---------------------------
Apply Corrections to Radar
Apply Filter to Gates
Apply Filter to data
Manually unfold Velocity
---------------------------
Query a Selectable Region of Interest
Extract a Selected Region of Points
```

gamaanderson, 2016-05-05:
ok thank you, I will adress this sometime in the near future

nguy, 2016-05-05:
What are you working on next?

gamaanderson, 2016-05-05:
in my list there is the layout and the background map, but those are long time tasks.

nguy, 2016-05-05:
Okay great! Let me look at my task list again in the next couple of days and let you know what I'm going to work on.

gamaanderson, 2017-01-29:
can this be closed?

nguy, 2017-02-03:
Yes

closed 2017-02-03 by nguy