nguy/artview #121: GateFilter bug

tjlang, 2016-01-26:
Not every Py-ART Radar object features fields with masks. They may just be standard ndarrays. I'm not sure what controls this, but in my experience I've had to design workarounds for when this occurs. Thus, if you request the mask attribute from such a field (such as with a GateFilter), you will get an AttributeError, like so:

Traceback (most recent call last):
  File "//anaconda/lib/python2.7/site-packages/artview/plugins/gatefilter.py", line 378, in apply_filters
    self.Vradar.value.fields[field]['data'].mask)
AttributeError: 'numpy.ndarray' object has no attribute 'mask'

ARTview needs to be able to account for this, either with better exception handling or by constructing the mask from scratch. Below is an example of how I've worked around this issue in [PyTDA](https://github.com/nasa/PyTDA). With this function, I am attempting to pull standard ndarrays from Py-ART fields. However, in addition to just pulling the data out of a masked array, I need to deal with occasions when there is no mask, or if there is and there is no _FillValue dictionary item:

<pre>
def get_sweep_data(radar, field_name, sweep_number):
    # Check if _FillValue exists
    try:
        fill_value = radar.fields[field_name]['_FillValue']
    except KeyError:
        fill_value = BAD_DATA_VAL
    # Check if masked array
    try:
        return radar.fields[field_name]['data'][
            radar.sweep_start_ray_index['data'][sweep_number]:
            radar.sweep_end_ray_index['data'][sweep_number]+1][:].filled(
                fill_value=fill_value)
    except AttributeError:
        return radar.fields[field_name]['data'][
            radar.sweep_start_ray_index['data'][sweep_number]:
            radar.sweep_end_ray_index['data'][sweep_number]+1][:]
</pre>

jjhelmus, 2016-01-26:
The recommended method for accessing the mask of **MaskedArrayed** (including those where the mask is nomask) which also works with ordinary **ndarrays** is to use the [`numpy.ma.getmaskarray`](http://docs.scipy.org/doc/numpy/reference/generated/numpy.ma.getmaskarray.html) function which will return a boolean array of the correct size.

In the same manner [`numpy.ma.filled`](http://docs.scipy.org/doc/numpy/reference/generated/numpy.ma.filled.html) can be used to obtain a ndarray from a ndarray or MaskedArray where any masked elements are filled with a given value.

In general using the function in the [numpy.ma namespace](http://docs.scipy.org/doc/numpy/reference/routines.ma.html) is a good idea.

jjhelmus, 2016-01-26:
I'll submit a PR to fix this bug this evening. Been meaning to get into the ARTView code base and this is a good excuse.

gamaanderson, 2016-01-26:
thank you for reporting this bug @tjlang.

@jjhelmus is there a significant overhead of making all pyart arrays masked?

nguy, 2016-01-26:
@tjlang thanks for the report. I thought that I tested for this in the original code, but I guess not...
Thanks to @jjhelmus for tackling this.

jjhelmus, 2016-01-26:
From what I recall most of the Py-ART's read functions return a MaskedArray for the field data, but there may be a few that return ndarrays.  The one case that comes to mind is when reading Cf/Radial NetCDF files where the underlying variable does not have a missing_value on _FillValue attribute.  That data will likely be returned as an ndarray.

There is no compelling reason not to make all field data returned as masked arrays but I'm not sure if can be guaranteed.  Users can put whatever they please in the 'data' key of a field dictionary but hopefully most of the time they will put either a MaskedArray or a ndarray.   Py-ART tries to cover both of these two cases in the processing and analysis routines which at least in my opinion has not been especially burdensome.

PR #123 (2016-01-28): BUG: GateFilter plugin works on ndarray fields
The GateFilter plugin works on radar volumes which contain fields where the
data is stored as an ndarray not a MaskedArray object.

closes #121
commit 33f621df (2016-01-28): BUG: GateFilter plugin works on ndarray fields

The GateFilter plugin works on radar volumes which contain fields where the
data is stored as an ndarray not a MaskedArray object.

closes #121
closed 2016-01-28 by nguy