nguy/artview #122: FileList does not account for change of directory

nguy, 2016-01-26:
If a user stays within the initial directory opened, ```FileList``` works just fine. However, if the user opens a file in a different directory, the ```FileList``` returns to the current working directory used as an input at initialization.
If the user chooses Open once again the updated directory is shown.

nguy, 2016-01-26:
This parallels some issues I'm currently having in PR #119 where I'm trying to work through the Menu instance. I'm trying to figure our the best way to access the driving Menu instance.

gamaanderson, 2016-01-26:
this is component conflict, there are (just) two solutions:
1. one component surrenders (e.g. the open moved from menu to Navigator)
2. the two components talk to one an other = new V Variable for the directory

nguy, 2016-01-26:
I thought this might be the case. This is what I was trying to get at with a previous discussion. In this case a new Vfilelist or VFileCollection could be what is needed and allow optional stashing of radar instances.
This of course could cause memory issues as you pointed out previously.

nguy, 2016-01-26:
So the question is do we build the Vvariable in the components/menu.py ?

gamaanderson, 2016-01-26:
I would not mix VfileCollection with VradarCollection, for the reason I want this stashing the most transparent as possible, not to say you would need a class to handle this properly, a list would not do the job. I would also make it simple with a Vdirectory, but Vfilelist is also valid.

You could allow Navigator and Menu to be unrelated, just like `FileList` is. But for that the Navigator would need to look much more like the `FileList`, but this is (as I have found out) kind of tricky. In the way it is now, you need to add this Vvariable to `Menu`.

nguy, 2016-01-26:
I think Vfilelist is more descriptive as it will hold long file paths from which the directory can then be found.

I don't see how ```FileList``` is more unrelated than ```FileNavigator```.

nguy, 2016-01-26:
Oh and I think you are right about not mixing the collections.

gamaanderson, 2016-01-26:
When I change folder in the FileList it does not affect the Menu, but this is not a problem since it can navigate and open on its own. The most basic way of making `FileNavitator` independent if a change directory button.

gamaanderson, 2016-01-26:
But I don't think this is satisfactory, it would really need more. Creating the Vvariable does look the better option

nguy, 2016-01-26:
I actually think that when the user changes the folder it should be updated as the current working directory. And that starting the ```FileList``` should bring up the list of most recent accessed data.

nguy, 2016-01-26:
Working on a Vfilelist at the moment. I'll push up when finished.

gamaanderson, 2016-01-26:
well, that is not how it is working right now. An as `FileList` is a directory view in form of a list, it does not really work to list last accessed data.

nguy, 2016-01-27:
Yep, but with a Vfileliest variable, I think it could be made to do such.

gamaanderson, 2016-01-27:
I have totally mistaken this issue (that is what happens when I don't sleep!).

I was thinking it was about the `FileNavigator`. There I do see a need of linking to menu. In the `FileList` I was award of this from the beginning and didn't and don't see as a problem, I believe `FileList` should be sovereign over its directory, I would no like it to change responding to `Menu` (I want to open a file, does not means I want to see that directory) and that `Menu` does not respond to `FileList` is a side effect.

Also, in the case of `FileList`, a `Vfilelist` does no work, since `FileList` is actually a `DirectoryView` (I hate those wrong names) and I want it to be it. If you want `FileList` to be a view of an arbitrary list of files we need a new plugin for that (what about `FileNavigator`?).

Please re-read my comments, see if you can make sense of those.

nguy, 2016-01-27:
Ha, no problem! Hope you get some sleep soon!

I actually think this conversation was good. I'm implementing a Vfilelist, which helps keep the ```FileNavigator``` a separate entity as well and be a plugin much like ```FileList``` is currently. I hope to get the last bugs worked out soon. 

By the way, I'm not opposed to changing the name FileList to DirectoryView since this is more explicit to what is happening and I think clear to a user as well.

nguy, 2016-01-28:
@gamaanderson Should I rename ```FileList``` to ```DirectoryList``` or other in PR #119 ?

nguy, 2016-02-04:
Just wanted to note that with the introduction of ```Vfilelist``` with **FileNavigator**, this could be adapted into the **Filelist** (or **DirectoryList**) if desired.

gamaanderson, 2016-02-04:
no, this can no be done, since Vfilelist does not define a directory.

nguy, 2016-02-05:
It does now. I modified it so that the full path is recorded in the list and then use ```os.path.basename``` and os.path.dirname``` to split it apart.

gamaanderson, 2016-02-05:
okay, it really does, not because of that, but because the documentation says:

        +-----------+-------------------+------------------------------------+
        |Vfilelist  | Hold filenames in |list containing string file         |
        |           | current working   |basenames                           |
        |           | directory         |                                    |
        +-----------+-------------------+------------------------------------+

I find however an unnecessary limitation to restrict that list to a directory. Why can't I make a plugin where I search several directory and select files to add to the list. For instance I want to follow a storm that happened during the night, the files are separated by days, so before midnight is in one and after is in other.

nguy, 2016-02-05:
Oops, that was an oversight on my part not to update this info. Can you update this in your next PR?

That's a great idea on the plugin! The roll of day can be a real tough issue with data management.

gamaanderson, 2017-01-29:
can this be closed?

nguy, 2017-02-03:
Yep.

closed 2017-02-03 by nguy