nguy/artview #126: new norm argument

gamaanderson, 2016-02-03:
PyART recently introduced a new argument for non-linear colormap (see [here](https://github.com/ARM-DOE/pyart/issues/390#issuecomment-178825935)). I'm not really sure how to add this to artview, I would like some ideas (@nguy,@jjhelmus).

Any way, this is just for artview 1.3. We need to finish 1.2 still.

nguy, 2016-02-03:
My first instinct would be to add a 'norm' keyword to the Vcmap. Have it default to None which uses linear Normalize() function and then add the norm=cmap['norm'] argument to the plotting routines.

To allow adjustments, I guess we could add a dialog box something like the limits on the drop-down Display Options toolbar.

gamaanderson, 2016-02-03:
and how does that new key and the current vmin and vmax work together?

Of course, adding a norm argument make little sense with out a plugin to edit the colormap, but we have being needing some improvement in this any way.

nguy, 2016-02-03:
Having a value of None should not have any effect since the vmin/vmax are already passed to the plot routines. It could be instituted in the ```artview.components.plot_radar._set_default_cmap``` and plot_grid version.

We could institute an additional step of calculating the norm, but take look [here](https://github.com/nguy/AWOT/blob/master/awot/graph/radar_horizontal.py#L173). This is a little bit different implementation, but the idea is the same. I feel this is something that could be done behind the scenes in Py-ART, by using the cmap and a keyword of boundaries. What do you think @jjhelmus ?

nguy, 2016-02-03:
I agree that this would be a nice feature for ARTview.

jjhelmus, 2016-02-03:
How to deal with the `norm` parameter in ARTView is tricky.  

My idea of adding to parameter was to allow users to create a categorical or non-linearly scaled colormap by passing a Matplotlib [BoundaryNorm](http://matplotlib.org/api/colors_api.html#matplotlib.colors.BoundaryNorm) or [LogNorm](http://matplotlib.org/api/colors_api.html#matplotlib.colors.LogNorm) instance.  This unlocks some interesting plotting possibilities but also means that the user needs to do some work to create the object to pass in as the `norm` argument. 

ARTView could allow a users to switch to a log scaling colorbar without the addition of too much additional code, the LogNorm instance would need to be created inside ARTView.  As @gamaanderson mentioned the categorical option will require more work as first some method allowing the user to specify the bounds would need to be created.  

> and how does that new key and the current vmin and vmax work together?

If norm is specified as anything other than `None` then vmin and vmax are ignored by setting them to `None`.  This is handled by the [parse_norm_vmin_vmax](https://github.com/ARM-DOE/pyart/blob/master/pyart/graph/common.py#L70-L77) function which is called in all of the public `plot_` methods.

jjhelmus, 2016-02-03:
> We could institute an additional step of calculating the norm, but take look here. This is a little bit different implementation, but the idea is the same. I feel this is something that could be [...]

I like this idea but I'd rather is not be implemented as an additional argument to the `plot_` methods.  Those methods are getting quite clutters as it is.  I think a more maintainable solution would be to have some helper function which generate the `cmap` and `norm` parameters from a set of discrete levels, basically what the `from_levels_and_colors` code is doing in the linked code.  Then instruct users to use these helpers to generate the parameters and then pass them to the `plot_` methods.  The other option I see to clean up these methods is to use kwargs to catch everything and then parse that dictionary.

gamaanderson, 2016-04-02:
FYI: I have started working on this.

nguy, 2016-04-04:
I'm sorry for the delay. I'll look at it this week. I wanted to keep my ARTview at stable place for the presentation (today)!

closed 2016-05-05 by nguy