nguy/artview #127: Change in LinkPlugins

gamaanderson, 2016-02-05:
what do you think about this
[image]

I particularly find if ugly, but it is more verbose.

gamaanderson, 2016-02-05:
or may be this

[image]

nguy, 2016-02-05:
I would say the second image with "from *dropdown1*'  and 'to *dropdown2*' moved to the top of the page then this would have two columns and be more straightforward.

gamaanderson, 2016-02-05:
The reason I make like this, is that it is supposed to be a phrase 
`limits from Display1 is linked to Display2` 

if I make

    from Display1
    to Display2
    limits   |  is linked

this idea get lots, don't it?

nguy, 2016-02-05:
It could be something like:
```
Linking from Display1     to Display2
            limits                |  is linked
```

gamaanderson, 2016-02-05:
ok now I understand, I will try to see how it looks

gamaanderson, 2016-02-05:
it this it?
[image]

nguy, 2016-02-05:
I think that is much more straightforward. @tjlang @pfhein what do you guys think?

pfhein, 2016-02-05:
Yes, I like the latest version. To nitpick, `Linking from Display1 to Display2` makes it sound like the linking only goes one way.  Would `Display1 linked to Display2` be better? Or something else?

gamaanderson, 2016-02-05:
the linking process in not only one way, but it is asymmetrical:

initially `Display1` has `value1` and `Display2` has `value2`
one of those must go away, in this case `value1` is lost and `Display1` will end with `value2`.

pfhein, 2016-02-05:
Yes, but how to say that. Maybe another way to say it is `Display1 and Display2 linked together` or maybe better just say `Display1 and Display2 Shared Variables` and below you can see which ones are linked.

gamaanderson, 2016-02-05:
I choose the use of preposition to emphasize that
`from Display1 to Display2`
is not the same as
`from Display2 to Display1`

pfhein, 2016-02-05:
Okay - I am being just dense.   Asymmetrical means asymmetrical.  Sorry!  Your wording then makes more sense.

gamaanderson, 2016-02-06:
I added a help button. 
[image]

nguy, 2016-02-09:
Nice, that provide some good help I think.

closed 2016-02-25 by gamaanderson