nguy/artview #129: Release 1.2

gamaanderson, 2016-02-06:
I have the impression there are several small things that need to be done before release 1.2 and I'm afraid I may forget something so I would like to open this issue for reference.

I also invite people to start looking for error, in the current master or in PR #128.

- [x] issue  #61
- [x] issue #127 
- [x] issue #118 
- [x] issue #124

nguy, 2016-02-09:
+1 for the list!

The one thing I think could wait is the select region stuff. This could be a sub version of 1.2 in my opinion.

gamaanderson, 2016-02-09:
if you mean the SelectRegion_dev, this was kind of a question on you. But I would like to know why; what is missing in the dev version?

nguy, 2016-02-10:
Honestly I haven't looked at it in a while. Let me take a look and get back to you.

nguy, 2016-02-19:
@jjhelmus When we get this further along, could you help us build a conda package?

jjhelmus, 2016-02-22:
I'd be happy to help get a conda package built for ARTView, just ping me when you make or are about to make the release and I can start testing out some recipes.

nguy, 2016-02-22:
Thanks! It should be pretty soon.

jjhelmus, 2016-02-22:
Great, I'm traveling this week so might not be as prompt, but I'll keep an eye out.

nguy, 2016-02-22:
@gamaanderson What do you propose for the select region? I would like to keep the display functionality, but agree that having two separate methods will become problematic.

gamaanderson, 2016-02-22:
what funcionalities of do you like to mantain,  that you cannot do in one way or the other in with the new version? I think they are all there.

gamaanderson, 2016-02-22:
@jjhelmus what would be the install instruction once the conda package is made?

jjhelmus, 2016-02-22:
Run `conda install -c jjhelmus artview`.  The jjhelmus part could be replaced with another conda "channel", that has just need where I've been putting the packages I've built.

nguy, 2016-02-22:
I believe the functionality is there, I meant that I'd like to maintain the setup from the Display as it is for now in terms of easily plotting histogram, bringing up table, applying a mask, etc. The newer select region is more modular probably and therefore in the long run more flexible. I just want to make sure we don't lose functionality for the user.

nguy, 2016-02-22:
This sort of goes back to a way back discussion I had with @scollis about open source radar or meteorology having a channel on conda. Would this be something to explore?

scollis, 2016-02-22:
@jjhelmus is the Guru here.. But yes.. this would be a great idea..

gamaanderson, 2016-02-23:
@nguy since the code is already there, I have no problem to let the current SelectRegion accessible through the display. But it will become a part of display and lose its status as a components, therefore not accessible through the artview module or any mode.

nguy, 2016-02-23:
I know it's a balance, but I feel that this is a tool for quick data querying. Let's keep it for the time being, with the thought of any other tools developed should use the newer version. Do you think it wise to rename the code to indicate that it is display-based and not a component?

For example, we could rename the class from ```SelectRegion``` to ```DisplaySelectRegion``` or something and have display prominent in all naming. We could then also rename ```SelectRegion_dev``` to ```SelectRegion```.

gamaanderson, 2016-02-23:
ok, I will rename it.

nguy, 2016-02-23:
Thanks @gamaanderson. Could you also replace the the contact in the README file to the user forum email?

nguy, 2016-02-24:
We can probably close some of the outstanding Issues that have been addressed as well. Let me know if you want me to do any of this.

nguy, 2016-02-26:
Okay, it appears that rebuilding docs and rolling version are the two parts that are left. I'll do that today and submit a PR.

At this point we should probably then build the conda package and go forward with the release. If we manage to get this sorted, it would be fun to release on Monday since it's a leap year!

nguy, 2016-02-26:
Ended up pushing the new docs and rolling the rev directly into the master.

@jjhelmus Can we get started with the conda release if you have time?

gamaanderson, 2016-02-26:
Do you think we should do a "what new" in relation to the latest release? It is so much, that I have my doubts if it makes sense.

nguy, 2016-02-26:
I think that's a great idea. We could put this in the README maybe. Then I was going to post that What's New and announcement in the Users forum and the pyaos mailing list.

gamaanderson, 2016-02-26:
The README is already too long!

nguy, 2016-02-26:
Ha, that is true! We should probably move some of this to the Documentation.

jjhelmus, 2016-02-29:
@nguy I'll start working on the artview conda build this week.  I'll keep you all updated if I run into any question.

nguy, 2016-03-01:
@jjhelmus I just pushed a minor update. I don't think this affects the conda build, but just in case....

gamaanderson, 2016-03-02:
yes, it does afect. If the conda package is build from the old code then the tool would continue broken.

jjhelmus, 2016-03-02:
I created test artview conda packages from the ae1331f56 commit which I labeled as "1.2.3".  These are available on my [Anaconda.org channel](https://anaconda.org/jjhelmus/artview) with the label "testing".  

If you use `git tag` to [tag a release](https://git-scm.com/book/en/v2/Git-Basics-Tagging) it will be easier to create the conda package for 1.2.  Just let me know the tag/commit that should be considered 1.2 for the conda package and I can update the package quickly.

gamaanderson, 2016-03-02:
Thank you a lot @jjhelmus 

@nguy PR #132 can be merged. I would also like to call attention to the `ISRELEASED` flag in **setup.py** line 57, in case you haven't already seen it.

nguy, 2016-03-02:
I want to make sure I follow what to do here. 

1. I can merge #132 (with the ISRELEASED flag turned on - thanks @gamaanderson I did not see this.)
2. Apply the tag to it with something like conda release as description. 
3. Do I need to do a formal release via Github also?
4. After all of this the "real" artview conda package will be released.

nguy, 2016-03-02:
Or do I need to just do step 2 and the rest can come afterward. Sorry, try to figure out the workflow here.

jjhelmus, 2016-03-02:
You can do steps 3 and 4 anytime after after you change ISRELEASED to True and tag the release, there is no immediacy.  Setting ISRELEASE to True indicates that commit corresponds to the "official" release version and tagging makes it easier to find rather than having to use the commit hash.  It is good to make a second commit immediately after to flip ISRELEASED back to False and roll the version number.  

Examples from the last release of Py-ART:

* [REL Release 1.6.0](https://github.com/ARM-DOE/pyart/commit/7876d8733d43818e752e009ef51e3e569f331501)
* [MAINT: start v1.7+dev](https://github.com/ARM-DOE/pyart/commit/c54a29916defa36b8f16ef849c8c1c35a7d755dd)

There are other workflows that projects use for this process that involves release branches, etc, but I've never found that necessary.

nguy, 2016-03-02:
Thanks @jjhelmus for walking us through this!

Okay the tag is done.

nguy, 2016-03-02:
The method for Py-ART seems to be a good one to follow. @gamaanderson any thoughts?
This would give us a benchmark for development.

jjhelmus, 2016-03-02:
FYI, `git push` doesn't push tags by default, you need to use `git push --tags`.

nguy, 2016-03-02:
done, thanks

nguy, 2016-03-02:
FYI, this will close Issue #105 I believe.

Also, @jjhelmus would you be willing to help with future conda releases? If so, we should ping you when appropriate?

jjhelmus, 2016-03-02:
Happy to.  I'll set up the system for building the conda package on [conda-forge](https://conda-forge.github.io/) so that anyone can make a release by submitting a pull request.

nguy, 2016-03-02:
Great! You'll probably need to lead me through that as well eventually.

nguy, 2016-03-02:
We should sit and chat at SEA conference to discuss parallel developments.

jjhelmus, 2016-03-02:
Sounds like a plan.

gamaanderson, 2016-03-02:
I have always been in favor of following pyart example when applicable.

nguy, 2016-03-02:
Me too, let @jjhelmus do the hard work. ;p

nguy, 2016-03-02:
Okay back on topic, do we need anything else for the conda package then?

jjhelmus, 2016-03-02:
A full suite of conda packages for artview 1.2 made and available on [Anaconda.org](https://anaconda.org/jjhelmus/artview/files).  I need to fix a few items on conda-forge before moving the build there.

nguy, 2016-03-02:
Excellent, thank you again and let us know anything else we can do to help.

jjhelmus, 2016-03-07:
I created a [conda-forge feedstock for artview](https://github.com/conda-forge/artview-feedstock/) which allows for reproducible builds of the conda package using online CI services.  Anyone can submit a pull request that repository when updates are needed.  The command to install the packages produced by this setup is `conda install -c https://conda.anaconda.org/conda-forge artview`.  I prefer this more open method for building but will continue to maintain the packages on my personal anaconda.org channel, jjhelmus.

nguy, 2016-03-08:
The official V1.2 release is in the wild!

nguy, 2016-03-09:
Realized I forgot to do the ISRELEASED. So pushed this as REL commit and updated the Release tag.
Then pushed a v1.3dev commit afterwards.

gamaanderson, 2016-03-10:
Can this be closed?

nguy, 2016-03-10:
Yep

closed 2016-03-10 by gamaanderson